espressif/esp-skainet #2: The board can't be waken up

Tflgit, 2019-10-28:
I have run the demo "garbage_classification" on the board ESP32-LyraT V4.3, but the board can't be waken up by the word "hilexin". So I tryed to debug did it read something in "**ringbuf**". Here is the log
I (29) boot: ESP-IDF v3.3 2nd stage bootloader
I (29) boot: compile time 09:29:28
I (29) boot: Enabling RNG early entropy source...
I (33) qio_mode: Enabling default flash chip QIO
I (38) boot: SPI Speed      : 80MHz
I (42) boot: SPI Mode       : QIO
I (46) boot: SPI Flash Size : 4MB
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 factory          factory app      00 00 00010000 003c0000
I (69) boot:  1 nvs              WiFi data        01 02 003d0000 00004000
I (76) boot: End of partition table
I (81) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2c0394 (2884500) map
I (853) esp_image: segment 1: paddr=0x002d03bc vaddr=0x3ffb0000 size=0x02220 (  8736) load
I (856) esp_image: segment 2: paddr=0x002d25e4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/tfl/workspace/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (860) esp_image: segment 3: paddr=0x002d29ec vaddr=0x40080400 size=0x0d624 ( 54820) load
I (887) esp_image: segment 4: paddr=0x002e0018 vaddr=0x400d0018 size=0x2b0e0 (176352) map
0x400d0018: _flash_cache_start at ??:?

I (933) esp_image: segment 5: paddr=0x0030b100 vaddr=0x4008da24 size=0x01834 (  6196) load
0x4008da24: xTaskResumeAll at /home/tfl/workspace/esp/esp-idf/components/freertos/tasks.c:4560

I (945) boot: Loaded app from partition at offset 0x10000
I (945) boot: Disabling RNG early entropy source...
I (945) psram: This chip is ESP32-D0WD
I (950) spiram: Found 64MBit SPI RAM device
I (954) spiram: SPI RAM mode: flash 80m sram 80m
I (960) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (967) cpu_start: Pro cpu up.
I (971) cpu_start: Application information:
I (975) cpu_start: Project name:     garbage_classification
I (982) cpu_start: App version:      v0.1.0-10-g74d715f-dirty
I (988) cpu_start: Compile time:     Oct 24 2019 09:29:25
I (994) cpu_start: ELF file SHA256:  34e2ddf0cebbc776...
I (1000) cpu_start: ESP-IDF:          v3.3
I (1005) cpu_start: Starting app cpu, entry point is 0x400815cc
0x400815cc: call_start_cpu1 at /home/tfl/workspace/esp/esp-idf/components/esp32/cpu_start.c:270

I (0) cpu_start: App cpu up.
I (1505) spiram: SPI SRAM memory test OK
I (1505) heap_init: Initializing. RAM available for dynamic allocation:
I (1505) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1512) heap_init: At 3FFB33A0 len 0002CC60 (179 KiB): DRAM
I (1518) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1524) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1531) heap_init: At 4008F258 len 00010DA8 (67 KiB): IRAM
I (1537) cpu_start: Pro cpu start user code
I (1542) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (220) cpu_start: Chip Revision: 1
I (220) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (225) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (235) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (245) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (245) I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
I (275) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (285) MEDIA_HAL: I2S_OUT_VOL_DEFAULT[60]
I (285) MSC_DSP: **CONFIG_CODEC_CHIP_IS_ES8388**
I (285) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:0 
Quantized wakeNet5: wakeNet5_v1_hilexin_5_0.95_0.90, mode:0 (Oct 11 2019 20:08:50)
Quantized MN1 (Oct 14 2019 20:06:13)
I (305) MN: ---------------------SPEECH COMMANDS---------------------
I (315) MN: Command ID0, phrase 0: wei sheng zhi
I (315) MN: Command ID0, phrase 1: shi zhi jin
· · · · · ·
I (565) MN: ---------------------------------------------------------

chunk_num = 200
@count 1 read 960 bytes
@count 2 read 960 bytes
@count 3 read 960 bytes
@count 4 read 960 bytes
-----------awaits to be waken up-----------
@count 5 read 960 bytes
@count 6 read 960 bytes
· · · · · ·

From the log, the board frequently reads audio data, even if no one speaks. Is this data noise? Why can't the board recognize the wake-up command?

WeldonWangwang, 2019-10-29:
Hi Tflgit,
To run ESP-Skainet, you need to have an ESP32-LyraT-Mini. ESP-Skainet currently does not support ESP32-LyraT V4.3. So if you use ESP32-LyraT V4.3 without modifying the driver code, the I2S will not work.
**We will update ESP-Skainet to support ESP32-LyraT V4.3 recently**. Before we update, if you are familiar with ESP-ADF, you can also refer to ESP-ADF to update the hardware driver yourself.

Tflgit, 2019-10-29:
@WeldonWangwang ，I have modified the I2S_NUM in the app code file "recsrc.c", and added the board config choice CONFIG_ESP_LYRAT_V4_3_BOARD with CONFIG_CODEC_CHIP_IS_ES8388. Now, this device is running,  you can see in the log.

Tflgit, 2019-10-29:
By the way, the ESP_IDF version is 3.3. Is this compatible?

WeldonWangwang, 2019-10-29:
ESP32-LyraT V4.3 has two mics. How many channels do you configure when configuring I2S in 'components/hardware_driver/MediaHal/MediaHal.c '?

And the '.codec_set_adc_input = Es8388ConfigAdcInput,' should be modified to '.codec_set_adc_input = Es8374ConfigAdcInput,'

All of the features of ESP-Skainet have been tested based on ESP_IDF_3.2, so it is recommended to use ESP_IDF_3.2, we will also perform compatibility test verification for 3.3 recently.

Tflgit, 2019-10-29:
I2S_NUM is configured to I2S_NUM_0.

The audio codec chip is ES8388(CONFIG_CODEC_CHIP_IS_ES8388), Should the .codec_set_adc_input be defined to Es8374ConfigAdcInput?

There is not version 3.2 branch in some submodules of ESP_IDF.
git clone -b v3.2 --recursive https://github.com/espressif/esp-idf.git
正克隆到 'esp-idf'...
remote: Enumerating objects: 1684, done.
remote: Counting objects: 100% (1684/1684), done.
remote: Compressing objects: 100% (966/966), done.
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: The remote end hung up unexpectedly
fatal: 过早的文件结束符（EOF）
fatal: index-pack failed

WeldonWangwang, 2019-10-29:
There is an error in the code for 8388 in the current version. You need to change the function Es8388ConfigAdcInput to:
int Es8388ConfigAdcInput(AdcInput input)
{
    return 0;
}

or just modify '.codec_set_adc_input = Es8388ConfigAdcInput,' should be modified to '.codec_set_adc_input = Es8374ConfigAdcInput,'

The problem with clone IDF is due to excessive resources and network speed problems. Please try again.

WeldonWangwang, 2019-10-31:
@Tflgit  The latest code in the master branch supports ESP32-LyraT V4.3 now, please update the repository and try again.

Tflgit, 2019-10-31:
@WeldonWangwang Thanks! Now, the board can be waken up.

closed 2019-10-31 by Tflgit