espressif/esp-skainet #3: 更换唤醒词，出现驱动解码芯片出错！

xuhongv, 2019-10-29:
- 我给客户演示了这个仓库的效果，还不错！就是想换个 唤醒词，刚好看到你们那有《你好小智》这个唤醒词，然后我改了下，竟然串口上电很大几率出现如下截图的红色字符，貌似是初始化解码芯片出错了，但是我换回来 唤醒词 “ 你好乐鑫” 又没事了！

- 也许是这个原因，导致后面无法唤醒板子；

[image]

xuhongv, 2019-10-29:
- 再上一个“hi 乐鑫”唤醒词的串口打印，说明我的板子是没问题的：

[image]

xuhongv, 2019-10-29:
- 找到问题了，如果是 电源供电 上电就会出现第一个图的异常，但是如果是 lyart-mini 板子的 rst 按键复位重启，就正常了，硬件问题？？

WeldonWangwang, 2019-10-29:
你好，这是 codec 驱动中的一个问题，因为在起 codec 时 MCLK 和 codec 顺序不一致导致的，我们将会在最近进行更新！

xuhongv, 2019-10-29:
- 还有一个问题，那个播放的语音“有害垃圾”，“回收垃圾”是 以数组形式定义的，请问下，这个数组是 如何得到的？ 语音转数组？类似 mp3 文件转数组？

WeldonWangwang, 2019-10-29:
是的，为了方便播放，可以将 wav 文件（16K，16bit，单声道）转为数组进行播放，如果有需要我们后续可以开放 wav 转为 '.h' 文件的脚本。

xuhongv, 2019-10-29:
- 感谢！迫切需要！！

closed 2019-10-29 by xuhongv

Shaopus, 2019-11-04:
> 是的，为了方便播放，可以将 wav 文件（16K，16bit，单声道）转为数组进行播放，如果有需要我们后续可以开放 wav 转为 '.h' 文件的脚本。

你好，能提供下这样的脚本嘛？谢谢~

WeldonWangwang, 2019-11-08:
@xuhongv 输出语音转数组的工具已经更新，烦请使用 master 分支最新代码，在 tools 中既可进行 wav 转数组