espressif/esp-skainet #4: 降噪demo，耳机没有声音

Tflgit, 2019-10-31:
我测试“noise_suppression”demo时，耳机没有声音，调试串口也没有输出。
使用的是手机耳机接"PHONEJACK"口。
唤醒词选择是nihaoxiaozhi (WakeNet5)。
这个demo需要先唤醒吗？

WeldonWangwang, 2019-11-04:
@Tflgit 该 demo 不需要唤醒即可使用，请问您使用的是什么开发板？

Tflgit, 2019-11-05:
我使用的开发板是ESP32-LyraT V4.3，代码已经更新到最新。
I (217) cpu_start: Pro cpu up.
I (221) cpu_start: Application information:
I (226) cpu_start: Project name:     noise_suppression
I (232) cpu_start: App version:      v0.1.0-20-g060e269
I (238) cpu_start: Compile time:     Nov  5 2019 08:46:26
I (244) cpu_start: ELF file SHA256:  e3e2d7ecff50d685...
I (250) cpu_start: ESP-IDF:          v3.3
I (254) cpu_start: Starting app cpu, entry point is 0x400814a0
0x400814a0: call_start_cpu1 at /home/tfl/workspace/esp/esp-idf/components/esp32/cpu_start.c:270

I (0) cpu_start: App cpu up.
I (754) spiram: SPI SRAM memory test OK
I (754) heap_init: Initializing. RAM available for dynamic allocation:
I (755) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (761) heap_init: At 3FFB30B8 len 0002CF48 (179 KiB): DRAM
I (767) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (773) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (780) heap_init: At 4008E890 len 00011770 (69 KiB): IRAM
I (786) cpu_start: Pro cpu start user code
I (791) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (139) cpu_start: Chip Revision: 1
I (140) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (144) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (154) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (164) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (164) I2S: APLL: Req RATE: 16000, real rate: 15999.986, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 4095996.500, SCLK: 511999.562500, diva: 1, divb: 0
I (194) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (194) MEDIA_HAL: I2S_OUT_VOL_DEFAULT[60]
I (204) MEDIA_HAL: CONFIG_CODEC_CHIP_IS_ES8388
NS STATE: 1

WeldonWangwang, 2019-11-05:
从打印看，是正常启动的，并且显示上电时降噪为开，烦请查看耳机孔是否插紧或是硬件有损～

closed 2020-04-01 by feizi