espressif/esp-skainet #6: 输出语音如何制作成数组？

Shaopus, 2019-11-08:
你好，

请问一下，如何将语音通过喇叭输出，我在之前的Issues中看到这个[WAV](https://github.com/espressif/esp-skainet/issues/3#issuecomment-547310511)，请问是这些步骤是怎么操作？

请尽快回复，谢谢！

WeldonWangwang, 2019-11-08:
@Shaopus 已经更新，烦请使用 master 分支最新代码，在 tools 中既可进行 wav 转数组。

Shaopus, 2019-11-08:
> @Shaopus 已经更新，烦请使用 master 分支最新代码，在 tools 中既可进行 wav 转数组。

Thank you！我已更新。

但是我有个问题，怎么才能生产 16k/16bit WAV 的音频，有什么方法吗？

请尽快回复，谢谢~

xuhongv, 2019-11-08:
@WeldonWangwang 

这个脚本使用方法是如何使用？直接扔 wav文件放在同一级？ 我不懂 py ，麻烦使用文档详细一点

xuhongv, 2019-11-09:
[image]
@Shaopus 你看看，这个应该可以的

xuhongv, 2019-11-09:
[image]
@WeldonWangwang 报错

Shaopus, 2019-11-11:
> [image]
> @WeldonWangwang 报错

在前面加python，再执行

xuhongv, 2019-11-11:
@WeldonWangwang  @Shaopus 没辙了，我自己手机录音出来的 mp3 文件，之后通过迅捷音频转换为 单声道 16bit/s ，再通过脚本，发现播放出来的声音播放慢且卡顿；

Shaopus, 2019-11-11:
> @WeldonWangwang @Shaopus 没辙了，我自己手机录音出来的 mp3 文件，之后通过迅捷音频转换为 单声道 16bit/s ，再通过脚本，发现播放出来的声音播放慢且卡顿；

- [在线语音转换](http://tools.bugscaner.com/tts/)，选择语速3，合成，导出，
- 再用迅捷音频转换工具转换，选择WAV格式，16bit，导出，然后右键属性查看比特率是否是256kbps
- 最后使用tools工具进行转换成数组

用了上述操作，我的已经没有问题了，你可以试试看

xuhongv, 2019-11-11:
嗯嗯！已经ok，非常感谢！
@Shaopus

closed 2019-11-11 by feizi