espressif/esp-skainet #7: 使用Cmake编译报错

HoGC, 2019-11-10:
以将组件放置在工程的components目录下，但是编译还是报错，报如下错误
```
CMake Error at F:/esp/components/esp-sr/CMakeLists.txt:15 (target_link_libraries):
  Cannot specify link libraries for target "esp-sr" which is not built by
  this project.
```

feizi, 2019-11-11:
请参考 examples/get_started 中 CMakeLists.txt 设置

HoGC, 2019-11-11:
项目CMakeLists.txt没有错误，我已经组件放到项目下的components目录中，所以并不需要设置另外的组件搜索目录。编译错误是出现在esp-sr组件中的CMakeLists.txt，无法使用target_link_libraries链接到静态库

feizi, 2019-11-14:
你可以编译 get_started 吗？

HoGC, 2019-11-14:
编译后报一样的错误。
还尝试了espressif的docker的编译镜像，一样的问题。
```
CMake Error at F:esp-skainet/components/esp-sr/CMakeLists.txt:15 (target_link_libraries):
  Cannot specify link libraries for target "esp-sr" which is not built by
  this project.
```

feizi, 2019-11-15:
我们测试get_started是可以的，你检查一下 submodule是不是正常下载了，或是使用  
git clone --recursive https://github.com/espressif/esp-skainet.git 克隆仓库

HoGC, 2019-11-15:
你们测试的是最新的IDF吗

feizi, 2019-11-18:
我们测试的是 IDF v3.2

XiaochaoGONG, 2019-11-20:
尝试将提示错误的 CMakeLists.txt 中的 “esp-sr” 替换成 ${COMPONENT_TARGET}

feizi, 2019-12-03:
If you get the same compilation error：  
method1： replace your IDF with the IDF in ESP-Skainet.
method2： open the CMakeList.txt in ./components/esp-sr/,  replace  ${COMPONENT_NAME} with ${COMPONENT_TARGET} or ${COMPONENT_LIB}.

closed 2019-12-03 by feizi