espressif/esp-skainet #11: ESP32 Audio Kit可以跑这个吗

lxdemon, 2020-01-02:
你好

我手上有一块ESP32 Audio Kit V2.2的开发板，请问可以跑这个吗，我看列出来的硬件只有Lyrat和Lyrat Mini，如果可以的话，应该怎么做，是不是要改硬件的配置

feizi, 2020-01-03:
我们暂时没有这块开发板的驱动，如果您有驱动代码，只需修改codec的驱动，保证录音正确就可以使用该工程的example。

lxdemon, 2020-01-06:
Hi,
我现在可以通过耳机听到录音，但是好像不能唤醒，控制台信息：
-----------awaits to be waken up-----------
chunk_num = 200

附上代码片段截图
[image]
图中箭头所指的i2s_write可以让我在耳机中听到环境录音，但是无法唤醒，不知道是否需要对数据做些处理？

硬件：ESP32 Audio Kit V2.2
例程：get_started

lxdemon, 2020-01-06:
重新附上代码片段截图
[image]

lxdemon, 2020-01-06:
图片好像加载不出来，附上代码片段：
```
void recsrcTask(void *arg)
{
    size_t bytes_read;
    int16_t *rsp_in = malloc(AEC_FRAME_BYTES * 2);
    int16_t *aec_rec = malloc(AEC_FRAME_BYTES);
    int16_t *aec_ref = malloc(AEC_FRAME_BYTES);
    int16_t *aec_out = malloc(AEC_FRAME_BYTES);
    void *aec_handle = aec_create(16000, AEC_FRAME_LENGTH_MS, AEC_FILTER_LENGTH);

    while (1) {

        size_t bytes_write = 0;
        i2s_read(I2S_NUM_0, rsp_in, 2 *AEC_FRAME_BYTES, &bytes_read, portMAX_DELAY);
        // vTaskDelay(10 / portTICK_PERIOD_MS);
        i2s_write(I2S_NUM_0, rsp_in, bytes_read, &bytes_write, portMAX_DELAY);

        for (int i = 0; i < AEC_FRAME_BYTES / 2; i++) {
            // According to test results, get better speech recognition performance 
            // when MIC gain is expanded four times for lyrat v4.3.
            aec_out[i] = (rsp_in[2 * i] + rsp_in[2 * i + 1]) * 2 ;
        }

        rb_write(aec_rb, aec_out, AEC_FRAME_BYTES, portMAX_DELAY);
    }
}
```

feizi, 2020-01-07:
1. 检查声音采样率是否为16000Hz，
2. 检查声音幅值是否饱和，如果耳机听上去正常，建议将aec_out[i] = (rsp_in[2 * i] + rsp_in[2 * i + 1]) * 2 修改为 aec_out[i] = rsp_in[2 * i];  和 aec_out[i] = rsp_in[2 * i + 1]; 分别调试，

如果以上均正常，还无法唤醒，建议将录音数据保存下来，发给我们看看可能存在问题。

closed 2020-01-14 by feizi

rxd1999, 2021-10-05:
大哥，你成功了吗？