espressif/esp-skainet #18: `.flash.rodata' will not fit in region `drom0_0_seg'

Shaopus, 2020-04-10:
开发板：[ESP32-LyraT-Mini](https://docs.espressif.com/projects/esp-adf/en/latest/get-started/get-started-esp32-lyrat-mini.html)
程序：使用了[WakeNet](https://github.com/espressif/esp-sr/tree/master/wake_word_engine/README_cn.md)、[MultiNet](https://github.com/espressif/esp-sr/tree/master/speech_command_recognition/README_cn.md)、esp-tts和蓝牙
出现情况：**region `drom0_0_seg' overflowed by 50504 bytes**

因为ESP32-LyraT-Mini中的ESP32-WROVER-B 的FLASH是8M的，这个情况我应该怎么处理？

feizi, 2020-04-10:
现在ESP32支持的app bin大小限制在4MB，超过就会报错， data 分区没有限制。 
按照你的出错log， 应该是超过最大限制 50504 bytes， 你需要想办法减小bin的大小

Shaopus, 2020-04-10:
> 现在ESP32支持的app bin大小限制在4MB，超过就会报错， data 分区没有限制。 
> 
> 按照你的出错log， 应该是超过最大限制 50504 bytes， 你需要想办法减小bin的大小

问题是esp-tts就有2M多，加上wakenet,multinet就占了4M多了，我应该怎么去减少bin的大小呢？

feizi, 2020-04-10:
暂时只能选择参数量少一点的wakenet，比如wakenet5X2（如果你默认使用wakenet5X3），这样应该可以少200KB左右    
后续我们会把esp-tts中的voice set独立出来，放在单独分区

gilimg, 2020-04-12:
hi, please inform ESP 32 DEV KIT Can support Wakenet Multinet

feizi, 2020-04-13:
No microphone on  ESP 32 DEV KIT, so please use ESP32-Lyrat series or ESP32-Korvo series.

gilimg, 2020-04-13:
I am use the external digital microphone and detect the target world We use the ESP32-dev development board and the WakeNet5X2(hilexin) model in our test. The performance is limited because ESP32  only has one microphone. We expect a better recognition performance.
if is possible use 2 deferent Wake World "English "   or "ESTER" AND hilexin.

Shaopus, 2020-04-14:
You could use ESP32-Korvo V1.1,  is a dual-core ESP32-based audio development board with **microphone arrays**.

feizi, 2020-04-14:
For near-field recognition, a single microphone can also achieve good performance. But the microphone configuration must be carefully adjusted, such as gain, signal-to-noise ratio, etc.

feizi, 2020-04-14:
For far-field recognition, please use a microphone array

feizi, 2020-05-18:
> > 现在ESP32支持的app bin大小限制在4MB，超过就会报错， data 分区没有限制。
> > 按照你的出错log， 应该是超过最大限制 50504 bytes， 你需要想办法减小bin的大小
> 
> 问题是esp-tts就有2M多，加上wakenet,multinet就占了4M多了，我应该怎么去减少bin的大小呢？

Now，esp tts chinese has used separate voice data partition. Please refer to the example [chinese_tts](https://github.com/espressif/esp-skainet/tree/master/examples/chinese_tts)

closed 2020-05-18 by feizi