espressif/esp-skainet #19: FPC Connector cable question for Korvo

bluelu, 2020-05-24:
Hi,

Could you add more details in (https://github.com/espressif/esp-skainet/blob/master/docs/en/hw-reference/esp32/user-guide-esp32-korvo-v1.1.md) on how the FPC cable is to be connected into the FPC connector.

Which way is the cable connected? Are the contacts of the cable downwards to the board or upwards? (I assume the same on both boards)

Thanks

ZhaoYang7, 2020-05-25:
Hi @bluelu, for both boards, please ensure the contacts are downwards to the board :)

bluelu, 2020-05-25:
Thank you!

Can you also add which driver is needed for the usb connection? Your main page is still empty on this topic:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/establish-serial-connection.html

I assume I can also connect two USB micro from different power supplies (e.g. one external and one from the pc without the risk of frying the board?)

bluelu, 2020-05-25:
If aynone is wondering, the board requires the CP210x driver and the flash process also works without pushing any buttons. Closing.

closed 2020-05-25 by bluelu