espressif/esp-skainet #20: Esp32-Korvo. Wrong sampling rate?

bluelu, 2020-05-29:
Hi,

I'm streaming audio data over from the esp32-korvo to a server. All the examples returned very bad sound quality (could not hear my voice), so I reduced it to the absolute minimum to test (based on the getting started example from: https://github.com/espressif/esp-skainet/blob/master/examples/get_started/main/recsrc.c)

I only keep the code which reads form the i2s bus, do some basic transformation and then send it to a server which writes out the wave file ( AudioFormat format = new AudioFormat(16000, 16, 1, false, false); )

Raw audio output:
`
  {
        size_t bytes_read;
        int16_t *rsp_in = (int16_t *)malloc(AEC_FRAME_BYTES * I2S_CHANNEL_NUM);
        int16_t *aec_ref = (int16_t *)malloc(AEC_FRAME_BYTES);

        int nch = 1;
        int16_t *aec_rec = (int16_t *)malloc(AEC_FRAME_BYTES * nch);
        int16_t *aec_out = (int16_t *)malloc(AEC_FRAME_BYTES * nch);
        void *aec_handle = aec_create(16000, AEC_FRAME_LENGTH_MS, AEC_FILTER_LENGTH);

        while (1)
        {

            i2s_read(I2S_NUM_1, rsp_in, 2 * AEC_FRAME_BYTES, &bytes_read, portMAX_DELAY);
            for (int i = 0; i < AEC_FRAME_BYTES / 2; i++)
            {
                aec_out[i] = (rsp_in[2 * i] + rsp_in[2 * i + 1]) / 2;
            }
            rb_write(rec_rb, (uint8_t *)aec_out, AEC_FRAME_BYTES, portMAX_DELAY);
        }
    }
`

The audio is fine, however it is only half of normal speed. If I change the sample rate, to 32000 in the AudioFormat constructor, the sound is very good (however no postprocessing for noise, etc has yet been applied).

As the constructor of AEC specifically mentions that the sampling rate must be 16000 and not 32000, I tried changing the sample rate in MediaHal.c 
`i2s_config.bits_per_sample = 32; // for ES7210`
to 
`i2s_config.bits_per_sample = 16; // for ES7210`

For the raw audio output, I can now create the wave file with a frame size of 16.000, and the data output is cristal clear.

But none of the postprocessing filters of your example seem to work.

I now tried to apply the code to filter with both 2 and 3 mics (I have the 3 mic version), but the output gets very disturbed (e.g).
`
  {

        size_t bytes_read;
        int16_t *rsp_in = (int16_t *)malloc(AEC_FRAME_BYTES * I2S_CHANNEL_NUM);
        int16_t *aec_ref = (int16_t *)malloc(AEC_FRAME_BYTES);

        int nch = 3;
    int16_t *aec_rec = (int16_t *) malloc(AEC_FRAME_BYTES * nch);
    int16_t *aec_out = (int16_t *) malloc(AEC_FRAME_BYTES * nch);
    void *aec_handle = aec_create_multimic(16000, AEC_FRAME_LENGTH_MS, AEC_FILTER_LENGTH, nch);
        while (1)
        {
        
           i2s_read(I2S_NUM_1, rsp_in, 4 *AEC_FRAME_BYTES, &bytes_read, portMAX_DELAY);
        for (int i = 0; i < AEC_FRAME_BYTES / 2; i++) {
            aec_ref[i] = rsp_in[4 * i + 2];
            aec_rec[i] = rsp_in[4 * i + 1];
            aec_rec[i + AEC_FRAME_BYTES / 2] = rsp_in[4 * i + 3];
            if (nch == 3)
            {
                aec_rec[i + AEC_FRAME_BYTES] = rsp_in[4 * i];
            }
        }
        aec_process(aec_handle, aec_rec, aec_ref, aec_out);
        rb_write(rec_rb,  (uint8_t *) aec_out, AEC_FRAME_BYTES * nch, portMAX_DELAY);
        }
    }
`

How do I make the examples work?

Thanks

feizi, 2020-05-29:
Hi，
The I2S channel of ESP-Korvo board is 4. so you can not read I2S data as follows:
```
aec_out[i] = (rsp_in[2 * i] + rsp_in[2 * i + 1]) / 2;
```

You need to read it every four sampling points
```
i2s_read(I2S_NUM_1, rsp_in, 4 *AEC_FRAME_BYTES, &bytes_read, portMAX_DELAY);
aec_out[i] = (rsp_in[4 * i] + rsp_in[4 * i + 1]) / 2;
```

Please note the information of each channel data:
channel0: mircophone data, rsp_in[4 * i] 
channel1: mircophone data, rsp_in[4 * i+1] 
**channel2: reference data for AEC from I2S writer**     , rsp_in[4 * i+2] 
channel3: mircophone data , rsp_in[4 * i+3]

bluelu, 2020-05-29:
Thank you, that explains well why still the 32000 worked at the end.

I now apply the aec filter exactly like in the example (and written above), the the output is distorted while I still hear myself a little bit in the background (https://transfer.sh/%28/WjruG/test-199.wav%29.zip)? 

Hardware wise, I have not yet connected a speaker to lineout.
 
```
   {

        size_t bytes_read;
        int16_t *rsp_in = (int16_t *)malloc(AEC_FRAME_BYTES * I2S_CHANNEL_NUM);
        int16_t *aec_ref = (int16_t *)malloc(AEC_FRAME_BYTES);

        int nch = 3;
        int16_t *aec_rec = (int16_t *)malloc(AEC_FRAME_BYTES * nch);
        int16_t *aec_out = (int16_t *)malloc(AEC_FRAME_BYTES * nch);
        void *aec_handle = aec_create_multimic(16000, AEC_FRAME_LENGTH_MS, AEC_FILTER_LENGTH, nch);
        
        while (1)
        {

            i2s_read(I2S_NUM_1, rsp_in, 4 * AEC_FRAME_BYTES, &bytes_read, portMAX_DELAY);
            for (int i = 0; i < AEC_FRAME_BYTES / 2; i++)
            {
                aec_ref[i] = rsp_in[4 * i + 2];
                aec_rec[i] = rsp_in[4 * i + 1];
                aec_rec[i + AEC_FRAME_BYTES / 2] = rsp_in[4 * i + 3];
                if (nch == 3)
                {
                    aec_rec[i + AEC_FRAME_BYTES] = rsp_in[4 * i];
                }
            }
            aec_process(aec_handle, aec_rec, aec_ref, aec_out);
            rb_write(rec_rb, (uint8_t *)aec_out, AEC_FRAME_BYTES * nch, portMAX_DELAY);
        }
    }

```

feizi, 2020-06-01:
AEC test need a speaker， otherwise the reference signal and the input signal do not match.   
And the distance between speaker and microphone should be greater than 10cm.

closed 2020-06-22 by feizi

trupal112, 2023-06-30:
how can I increase the volume of the ESP32 Korvo board ? 

I tried many things but volume is not increase what should I do Is there any change in code 

code Link : - https://github.com/espressif/esp-skainet/tree/master/examples/en_speech_commands_recognition

ZorkyZorky, 2023-08-15:
I want to enable ace in esp-box.But now i even cannot wake up Alexsa.I have tried the following operation  in esp-box:

1.ret |= es7210_write_reg(ES7210_SDP_INTERFACE2_REG12, 0x02);    //in esp7210.c
2. .bits_per_sample        = I2S_BITS_PER_SAMPLE_32BIT,    //in bsp_i2s.c->I2S_CONFIG_DEFAULT()
3. .aec_init =true
4.i2s_read(I2S_NUM_0, feed_buffer, chunksize * 4 * sizeof(int16_t),&bytes_read, portMAX_DELAY); 
5.    for (int i = 0; i < chunksize; i++) {
        int16_t ref = feed_buffer[4 * i + 0];
        feed_buffer[3 * i + 0] = feed_buffer[4 * i + 1];
        feed_buffer[3 * i + 1] = feed_buffer[4 * i + 3];
        feed_buffer[3 * i + 2] = ref;
    }  //channel adjust

anything else should i do ？Thank you.