espressif/esp-skainet #22: mase issue

andrej102, 2020-07-01:
Hellow. 
1. I found an error in the MASE algorithm. When MASE amplifies the signal, the INT16 value overflow is ignored and significant signal distortion is obtained. The signal should not overflow when it reaches the upper and lower limits of the value but should simply be cut off. Please fix this

[image]

2. Also, the filter_strength parameter does not affect the result. Why do I get the same results when I select 0, 1,2,3,4,5,6?

gilimg, 2020-07-04:
Hi,

 

We are use wakeup HI LEXI

When I play music from SD, or external noise, wakeup word detection 30%,

1.	How can create a wakeup word ready pay for customize Offline wakeup word "ESTER".
2.	That requires minimum resources ESP32.
3.	Detect wakeup word overcomes external noise and when playing music high quality.

Thanks for cooperation

Moshe Gili

VP Business Development

  <https://mail.google.com/mail/e/1f4f1>     <tel:+97235757775> + <tel:+972544211611> 972 54-421-1611

  <https://mail.google.com/mail/e/260e>      <tel:+97235757775> + <tel:+97236347050> 1-516-593-6241

     <mailto:gilimg@gmail.com> gilimg@gmail.com

     <mailto:gilimg@hirisetechnology.com> gilimg@hirisetechnology.com

     P.0.BOX 6874,Hamagshimim 20,Petach-Tikva,IL

   <http://www.hirisetechnology.com/> www.hirisetechnology.com

 

 

 

 

 

 

 

 

 

 

 

I found an error in the MASE algorithm. When MASE amplifies the signal, the INT16 value overflow is ignored and significant signal distortion is obtained. The signal should not overflow when it reaches the upper and lower limits of the value but should simply be cut off. Please fix this

 <https://user-images.githubusercontent.com/42580297/86257048-b18b4c80-bbc1-11ea-8961-7a2cd5ce2d4d.png> 

2.	Also, the filter_strength parameter does not affect the result. Why do I get the same results when I select 0, 1,2,3,4,5,6?

ZhaoYang7, 2020-07-08:
Hi @andrej102 , 
1. Will fix the overwhelming issue soon.
2. If you're using NORMAL_MODE, the filter_strength parameter does not influence the outcome; if you're using WAKE_UP_MODE, with higher filter_strength, less RAM will be consumed, while de-noising performance drops

ZhaoYang7, 2020-07-08:
Hi @gilimg 
Thanks for your interest. For business contact, please refer to [Espressif Speech Wake Word Customization Process](https://github.com/espressif/esp-sr/blob/master/wake_word_engine/ESP_Wake_Words_Customization.md).

ZhaoYang7, 2020-07-08:
Hi @andrej102  ,
1. Will fix the overflow issue soon.
2. If you're using NORMAL_MODE, the filter_strength parameter does not influence the outcome; if you're using WAKE_UP_MODE, with higher filter_strength, less RAM will be consumed, while de-noising performance drops

andrej102, 2020-07-15:
> Hi @andrej102 ,
> 
> 1. Will fix the overflow issue soon.
> 2. If you're using NORMAL_MODE, the filter_strength parameter does not influence the outcome; if you're using WAKE_UP_MODE, with higher filter_strength, less RAM will be consumed, while de-noising performance drops

Thank you for your interest in my questions. 
In my project, fixing the overflow of numbers is very important, I look forward to it. 

I also have another question: 
the description of SKAINET mentions the presence of an audio beamformig, but it looks like it is the MASE algorithm. In many cases (I conducted real tests), the beamformig technology gives a significantly greater improvement in useful speech against the background of noise (an improvement of not 10-20% as in MASE, but several times), and consequently better recognition of useful speech. 

Question:
do you plan to supplement the library of sound processing function audio beamforming?

closed 2021-11-12 by feizi