espressif/esp-skainet #32: Cannot compile any examples under ESP-IDF 4.3

expresspotato, 2021-07-15:
There are several fixes and enhancements in 4.3, but none of the examples here compile with it.

```
Merged 1 ELF section
Generated /home/kevin/esp-skainet/examples/wake_word_detection/build/bootloader/bootloader.bin
[976/1001] Building C object esp-idf/hardware_driver/CMakeFiles/__idf_hardware_driver.dir/MediaHal/MediaHal.c.obj
FAILED: esp-idf/hardware_driver/CMakeFiles/__idf_hardware_driver.dir/MediaHal/MediaHal.c.obj 
/home/kevin/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I/home/kevin/esp-skainet/components/hardware_driver/MediaHal/Board -I/home/kevin/esp-skainet/components/hardware_driver/MediaHal/Codec -I/home/kevin/esp-skainet/components/hardware_driver/MediaHal -I/home/kevin/esp-skainet/components/hardware_driver/SDCardConfig -I/home/kevin/esp-skainet/components/hardware_driver/SystemSal -I/home/kevin/esp-skainet/components/hardware_driver/i2c_bus -I/home/kevin/esp-skainet/components/hardware_driver/userconfig -I/home/kevin/esp-skainet/components/hardware_driver/button/include -I/home/kevin/esp-skainet/components/hardware_driver/led/include -I/home/kevin/esp-idf/components/newlib/platform_include -I/home/kevin/esp-idf/components/freertos/include -I/home/kevin/esp-idf/components/freertos/port/xtensa/include -I/home/kevin/esp-idf/components/esp_hw_support/include -I/home/kevin/esp-idf/components/esp_hw_support/port/esp32/. -I/home/kevin/esp-idf/components/heap/include -I/home/kevin/esp-idf/components/log/include -I/home/kevin/esp-idf/components/lwip/include/apps -I/home/kevin/esp-idf/components/lwip/include/apps/sntp -I/home/kevin/esp-idf/components/lwip/lwip/src/include -I/home/kevin/esp-idf/components/lwip/port/esp32/include -I/home/kevin/esp-idf/components/lwip/port/esp32/include/arch -I/home/kevin/esp-idf/components/soc/include -I/home/kevin/esp-idf/components/soc/esp32/. -I/home/kevin/esp-idf/components/soc/esp32/include -I/home/kevin/esp-idf/components/hal/esp32/include -I/home/kevin/esp-idf/components/hal/include -I/home/kevin/esp-idf/components/esp_rom/include -I/home/kevin/esp-idf/components/esp_rom/esp32 -I/home/kevin/esp-idf/components/esp_rom/include/esp32 -I/home/kevin/esp-idf/components/esp_common/include -I/home/kevin/esp-idf/components/esp_system/include -I/home/kevin/esp-idf/components/esp32/include -I/home/kevin/esp-idf/components/driver/include -I/home/kevin/esp-idf/components/driver/esp32/include -I/home/kevin/esp-idf/components/esp_ringbuf/include -I/home/kevin/esp-idf/components/efuse/include -I/home/kevin/esp-idf/components/efuse/esp32/include -I/home/kevin/esp-idf/components/xtensa/include -I/home/kevin/esp-idf/components/xtensa/esp32/include -I/home/kevin/esp-idf/components/espcoredump/include -I/home/kevin/esp-idf/components/esp_timer/include -I/home/kevin/esp-idf/components/esp_ipc/include -I/home/kevin/esp-idf/components/esp_pm/include -I/home/kevin/esp-idf/components/vfs/include -I/home/kevin/esp-idf/components/esp_wifi/include -I/home/kevin/esp-idf/components/esp_wifi/esp32/include -I/home/kevin/esp-idf/components/esp_event/include -I/home/kevin/esp-idf/components/esp_netif/include -I/home/kevin/esp-idf/components/esp_eth/include -I/home/kevin/esp-idf/components/tcpip_adapter/include -I/home/kevin/esp-idf/components/app_trace/include -I/home/kevin/esp-idf/components/fatfs/diskio -I/home/kevin/esp-idf/components/fatfs/vfs -I/home/kevin/esp-idf/components/fatfs/src -I/home/kevin/esp-idf/components/wear_levelling/include -I/home/kevin/esp-idf/components/spi_flash/include -I/home/kevin/esp-idf/components/sdmmc/include -I/home/kevin/esp-idf/components/esp_adc_cal/include -I/home/kevin/esp-idf/components/nvs_flash/include -I/home/kevin/esp-idf/components/mbedtls/port/include -I/home/kevin/esp-idf/components/mbedtls/mbedtls/include -I/home/kevin/esp-idf/components/mbedtls/esp_crt_bundle/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw -Os -freorder-blocks -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.3-194-g5b5e46971-dirty\" -DESP_PLATFORM -MD -MT esp-idf/hardware_driver/CMakeFiles/__idf_hardware_driver.dir/MediaHal/MediaHal.c.obj -MF esp-idf/hardware_driver/CMakeFiles/__idf_hardware_driver.dir/MediaHal/MediaHal.c.obj.d -o esp-idf/hardware_driver/CMakeFiles/__idf_hardware_driver.dir/MediaHal/MediaHal.c.obj   -c /home/kevin/esp-skainet/components/hardware_driver/MediaHal/MediaHal.c
In file included from /home/kevin/esp-idf/components/driver/include/driver/i2c.h:25,
                 from /home/kevin/esp-skainet/components/hardware_driver/MediaHal/Codec/ES8388_interface.h:30,
                 from /home/kevin/esp-skainet/components/hardware_driver/MediaHal/MediaHal.c:27:
/home/kevin/esp-idf/components/freertos/include/freertos/FreeRTOS.h:963:27: error: conflicting types for 'SemaphoreHandle_t'
  #define xSemaphoreHandle SemaphoreHandle_t
                           ^~~~~~~~~~~~~~~~~
/home/kevin/esp-skainet/components/hardware_driver/SystemSal/lock.h:7:16: note: in expansion of macro 'xSemaphoreHandle'
 typedef void * xSemaphoreHandle;
                ^~~~~~~~~~~~~~~~
In file included from /home/kevin/esp-idf/components/driver/include/driver/i2c.h:26,
                 from /home/kevin/esp-skainet/components/hardware_driver/MediaHal/Codec/ES8388_interface.h:30,
                 from /home/kevin/esp-skainet/components/hardware_driver/MediaHal/MediaHal.c:27:
/home/kevin/esp-idf/components/freertos/include/freertos/semphr.h:37:23: note: previous declaration of 'SemaphoreHandle_t' was here
 typedef QueueHandle_t SemaphoreHandle_t;
                       ^~~~~~~~~~~~~~~~~
/home/kevin/esp-skainet/components/hardware_driver/MediaHal/MediaHal.c:75:5: warning: 'I2S_COMM_FORMAT_I2S' is deprecated [-Wdeprecated-declarations]
     .communication_format = I2S_COMM_FORMAT_I2S,
     ^
In file included from /home/kevin/esp-idf/components/hal/esp32/include/hal/i2s_ll.h:30,
                 from /home/kevin/esp-idf/components/hal/include/hal/i2s_hal.h:28,
                 from /home/kevin/esp-idf/components/driver/include/driver/i2s.h:24,
                 from /home/kevin/esp-skainet/components/hardware_driver/MediaHal/MediaHal.c:35:
/home/kevin/esp-idf/components/hal/include/hal/i2s_types.h:70:5: note: declared here
     I2S_COMM_FORMAT_I2S       __attribute__((deprecated)) = 0x01, /*!< I2S communication format I2S, correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~
/home/kevin/esp-skainet/components/hardware_driver/MediaHal/MediaHal.c:178:19: warning: initialization of 'int (*)(void *)' from incompatible pointer type 'int (*)(Es8311Config *)' {aka 'int (*)(struct <anonymous> *)'} [-Wincompatible-pointer-types]
     .codec_init = Es8311Init,
                   ^~~~~~~~~~
/home/kevin/esp-skainet/components/hardware_driver/MediaHal/MediaHal.c:178:19: note: (near initialization for 'MediaHalConfig.codec_init')
In file included from /home/kevin/esp-skainet/components/hardware_driver/MediaHal/MediaHal.c:31:
/home/kevin/esp-skainet/components/hardware_driver/MediaHal/Codec/es7210.h:184:12: warning: 'es7210_mute' declared 'static' but never defined [-Wunused-function]
 static int es7210_mute(unsigned char mute);
            ^~~~~~~~~~~
[981/1001] Building C object esp-idf/spiffs/CMakeFiles/__idf_spiffs.dir/spiffs/src/spiffs_nucleus.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
kevin@ubuntu:~/esp-skainet/examples/wake_word_detection$ 

```

expresspotato, 2021-07-15:
Commenting out:
`typedef void * xSemaphoreHandle;` in: `esp-skainet/components/hardware_driver/SystemSal/lock.h:7`

Worked...

feizi, 2021-07-20:
We will support ESP-IDF v4.4 and ESP32S3 chip.

closed 2021-11-23 by feizi