espressif/esp-skainet #37: speech_commands_recognition  demo的一些问题

Wang050198, 2021-11-25:
1. 你好，我使用的环境是cn_speech_commands_recognition（master） demo 程序，ESP32-S3-Korvo-v2.0开发板，IDF4.4版本，有以下问题望解答

2. 在使用sdkconfig_esp32s3r8_8+4.defaults和修改开发板设置CONFIG_ESP32_S3_KORVO_V2_0_BOARD后，编译会报错，查看了一下发现  I2S0_CONFIG()里面宏定义结构体里面成员变量和有差异，修改后可以编译通过（同s3开发板宏定义结构体都一样，只有这个与实际i2s_config_t 不同，希望可以告知原因）

```
define I2S0_CONFIG(){ \
        .param_cfg = { \
            .mode = I2S_MODE_MASTER | I2S_MODE_TX, \
            .sample_rate = 16000, \
            .communication_format = I2S_COMM_FORMAT_STAND_I2S, \
            .slot_bits_cfg = (I2S_BITS_PER_SLOT_16BIT << SLOT_BIT_SHIFT) | I2S_BITS_PER_SAMPLE_16BIT, \
            .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT, \
            .slot_channel_cfg = (2 << SLOT_CH_SHIFT) | 2, \
            .active_slot_mask = I2S_TDM_ACTIVE_CH0 | I2S_TDM_ACTIVE_CH1, \
            .left_align_en = false, \
            .big_edin_en = false, \
            .bit_order_msb_en = false, \
        }, \
        .dma_buf_count = 6, \
        .dma_buf_len = 160, \
        .use_apll = false, \
        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1 \
}
```

3.  另一个问题是在编译通过后会有下载程序到开发板，使用 idf.py  -p COMx monitor  查看端口输出会出现一下问题

```
C:\esp-skainet\esp-idf\tools\idf.py' '-p' 'COM3'"...
[0;33m--- WARNING: GDB cannot open serial ports accessed as COMx[0m
[0;33m--- Using \\.\COM3 instead...[0m
[0;33m--- idf_monitor on \\.\COM3 115200 ---[0m
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
 header: 0xffffffff
inESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
```

Wang050198, 2021-11-26:
更：我修改了自定义内存分区和对应文件内容（partitions.csv）
```
# Espressif ESP32 Partition Table
# Name,  Type, SubType, Offset,  Size
factory, app,  factory, 0x010000, 2048k
model,  data, spiffs,         , 1048K,
```
和sdkconfig文件中的flash配置大小从16M到8M
`CONFIG_ESPTOOLPY_FLASHSIZE="8MB"`

就可以正常运行到app_main()中了，但是有一个新的问题，main函数在
esp_afe_sr_data_t *afe_data = afe_handle->create_from_config(&afe_config); 
位置出现内存错误。
```
void app_main()
{
    sd_card_mount("/sdcard");
    codec_init();
#if defined CONFIG_ESP_LYRAT_MINI_V1_1_BOARD || defined CONFIG_ESP32_KORVO_V1_1_BOARD
    led_init();
#endif

#if CONFIG_IDF_TARGET_ESP32
    afe_handle = &esp_afe_sr_1mic;
#else
    afe_handle = &esp_afe_sr_2mic;
#endif

    afe_config_t afe_config = AFE_CONFIG_DEFAULT();
#if defined CONFIG_ESP32_S3_BOX_BOARD
    afe_config.aec_init = false;
    afe_config.vad_init = false;
#endif
    esp_afe_sr_data_t *afe_data = afe_handle->create_from_config(&afe_config);      //      运行到这里就会出现内存错误

    xTaskCreatePinnedToCore(&feed_Task, "feed", 4 * 1024, (void*)afe_data, 5, NULL, 0);
    xTaskCreatePinnedToCore(&detect_Task, "detect", 4 * 1024, (void*)afe_data, 5, NULL, 1);
#ifndef  CONFIG_ESP32_S3_BOX_BOARD
    xTaskCreatePinnedToCore(&play_music, "play", 2 * 1024, NULL, 5, NULL, 1);
    xTaskCreatePinnedToCore(&led_Task, "led", 2 * 1024, NULL, 5, NULL, 0);
#endif
}
```

以下为错误log信息

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759bc
0x403759bc: esp_restart_noos at C:/esp-skainet/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:143 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1718
load:0x403b6000,len:0x8c8
load:0x403ba000,len:0x2eb4
entry 0x403b61c0
I (29) boot: ESP-IDF v4.4-dev-3703-gddc44956bf-dirty 2nd stage bootloader
I (29) boot: compile time 17:20:04
I (29) boot: chip revision: 0
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (47) boot: Enabling RNG early entropy source...
W (53) bootloader_random: RNG for ESP32-S3 not currently supported
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 factory          factory app      00 00 00010000 00200000
[... 74 lines cut ...]
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4200ae95  PS      : 0x00060830  A0      : 0x82009e4d  A1      : 0x3fcf46e0
0x4200ae95: esp_aec3_hps16fft_init at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc/build/../components/esp_audio_processor/acoustic_echo_cancellation/esp_aec3_hps16fft.c:113 (discriminator 3)

A2      : 0x00000400  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x3fce24d4  A8      : 0x8200ae71  A9      : 0x3fcf46d0
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x38000000  A13     : 0x00000380
A14     : 0x00000381  A15     : 0x3fe00000  SAR     : 0x00000004  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0xffffffff

Backtrace:0x4200ae92:0x3fcf46e00x42009e4a:0x3fcf4710 0x42007674:0x3fcf4740 0x4206c901:0x3fcf47a0 0x403842b9:0x3fcf47c0
0x4200ae92: esp_aec3_hps16fft_init at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc/build/../components/esp_audio_processor/acoustic_echo_cancellation/esp_aec3_hps16fft.c:113 (discriminator 3)

0x42009e4a: afe_create_from_config at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc/build/../components/esp_audio_front_end/esp_afe_sr_2mic.c:117

0x42007674: app_main at c:\esp-skainet\examples\cn_speech_commands_recognition\build/../main/main.c:183

0x4206c901: main_task at C:/esp-skainet/esp-idf/components/freertos/port/port_common.c:129 (discriminator 2)

0x403842b9: vPortTaskWrapper at C:/esp-skainet/esp-idf/components/freertos/port/xtensa/port.c:130
```

希望我的问题可以得到回复，谢谢！

feizi, 2021-11-26:
speech_commands_recognition 的demo需要的model分区大约需要5M（根据选择的模型不同，所需大小也不同）。 
我看你修改后的model 分区只有1M,可定是不够的，你可以按照下面修改一下partition table试试
```
factory, app,  factory, 0x010000, 2M
model,  data, spiffs,         , 6M
```

Wang050198, 2021-11-26:
@feizi **谢谢回复，但是我修改了Model分区为，**
```
factory, app,  factory, 0x010000, 2M
model,  data, spiffs,         , 6M
```

上面的问题依然存在, log如下；
```
I (569) gpio: GPIO[38]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
codec init done!!!!
I (569) DRV8311: SET: volume:165

Item psram alloc failed. Size: 2068 = 1024 x 2 + 16 + 4
Item psram alloc failed. Size: 8212 = 2048 x 4 + 16 + 4
Item psram alloc failed. Size: 4120 = 1025 x 4 + 16 + 4
Item psram alloc failed. Size: 8212 = 2048 x 4 + 16 + 4
Item psram alloc failed. Size: 8212 = 2048 x 4 + 16 + 4
Item psram alloc failed. Size: 4116 = 1024 x 4 + 16 + 4
Item psram alloc failed. Size: 4116 = 1024 x 4 + 16 + 4
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4200ac49  PS      : 0x00060830  A0      : 0x8200997a  A1      : 0x3fcf46e0
0x4200ac49: esp_aec3_728_init at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc/build/../components/esp_audio_processor/acoustic_echo_cancellation/esp_aec3_728.c:100 (discriminator 3)

A2      : 0x3fce1298  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000001
A6      : 0x00000400  A7      : 0x00000000  A8      : 0x40c90fda  A9      : 0x3fcf46d0
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x38000000  A13     : 0x00000380
A14     : 0x00000381  A15     : 0x3fe00000  SAR     : 0x00000004  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0xffffffff

Backtrace:0x4200ac46:0x3fcf46e00x42009977:0x3fcf4710 0x42007485:0x3fcf4740 0x42063609:0x3fcf47a0 0x40383d41:0x3fcf47c0
0x4200ac46: esp_aec3_728_init at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc/build/../components/esp_audio_processor/acoustic_echo_cancellation/esp_aec3_728.c:101 (discriminator 3)

0x42009977: afe_create_from_config at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc/build/../components/esp_audio_front_end/esp_afe_sr_1mic.c:100

0x42007485: app_main at c:\esp-skainet\examples\cn_speech_commands_recognition\build/../main/main.c:183

0x42063609: main_task at C:/esp-skainet/esp-idf/components/freertos/port/port_common.c:129 (discriminator 2)

0x40383d41: vPortTaskWrapper at C:/esp-skainet/esp-idf/components/freertos/port/xtensa/port.c:130
```

感觉像是其他问题导致的，**Item psram alloc failed.** 这个错误是怎样产生的呢，我在自己malloc可以正常申请空间和使用，但是它却会报这样的错误。

`0x4200ac49: esp_aec3_728_init at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc/build/../components/esp_audio_processor/acoustic_echo_cancellation/esp_aec3_728.c:100 (discriminator 3)`
会不会是上面的错误提示中的esp_audio_processor库导致的呢？

feizi, 2021-11-26:
这个问题是系统的PSRAM不够了，你现在的PSRAM是8M吗

Wang050198, 2021-11-27:
@feizi 感谢回复！
我看到ESP32_S3_KORVO_V2_0的模组的内存是8 MB Flash + 8 MB PSRAM，工程中是有地方需要配置PSAM吗，抱歉没有找到对应的定义

Wang050198, 2021-11-27:
@feizi 确实我发现我的开发板的esp32s3是esp32s3r2，没有集成PSRAM，我更换了esp32s3R8后，Item psram alloc failed错误得到了解决，但是工程还是没有正常运行起来，
```
I (377) cpu_start: App cpu up.
I (687) spiram: SPI SRAM memory test OK
I (696) cpu_start: Pro cpu start user code
I (696) cpu_start: cpu freq: 240000000
I (696) cpu_start: Application information:
I (699) cpu_start: Project name:     speech_commands_recognition
I (705) cpu_start: App version:      v0.3.0-69-g4911fbe-dirty
I (712) cpu_start: Compile time:     Nov 27 2021 17:56:50
I (718) cpu_start: ELF file SHA256:  201241bff4f35095...
I (724) cpu_start: ESP-IDF:          v4.4-dev-3703-gddc44956bf-dirty
I (731) heap_init: Initializing. RAM available for dynamic allocation:
I (738) heap_init: At 3FC9F9F8 len 00040608 (257 KiB): D/IRAM
I (745) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (751) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (760) sleep: Configure to isolate all GPIO pins in sleep state
I (766) sleep: Enable automatic switching of GPIO sleep configuration
I (773) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (794) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
Initializing SPIFFS
Partition size: total: 5515976, used: 0
I (1144) MEDIA_HAL: CONFIG_CODEC_CHIP_IS_ES8311
W (1144) I2S: APLL not supported on current chip, use I2S_CLK_D2CLK as default clock source
I (1144) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (1154) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (1164) I2S: I2S0, MCLK output by GPIO0
I (1164) I2S: DMA Malloc info, datalen=blocksize=2400, dma_buf_count=3
I (1174) I2S: DMA Malloc info, datalen=blocksize=2400, dma_buf_count=3
I (1184) I2S: I2S1, MCLK output by GPIO0
I (1184) DRV8311: Enter into es8311_Codec_Startup()

I (1194) DRV8311: ES8311 in Slave mode

I (1194) DRV8311: Enter into es8311_pcm_hw_params()

I (1214) DRV8311: Exit es8311_Codec_Startup()

I (1214) DRV8311: ES8311 in I2S Format
I (1214) DRV8311: SET: volume:153

I (1214) MEDIA_HAL: I2S_OUT_VOL_DEFAULT[60]
WYB:Slave mode:80 !
WYB:ES7210_MCLK_CTL_REG02 = 0xC3!
I (1234) gpio: GPIO[38]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
codec init done!!!!
I (1234) DRV8311: SET: volume:165

Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.

Core  0 register dump:
PC      : 0x4c4c554e  PS      : 0x00060630  A0      : 0x8200763f  A1      : 0x3fca6410
A2      : 0x3fca6450  A3      : 0x3fce1c78  A4      : 0x0000002c  A5      : 0x3fca647c
A6      : 0x00000002  A7      : 0x00000002  A8      : 0x820082f4  A9      : 0x3fca63f0
A10     : 0x3c0f1d10  A11     : 0x00000002  A12     : 0x00000002  A13     : 0x3a83126f
A14     : 0x00002000  A15     : 0x3fcad680  SAR     : 0x00000010  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x40056fc5  LEND    : 0x40056fe7  LCOUNT  : 0xffffffff
```

Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.要怎么解决呢？

WeldonWangwang, 2021-11-29:
Hi @Wang050198 , 从你最新的 log 使用应该是 codec/ADC 驱动报错，请问你使用的硬件是 ESP32-S3-Korvo-v2.0 开发板吗？ esp-skainet 目前不支持这块开发板（codec + ADC），所以你如果要使用这块硬件的话， codec 和 ADC 的驱动需要自己集成一下。

Wang050198, 2021-12-03:
@feizi @WeldonWangwang 早上好，再请教一个问题，我的例程在编译后生成的model.bin文件，我发现model.bin文件生成了，但是模型文件里面是空的全是0xff，请问有可能是什么情况导致的呢

feizi, 2021-12-14:
ESP32-S3-Korvo-2 和 ESP32-S3-Korvo-1是两块不同的开发板，而不是一个开发板的不同版本。
现在esp-skainet同时支持这两款开发板了

closed 2022-08-23 by feizi