espressif/esp-skainet #38: ESP-BOX ISSUE

FRANCISEG, 2021-12-23:
I (659) MEDIA_HAL: CONFIG_CODEC_CHIP_IS_ES8311
I (659) I2S: DMA Malloc info, datalen=blocksize=1280, dma_buf_count=6
I (669) I2S: DMA Malloc info, datalen=blocksize=1280, dma_buf_count=6
I (669) I2S: I2S1, MCLK output by GPIO0
I (679) DRV8311: Enter into es8311_Codec_Startup()

I (689) DRV8311: ES8311 in Slave mode

I (689) DRV8311: Enter into es8311_pcm_hw_params()

I (709) DRV8311: Exit es8311_Codec_Startup()

I (709) DRV8311: ES8311 in I2S Format
I (709) DRV8311: SET: volume:153

I (709) MEDIA_HAL: I2S_OUT_VOL_DEFAULT[60]
WYB:Slave mode:80 !
WYB:ES7210_MCLK_CTL_REG02 = 0xC3!
I (729) gpio: GPIO[46]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
codec init done!!!!
I (729) DRV8311: SET: volume:165

ESP Chinese TTS v1.1 (Nov 12 2021 15:44:10)
欢迎使用乐鑫语音合成
I (739) tts_parser: unicode:0x6b22 -> huan1
I (749) tts_parser: unicode:0x8fce -> ying2
I (749) tts_parser: unicode:0x4f7f -> shi3
I (759) tts_parser: unicode:0x7528 -> yong4
I (759) tts_parser: unicode:0x4e50 -> le4
I (759) tts_parser: unicode:0x946b -> xin1
I (769) tts_parser: unicode:0x8bed -> yu3
I (769) tts_parser: unicode:0x97f3 -> yin1
I (779) tts_parser: unicode:0x5408 -> he2
I (779) tts_parser: unicode:0x6210 -> cheng2
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42005f12  PS      : 0x00060c30  A0      : 0x82006258  A1      : 0x3fcf3f70  
0x42005f12: esp_tts_stream_amr at /home/sunxiangyu/aishare/aigit2/esp_tts/build/../components/speech_synthesis/esp_tts_chinese/esp_tts.c:425

A2      : 0x00000002  A3      : 0x3c03bff0  A4      : 0x6e463b32  A5      : 0x3fce0c94  
A6      : 0x3fcf3fc0  A7      : 0xb918ecc8  A8      : 0xf54970a4  A9      : 0x3fcf3f50  
A10     : 0x00000001  A11     : 0x3c06503c  A12     : 0x3fce0cdc  A13     : 0x0000003c  
A14     : 0x00000003  A15     : 0xff000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xf54970a4  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace:0x42005f0f:0x3fcf3f700x42006255:0x3fcf3f90 0x420057d7:0x3fcf3fb0 0x42026fbc:0x3fcf43f0 
0x42005f0f: esp_tts_stream_amr at /home/sunxiangyu/aishare/aigit2/esp_tts/build/../components/speech_synthesis/esp_tts_chinese/esp_tts.c:425

0x42006255: esp_tts_stream_play at /home/sunxiangyu/aishare/aigit2/esp_tts/build/../components/speech_synthesis/esp_tts_chinese/esp_tts.c:548

0x420057d7: app_main at /home/winter/item/esp-skainet/examples/chinese_tts/build/../main/main.c:85 (discriminator 1)

0x42026fbc: main_task at /home/winter/item/esp/esp-idf/components/freertos/port/port_common.c:129

lib文件报错，使用method1或者method2，也一样错误

FRANCISEG, 2021-12-23:
使用IDF4.4

feizi, 2022-01-04:
Have fixed it, pls try new commit.

FRANCISEG, 2022-01-04:
It still doesn't work

FRANCISEG, 2022-01-04:
method2：

FRANCISEG, 2022-01-04:
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x12e4
load:0x403b6000,len:0x9c0
load:0x403ba000,len:0x2e08
entry 0x403b6184
W (29) bootloader_random: RNG for ESP32-S3 not currently supported
W (84) bootloader_random: RNG for ESP32-S3 not currently supported
I (91) cpu_start: Pro cpu up.
I (91) cpu_start: Starting app cpu, entry point is 0x40375184
0x40375184: call_start_cpu1 at /home/winter/item/esp/esp-idf/components/esp_system/port/cpu_start.c:156

I (69) cpu_start: App cpu up.
I (105) cpu_start: Pro cpu start user code
I (105) cpu_start: cpu freq: 160000000
I (105) cpu_start: Application information:
I (108) cpu_start: Project name:     chinese_tts
I (113) cpu_start: App version:      v0.3.0-95-g1d9a141-dirty
I (119) cpu_start: Compile time:     Jan  4 2022 15:34:28
I (125) cpu_start: ELF file SHA256:  5c7f5df1b806df70...
I (131) cpu_start: ESP-IDF:          v4.4-beta1-20-g35b20ca-dirty
I (138) heap_init: Initializing. RAM available for dynamic allocation:
I (145) heap_init: At 3FC94EA8 len 0004B158 (300 KiB): D/IRAM
I (152) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (158) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (165) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (171) spi_flash: detected chip: gd
I (175) spi_flash: flash io: qio
I (180) sleep: Configure to isolate all GPIO pins in sleep state
I (186) sleep: Enable automatic switching of GPIO sleep configuration
I (193) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (214) MEDIA_HAL: CONFIG_CODEC_CHIP_IS_ES8311
I (214) I2S: DMA Malloc info, datalen=blocksize=1280, dma_buf_count=6
I (224) I2S: DMA Malloc info, datalen=blocksize=1280, dma_buf_count=6
I (224) I2S: I2S1, MCLK output by GPIO0
I (234) DRV8311: Enter into es8311_Codec_Startup()

I (244) DRV8311: ES8311 in Slave mode

I (244) DRV8311: Enter into es8311_pcm_hw_params()

I (264) DRV8311: Exit es8311_Codec_Startup()

I (264) DRV8311: ES8311 in I2S Format
I (264) DRV8311: SET: volume:153

I (264) MEDIA_HAL: I2S_OUT_VOL_DEFAULT[60]
WYB:Slave mode:80 !
WYB:ES7210_MCLK_CTL_REG02 = 0xC3!
I (284) gpio: GPIO[46]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
codec init done!!!!
I (284) DRV8311: SET: volume:165

inti voice set:template
ESP Chinese TTS v1.2 (Dec 30 2021 14:27:48)
欢迎使用乐鑫语音合成
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42006cd9  PS      : 0x00060c30  A0      : 0x82007247  A1      : 0x3fcf3f00  
0x42006cd9: esp_tts_parser_extern at /home/sunxiangyu/aishare/aigit2/esp_tts/build/../components/speech_synthesis/esp_tts_chinese/esp_tts_parser.c:589

A2      : 0x3c032c5f  A3      : 0x3fce0d74  A4      : 0x00001d22  A5      : 0x9b79659b  
A6      : 0x9b799fe1  A7      : 0x00000001  A8      : 0x82006ccc  A9      : 0x3fcf3ee0  
A10     : 0x00006b22  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x00000080  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x9b799fe1  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffff8  

Backtrace:0x42006cd6:0x3fcf3f000x42007244:0x3fcf3f30 0x42005bc0:0x3fcf3f80 0x4200581f:0x3fcf3fa0 0x420270e4:0x3fcf43f0 
0x42006cd6: esp_tts_parser_extern at /home/sunxiangyu/aishare/aigit2/esp_tts/build/../components/speech_synthesis/esp_tts_chinese/esp_tts_parser.c:589

0x42007244: esp_tts_parser_chinese at /home/sunxiangyu/aishare/aigit2/esp_tts/build/../components/speech_synthesis/esp_tts_chinese/esp_tts_parser.c:893

0x42005bc0: esp_tts_parse_chinese at /home/sunxiangyu/aishare/aigit2/esp_tts/build/../components/speech_synthesis/esp_tts_chinese/esp_tts.c:167

0x4200581f: app_main at /home/winter/item/esp-skainet/examples/chinese_tts/build/../main/main.c:84

0x420270e4: main_task at /home/winter/item/esp/esp-idf/components/freertos/port/port_common.c:129

feizi, 2022-01-05:
The address to flash voice_data should be the offset of voice_data in parations.csv. 
I have corrected the wrong address in flash_voicedata.sh. pls try again.

FRANCISEG, 2022-01-07:
the  log i try is first download :
source flash_voicedata.sh ../../components/esp-sr/esp-tts/esp_tts_chinese/esp_tts_voice_data_xiaole.dat  /dev/ttyUSB0 ;
then download method2 bin ,use  idf.py flash monitor -p /dev/ttyUSB0 .

FRANCISEG, 2022-01-07:
But still wrong

feizi, 2022-01-07:
I can run chinese_tts by method2, and your actions are correct. 
What error did you encounter?

closed 2022-01-24 by feizi