espressif/esp-skainet #42: 关于编译wake_word_detection程序的问题

PandaChen-pro, 2022-01-12:
板子使用的esp32-korvo-v1.1
在编译wake_word_detection这个程序的时候出现了cc1.exe: some warnings being treated as errors的错误，在stackoverflow上找到了解决方法，在makefile中加入CFLAGS = "-Wno-error"
CXXFLAGS = "-Wno-error"来忽略警告引起的错误，但是仍然不起作用，请问有什么解决方案

979/988] Building C object esp-idf/player/CMakeFiles/__idf_player.dir/esp_skainet_player.c.obj
In file included from C:/Users/MSI/esp/esp-idf/components/freertos/include/freertos/semphr.h:35,
                 from C:/Users/MSI/esp/esp-idf/components/driver/include/driver/i2s.h:20,
                 from C:/Users/MSI/Desktop/esp-skainet-master/components/player/esp_skainet_player.c:8:
C:/Users/MSI/Desktop/esp-skainet-master/components/player/esp_skainet_player.c: In function 'esp_skainet_stream_in_task':
C:/Users/MSI/Desktop/esp-skainet-master/components/player/esp_skainet_player.c:93:31: warning: passing argument 1 of 'xQueueGenericSend' from incompatible pointer type [-Wincompatible-pointer-types]
              xQueueSend(player->player_queue, buffer, portMAX_DELAY);
C:/Users/MSI/esp/esp-idf/components/freertos/include/freertos/queue.h:421:80: note: in definition of macro 'xQueueSend'
 #define xQueueSend( xQueue, pvItemToQueue, xTicksToWait ) xQueueGenericSend( ( xQueue ), ( pvItemToQueue ), ( xTicksToWait ), queueSEND_TO_BACK )
                                                                                ^~~~~~
C:/Users/MSI/esp/esp-idf/components/freertos/include/freertos/queue.h:572:45: note: expected 'QueueHandle_t' {aka 'struct QueueDefinition *'} but argument is of type 'struct RingBuf *' BaseType_t xQueueGenericSend( QueueHandle_t xQueue, const void * const pvItemToQueue, TickType_t xTicksToWait, const BaseType_t xCopyPosition ) PRIVILEGED_FUNCTION;
                               ~~~~~~~~~~~~~~^~~~~~
C:/Users/MSI/Desktop/esp-skainet-master/components/player/esp_skainet_player.c:94:31: warning: passing argument 1 of 'xQueueGenericSend' from incompatible pointer type [-Wincompatible-pointer-types]
              xQueueSend(player->player_queue, buffer+player->frame_size, portMAX_DELAY);
C:/Users/MSI/esp/esp-idf/components/freertos/include/freertos/queue.h:421:80: note: in definition of macro 'xQueueSend'
 #define xQueueSend( xQueue, pvItemToQueue, xTicksToWait ) xQueueGenericSend( ( xQueue ), ( pvItemToQueue ), ( xTicksToWait ), queueSEND_TO_BACK )
                                                                                ^~~~~~
C:/Users/MSI/esp/esp-idf/components/freertos/include/freertos/queue.h:572:45: note: expected 'QueueHandle_t' {aka 'struct QueueDefinition *'} but argument is of type 'struct RingBuf *' BaseType_t xQueueGenericSend( QueueHandle_t xQueue, const void * const pvItemToQueue, TickType_t xTicksToWait, const BaseType_t xCopyPosition ) PRIVILEGED_FUNCTION;
                               ~~~~~~~~~~~~~~^~~~~~
C:/Users/MSI/Desktop/esp-skainet-master/components/player/esp_skainet_player.c:98:30: warning: passing argument 1 of 'xQueueGenericSend' from incompatible pointer type [-Wincompatible-pointer-types]
             xQueueSend(player->player_queue, buffer, portMAX_DELAY);
C:/Users/MSI/esp/esp-idf/components/freertos/include/freertos/queue.h:421:80: note: in definition of macro 'xQueueSend'

WeldonWangwang, 2022-01-13:
Hi, @PandaChen-pro ,

这个问题如果不能在本地通过改写 make 解决的话，建议可以直接将 https://github.com/espressif/esp-skainet/blob/aa4f57e5fa0a52e2db6ceb1d9830af4c05cabccf/components/hardware_driver/boards/esp32-korvo/bsp_board.c#L262 该函数注释掉先使用，注释掉该函数不影响功能。

我们会最近修复这个问题，谢谢反馈！

另：ESP32S3 支持的唤醒模型相比 ESP32 有很大提升，如方便，我们建议后续可以使用 ESP32S3.

PandaChen-pro, 2022-01-13:
> 嗨，@PandaChen-pro，
> 
> 这个如果不能在本地通过改写问题提出解决方案，建议可以直接将
> 
> https://github.com/espressif/esp-skainet/blob/aa4f57e5fa0a52e2db6ceb1d9830af4c05cabccf/components/hardware_driver/boards/esp32-korvo/bsp_board.c#L262
> 
> 该函数注释掉先使用，注释掉该函数不影响功能。
> 我们会最近修复这个问题，谢谢反馈！
> 
> 另 ESP3S3 支持的触控模式相比 ESP32 有很大的提升，如方便，我们建议使用 ESP3S32。

您好。按照您的方式，将static esp_err_t bsp_codec_prob(int *codec_type)此函数注释，仍然还是报错，具体的报错日志如下：

[964/988] Building C object esp-idf/hardware_driver/CMakeFiles/__idf_hardware_driver.dir/boards/esp32-korvo/bsp_board.c.obj
FAILED: esp-idf/hardware_driver/CMakeFiles/__idf_hardware_driver.dir/boards/esp32-korvo/bsp_board.c.obj 
D:\Develop_Tools\.espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -Iconfig -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/include -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/codec/include -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/codec/es7210 -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/codec/es8311 -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/codec/es8388 -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/i2c_bus/include -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/boards/include -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/led/include -IC:/Users/MSI/esp/esp-idf/components/newlib/platform_include -IC:/Users/MSI/esp/esp-idf/components/freertos/include -IC:/Users/MSI/esp/esp-idf/components/freertos/port/xtensa/include -IC:/Users/MSI/esp/esp-idf/components/esp_hw_support/include -IC:/Users/MSI/esp/esp-idf/components/esp_hw_support/port/esp32/. -IC:/Users/MSI/esp/esp-idf/components/heap/include -IC:/Users/MSI/esp/esp-idf/components/log/include -IC:/Users/MSI/esp/esp-idf/components/lwip/include/apps -IC:/Users/MSI/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/MSI/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/MSI/esp/esp-idf/components/lwip/port/esp32/include -IC:/Users/MSI/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/MSI/esp/esp-idf/components/soc/include -IC:/Users/MSI/esp/esp-idf/components/soc/esp32/. -IC:/Users/MSI/esp/esp-idf/components/soc/esp32/include -IC:/Users/MSI/esp/esp-idf/components/hal/esp32/include -IC:/Users/MSI/esp/esp-idf/components/hal/include -IC:/Users/MSI/esp/esp-idf/components/esp_rom/include -IC:/Users/MSI/esp/esp-idf/components/esp_rom/esp32 -IC:/Users/MSI/esp/esp-idf/components/esp_rom/include/esp32 -IC:/Users/MSI/esp/esp-idf/components/esp_common/include -IC:/Users/MSI/esp/esp-idf/components/esp_system/include -IC:/Users/MSI/esp/esp-idf/components/esp32/include -IC:/Users/MSI/esp/esp-idf/components/driver/include -IC:/Users/MSI/esp/esp-idf/components/driver/esp32/include -IC:/Users/MSI/esp/esp-idf/components/esp_ringbuf/include -IC:/Users/MSI/esp/esp-idf/components/efuse/include -IC:/Users/MSI/esp/esp-idf/components/efuse/esp32/include -IC:/Users/MSI/esp/esp-idf/components/xtensa/include -IC:/Users/MSI/esp/esp-idf/components/xtensa/esp32/include -IC:/Users/MSI/esp/esp-idf/components/espcoredump/include -IC:/Users/MSI/esp/esp-idf/components/esp_timer/include -IC:/Users/MSI/esp/esp-idf/components/esp_ipc/include -IC:/Users/MSI/esp/esp-idf/components/esp_pm/include -IC:/Users/MSI/esp/esp-idf/components/vfs/include -IC:/Users/MSI/esp/esp-idf/components/esp_wifi/include -IC:/Users/MSI/esp/esp-idf/components/esp_wifi/esp32/include -IC:/Users/MSI/esp/esp-idf/components/esp_event/include -IC:/Users/MSI/esp/esp-idf/components/esp_netif/include -IC:/Users/MSI/esp/esp-idf/components/esp_eth/include -IC:/Users/MSI/esp/esp-idf/components/tcpip_adapter/include -IC:/Users/MSI/esp/esp-idf/components/app_trace/include -IC:/Users/MSI/esp/esp-idf/components/fatfs/diskio -IC:/Users/MSI/esp/esp-idf/components/fatfs/vfs -IC:/Users/MSI/esp/esp-idf/components/fatfs/src -IC:/Users/MSI/esp/esp-idf/components/wear_levelling/include -IC:/Users/MSI/esp/esp-idf/components/spi_flash/include -IC:/Users/MSI/esp/esp-idf/components/sdmmc/include -IC:/Users/MSI/esp/esp-idf/components/spiffs/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.3.2-dirty\" -DESP_PLATFORM -MD -MT esp-idf/hardware_driver/CMakeFiles/__idf_hardware_driver.dir/boards/esp32-korvo/bsp_board.c.obj -MF esp-idf\hardware_driver\CMakeFiles\__idf_hardware_driver.dir\boards\esp32-korvo\bsp_board.c.obj.d -o esp-idf/hardware_driver/CMakeFiles/__idf_hardware_driver.dir/boards/esp32-korvo/bsp_board.c.obj   -c C:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/boards/esp32-korvo/bsp_board.c
In file included from C:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/boards/include/bsp_board.h:35,
                 from C:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/boards/esp32-korvo/bsp_board.c:25:
C:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/boards/include/esp32_korvo_v1_1_board.h:103:32: warning: backslash-newline at end of file
 #define I2S_CONFIG_DEFAULT() { \

[... cut ...]
C:/Users/MSI/Desktop/esp-skainet-master/components/player/esp_skainet_player.c:241:25: warning: passing argument 2 of 'file_list_scan' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  file_list_scan(player, path);
                         ^~~~
C:/Users/MSI/Desktop/esp-skainet-master/components/player/esp_skainet_player.c:170:40: note: expected 'char *' but argument is of type 'const char *'
 int file_list_scan(void *handle, char *path)
                                  ~~~~~~^~~~
[977/988] Building C object esp-idf/sr_ringbuf/CMakeFiles/__idf_sr_ringbuf.dir/lock.c.obj
ninja: build stopped: subcommand failed.

WeldonWangwang, 2022-01-13:
hi @PandaChen-pro :
1. 通过查看你的log，请问你是否使用的是 make 编译方式？目前 Skainet 只支持 Cmake 编译。
2. 注释掉的函数应该是 bsp_i2s_deinit，不是 bsp_codec_prob.

PandaChen-pro, 2022-01-13:
> 非常感谢您的回复。我使用的是vscode编译，注释的函数确实是bsp_i2s_deinit
// static esp_err_t bsp_i2s_deinit(i2s_port_t i2s_num)
// {
//     esp_err_t ret_val = ESP_OK;

//     ret_val |= i2s_stop(I2S_NUM_0);
//     ret_val |= i2s_driver_uninstall(i2s_num);

//     return ret_val;
// }

PandaChen-pro, 2022-01-13:
非常抱歉这么晚打扰您，我进行了很多次其他example案例的编译，同样是报错，发现似乎一直是bsp_board.c文件的问题，请问除了注释bsp_i2s_deinit这个函数外，是否还要进行其他操作。

// static esp_err_t bsp_i2s_deinit(i2s_port_t i2s_num)
// {
//     esp_err_t ret_val = ESP_OK;
//     ret_val |= i2s_stop(I2S_NUM_0);
//     ret_val |= i2s_driver_uninstall(i2s_num);

//     return ret_val;
// }

下面是错误日志：

[2/9] Building C object esp-idf/hardware_driver/CMakeFiles/__idf_hardware_driver.dir/boards/esp32-korvo/bsp_board.c.obj
FAILED: esp-idf/hardware_driver/CMakeFiles/__idf_hardware_driver.dir/boards/esp32-korvo/bsp_board.c.obj
ccache D:\Develop_Tools\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -Iconfig -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/include -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/codec/include -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/codec/es7210 -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/codec/es8311 -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/codec/es8388 -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/i2c_bus/include -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/boards/include -IC:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/led/include -ID:/Develop_Tools/esp-idf/components/newlib/platform_include -ID:/Develop_Tools/esp-idf/components/freertos/include -ID:/Develop_Tools/esp-idf/components/freertos/port/xtensa/include -ID:/Develop_Tools/esp-idf/components/esp_hw_support/include -ID:/Develop_Tools/esp-idf/components/esp_hw_support/port/esp32/. -ID:/Develop_Tools/esp-idf/components/heap/include -ID:/Develop_Tools/esp-idf/components/log/include -ID:/Develop_Tools/esp-idf/components/lwip/include/apps -ID:/Develop_Tools/esp-idf/components/lwip/include/apps/sntp -ID:/Develop_Tools/esp-idf/components/lwip/lwip/src/include -ID:/Develop_Tools/esp-idf/components/lwip/port/esp32/include -ID:/Develop_Tools/esp-idf/components/lwip/port/esp32/include/arch -ID:/Develop_Tools/esp-idf/components/soc/include -ID:/Develop_Tools/esp-idf/components/soc/esp32/. -ID:/Develop_Tools/esp-idf/components/soc/esp32/include -ID:/Develop_Tools/esp-idf/components/hal/esp32/include -ID:/Develop_Tools/esp-idf/components/hal/include -ID:/Develop_Tools/esp-idf/components/esp_rom/include -ID:/Develop_Tools/esp-idf/components/esp_rom/esp32 -ID:/Develop_Tools/esp-idf/components/esp_rom/include/esp32 -ID:/Develop_Tools/esp-idf/components/esp_common/include -ID:/Develop_Tools/esp-idf/components/esp_system/include -ID:/Develop_Tools/esp-idf/components/esp32/include -ID:/Develop_Tools/esp-idf/components/driver/include -ID:/Develop_Tools/esp-idf/components/driver/esp32/include -ID:/Develop_Tools/esp-idf/components/esp_ringbuf/include -ID:/Develop_Tools/esp-idf/components/efuse/include -ID:/Develop_Tools/esp-idf/components/efuse/esp32/include -ID:/Develop_Tools/esp-idf/components/xtensa/include -ID:/Develop_Tools/esp-idf/components/xtensa/esp32/include -ID:/Develop_Tools/esp-idf/components/espcoredump/include -ID:/Develop_Tools/esp-idf/components/esp_timer/include -ID:/Develop_Tools/esp-idf/components/esp_ipc/include -ID:/Develop_Tools/esp-idf/components/esp_pm/include -ID:/Develop_Tools/esp-idf/components/vfs/include -ID:/Develop_Tools/esp-idf/components/esp_wifi/include -ID:/Develop_Tools/esp-idf/components/esp_wifi/esp32/include -ID:/Develop_Tools/esp-idf/components/esp_event/include -ID:/Develop_Tools/esp-idf/components/esp_netif/include -ID:/Develop_Tools/esp-idf/components/esp_eth/include -ID:/Develop_Tools/esp-idf/components/tcpip_adapter/include -ID:/Develop_Tools/esp-idf/components/app_trace/include -ID:/Develop_Tools/esp-idf/components/fatfs/diskio -ID:/Develop_Tools/esp-idf/components/fatfs/vfs -ID:/Develop_Tools/esp-idf/components/fatfs/src -ID:/Develop_Tools/esp-idf/components/wear_levelling/include -ID:/Develop_Tools/esp-idf/components/spi_flash/include -ID:/Develop_Tools/esp-idf/components/sdmmc/include -ID:/Develop_Tools/esp-idf/components/spiffs/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.3.1\" -DESP_PLATFORM -MD -MT esp-idf/hardware_driver/CMakeFiles/__idf_hardware_driver.dir/boards/esp32-korvo/bsp_board.c.obj -MF esp-idf\hardware_driver\CMakeFiles\__idf_hardware_driver.dir\boards\esp32-korvo\bsp_board.c.obj.d -o esp-idf/hardware_driver/CMakeFiles/__idf_hardware_driver.dir/boards/esp32-korvo/bsp_board.c.obj   -c C:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/boards/esp32-korvo/bsp_board.c
In file included from C:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/boards/include/bsp_board.h:35,
                 from C:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/boards/esp32-korvo/bsp_board.c:25:
C:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/boards/include/esp32_korvo_v1_1_board.h:103:32: warning: backslash-newline at end of file
 #define I2S_CONFIG_DEFAULT() { \

C:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/boards/esp32-korvo/bsp_board.c:226:107: error: unknown type name 'i2s_bits_per_chan_t'; did you mean 'i2s_bits_per_sample_t'?
[... cut ...]
     bsp_i2s_init(I2S_NUM_1, 16000, I2S_CHANNEL_FMT_RIGHT_LEFT, I2S_BITS_PER_CHAN_32BIT);
                                                                ^~~~~~~~~~~~~~~~~~~~~~~
                                                                I2S_BITS_PER_SAMPLE_32BIT
C:/Users/MSI/Desktop/esp-skainet-master/components/hardware_driver/boards/esp32-korvo/bsp_board.c:395:64: note: each undeclared identifier is reported only once for each function it appears in
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

WeldonWangwang, 2022-01-14:
@PandaChen-pro 请问你使用的 IDF 版本是多少呀？分支和commit？

WeldonWangwang, 2022-01-14:
> @PandaChen-pro 请问你使用的 IDF 版本是多少呀？分支和commit？

目前 Skaient  master 分支只支持 IDF release/v4.4 https://github.com/espressif/esp-idf/tree/release/v4.4

PandaChen-pro, 2022-01-14:
目前

> > @PandaChen-pro 请问你使用的 IDF 版本是多少呀？分支和commit？
> 
> 目前 Skaient master 分支只支持 IDF release/v4.4 https://github.com/espressif/esp-idf/tree/release/v4.4

目前用的IDF版本是v4.3，是需要进行更新吗

WeldonWangwang, 2022-01-14:
@PandaChen-pro 
建议将 IDF 更新到 release/v4.4, 因为 log 中报错的结构体在 4.3 中没有定义，4.4 针对 I2S 进行了接口改动。

PandaChen-pro, 2022-01-14:
> @PandaChen-pro 建议将 IDF 更新到 release/v4.4, 因为 log 中报错的结构体在 4.3 中没有定义，4.4 针对 I2S 进行了接口改动。

非常感谢您的帮助，现在已经能够正常编译了，但是在烧录后，一直会重复出现下面的字符串，不能正常工作，请问这应该怎么解决。

ELF file SHA256: 1c17032cd93c17ab

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6724
load:0x40078000,len:14812
xtensa-esp32-elfaddr2line -pfiaC -e c:\users\msi\desktop\esp32\esp-skainet\examples\wake_word_detection\build\wake_word_detection.elf 0x40078000: [WinError 2] ?????????????????????????????????
load:0x40080400,len:3792
xtensa-esp32-elfaddr2line -pfiaC -e c:\users\msi\desktop\esp32\esp-skainet\examples\wake_word_detection\build\wake_word_detection.elf 0x40080400: [WinError 2] ?????????????????????????????????
entry 0x40080694
xtensa-esp32-elfaddr2line -pfiaC -e c:\users\msi\desktop\esp32\esp-skainet\examples\wake_word_detection\build\wake_word_detection.elf 0x40080694: [WinError 2] ?????????????????????????????????
I (27) boot: ESP-IDF v4.4-beta1-2-gdf0929009a 2nd stage bootloader
I (27) boot: compile time 17:07:24
I (27) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 3, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=34f5ch (216924) map
I (189) esp_image: segment 1: paddr=00044f84 vaddr=3ffb0000 size=025c0h (  9664) load
I (193) esp_image: segment 2: paddr=0004754c vaddr=40080000 size=08acch ( 35532) load
I (210) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=55f24h (352036) map
I (337) esp_image: segment 4: paddr=000a5f4c vaddr=40088acc size=073f8h ( 29688) load
I (350) esp_image: segment 5: paddr=000ad34c vaddr=50000000 size=00010h (    16) load
I (358) boot: Loaded app from partition at offset 0x10000
I (358) boot: Disabling RNG early entropy source...
I (370) cpu_start: Pro cpu up.
I (370) cpu_start: Starting app cpu, entry point is 0x400814d0
xtensa-esp32-elfaddr2line -pfiaC -e c:\users\msi\desktop\esp32\esp-skainet\examples\wake_word_detection\build\wake_word_detection.elf 0x400814d0: [WinError 2] ?????????????????????????????????
I (357) cpu_start: App cpu up.
I (385) cpu_start: Pro cpu start user code
I (385) cpu_start: cpu freq: 160000000
I (385) cpu_start: Application information:
I (389) cpu_start: Project name:     wake_word_detection
I (395) cpu_start: App version:      v0.3.0-104-gaa4f57e-dirty
I (402) cpu_start: Compile time:     Jan 14 2022 17:07:06
I (408) cpu_start: ELF file SHA256:  1c17032cd93c17ab...
I (414) cpu_start: ESP-IDF:          v4.4-beta1-2-gdf0929009a
I (420) heap_init: Initializing. RAM available for dynamic allocation:
I (427) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (433) heap_init: At 3FFB44C8 len 0002BB38 (174 KiB): DRAM
I (440) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (446) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (452) heap_init: At 4008FEC4 len 0001013C (64 KiB): IRAM
I (460) spi_flash: detected chip: gd
I (463) spi_flash: flash io: dio
W (467) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (481) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (493) I2S: DMA Malloc info, datalen=blocksize=1280, dma_buf_count=6
I (503) I2S: I2S1, MCLK output by GPIO0
I (503) I2S: DMA Malloc info, datalen=blocksize=320, dma_buf_count=6
I (513) I2S: I2S0, MCLK output by GPIO0
[... 11 lines cut ...]
I (573) ES7210: Enable ES7210_INPUT_MIC4
I (583) DRV8311: ES8311 in Slave mode
I (593) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
Item psram alloc failed. Size: 2068 = 1024 x 2 + 16 + 4
Deprecated Quantized wakeNet5: wakeNet5_v1_hilexin_5_0.95_0.90, mode:0 (Dec 10 2021 20:15:15)
3
Item psram alloc failed. Size: 2068 = 2048 x 1 + 16 + 4
Initial ONE-MIC auido front-end for speech recognition, mode:1, (Dec 10 2021 20:29:24)
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d7ba8  PS      : 0x00060c30  A0      : 0x800d695c  A1      : 0x3ffe3910
xtensa-esp32-elfaddr2line -pfiaC -e c:\users\msi\desktop\esp32\esp-skainet\examples\wake_word_detection\build\wake_word_detection.elf 0x400d7ba8: [WinError 2] ?????????????????????????????????
A2      : 0x3ffaf3ec  A3      : 0x3ffecdb4  A4      : 0x00060021  A5      : 0x00060023
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00001000  A12     : 0x00000200  A13     : 0x3ffb7510
A14     : 0x00000000  A15     : 0x00001000  SAR     : 0x0000001c  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
[... cut ...]
xtensa-esp32-elfaddr2line -pfiaC -e c:\users\msi\desktop\esp32\esp-skainet\examples\wake_word_detection\build\wake_word_detection.elf 0x400d7ba5: [WinError 2] ?????????????????????????????????
xtensa-esp32-elfaddr2line -pfiaC -e c:\users\msi\desktop\esp32\esp-skainet\examples\wake_word_detection\build\wake_word_detection.elf 0x400d6959: [WinError 2] ?????????????????????????????????
xtensa-esp32-elfaddr2line -pfiaC -e c:\users\msi\desktop\esp32\esp-skainet\examples\wake_word_detection\build\wake_word_detection.elf 0x4008baad: [WinError 2] ?????????????????????????????????

PandaChen-pro, 2022-01-14:
接收到的消息如下：
[2022-01-14 17:27:01.307]# RECV ASCII>
[0;32mI (337) esp_image: segment 4: paddr=000a5f4c vaddr=40088acc size=073f8h ( 29688) load[0m
[0;32mI (350) esp_image: segment 5: paddr=000ad34c vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (358) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (358) boot: Disabling RNG early entropy source...[0m
[0;32mI (370) cpu_start: Pro cpu up.[0m
[0;32mI (370) cpu_start: Starting app cpu, entry point is 0x400814d0[0m
[0;32mI (357) cpu_start: App cpu up.[0m
[0;32mI (385) cpu_st

WeldonWangwang, 2022-01-14:
@PandaChen-pro 从 log 看，应该是芯片配置问题，请先删掉工程目录下的 build 文件夹，然后拷贝 sdkconfig.defaults.esp32 为 sdkconfig ，再次编译。如仍有问题，麻烦提供你目前工程中使用的 sdkconfig 文件。

PandaChen-pro, 2022-01-14:
> @PandaChen-pro 从 log 看，应该是芯片配置问题，请先删掉工程目录下的 build 文件夹，然后拷贝 sdkconfig.defaults.esp32 为 sdkconfig ，再次编译。如仍有问题，麻烦提供你目前工程中使用的 sdkconfig 文件。

非常感谢您的回复，我是用`idf.py fullclean`命令将build文件夹清楚后，重新编译烧录后仍为这样的错误，使用的sdkconfig如下：

CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions_esp32.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions_esp32.csv"
CONFIG_PARTITION_TABLE_OFFSET=0x8000
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32_SPIRAM_SUPPORT=y
CONFIG_SPIRAM_SPEED_80M=y
# CONFIG_ESP_SYSTEM_PANIC_PRINT_HALT is not set
CONFIG_ESP_SYSTEM_PANIC_PRINT_REBOOT=y
# CONFIG_ESP_SYSTEM_PANIC_SILENT_REBOOT is not set
# CONFIG_ESP_SYSTEM_PANIC_GDBSTUB is not set
# CONFIG_ESP_SYSTEM_GDBSTUB_RUNTIME is not set

CONFIG_MODEL_IN_SPIFFS=y
# CONFIG_MODEL_IN_SDCARD is not set
CONFIG_USE_WAKENET=y
CONFIG_SR_WN_MODEL_WN5_QUANT=y
CONFIG_SR_WN_WN5_HILEXIN=y
# CONFIG_USE_MULTINET is not set

PandaChen-pro, 2022-01-14:
很抱歉再次打扰您，现在已经能够正常烧录进行监听了，但是当我说出命令词时，没有反应。
输出日志如下：
```
entry 0x400806b4
I (27) boot: ESP-IDF v4.4-beta1-2-gdf0929009a 2nd stage bootloader
I (27) boot: compile time 19:12:18
I (27) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32: SPI Speed      : 80MHz
I (49) boot.esp32: SPI Mode       : QIO
I (53) boot.esp32: SPI Flash Size : 16MB
I (58) boot: Enabling RNG early entropy source...
I (63) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 factory          factory app      00 00 00010000 00500000
I (82) boot: End of partition table
I (86) boot_comm: chip revision: 3, min. application chip revision: 0
I (93) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=390a4h (233636) map
I (166) esp_image: segment 1: paddr=000490cc vaddr=3ffb0000 size=02afch ( 11004) load
I (169) esp_image: segment 2: paddr=0004bbd0 vaddr=40080000 size=04448h ( 17480) load
I (177) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=5e790h (386960) map
I (287) esp_image: segment 4: paddr=000ae7b8 vaddr=40084448 size=114e8h ( 70888) load
I (310) esp_image: segment 5: paddr=000bfca8 vaddr=50000000 size=00010h (    16) load
I (320) boot: Loaded app from partition at offset 0x10000
I (321) boot: Disabling RNG early entropy source...
I (332) psram: This chip is ESP32-D0WD
I (333) spiram: Found 64MBit SPI RAM device
I (333) spiram: SPI RAM mode: flash 80m sram 80m
I (338) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (345) cpu_start: Pro cpu up.
I (349) cpu_start: Starting app cpu, entry point is 0x400817ec
0x400817ec: call_start_cpu1 at C:/Users/MSI/Desktop/esp32/esp-skainet/esp-idf/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (851) spiram: SPI SRAM memory test OK
I (859) cpu_start: Pro cpu start user code
I (859) cpu_start: cpu freq: 240000000
I (859) cpu_start: Application information:
I (862) cpu_start: Project name:     wake_word_detection
I (868) cpu_start: App version:      v0.3.0-104-gaa4f57e-dirty
I (874) cpu_start: Compile time:     Jan 14 2022 19:25:15
I (880) cpu_start: ELF file SHA256:  568f2f412c549f60...
I (886) cpu_start: ESP-IDF:          v4.4-beta1-2-gdf0929009a
I (893) heap_init: Initializing. RAM available for dynamic allocation:
I (900) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (906) heap_init: At 3FFB4A50 len 0002B5B0 (173 KiB): DRAM
I (912) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (918) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (925) heap_init: At 40095930 len 0000A6D0 (41 KiB): IRAM
I (931) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (939) spi_flash: detected chip: gd
I (943) spi_flash: flash io: qio
I (948) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (958) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (968) I2S: DMA Malloc info, datalen=blocksize=1280, dma_buf_count=6
I (978) I2S: I2S1, MCLK output by GPIO0
I (978) I2S: DMA Malloc info, datalen=blocksize=320, dma_buf_count=6
I (988) I2S: I2S0, MCLK output by GPIO0
I (988) board: Detected codec at 0x40. Name : ES7210
I (998) board: Detected codec at 0x18. Name : ES8311
I (998) ES7210: ES7210 in Slave mode
I (1008) ES7210: Enable ES7210_INPUT_MIC1
I (1008) ES7210: Enable ES7210_INPUT_MIC2
I (1018) ES7210: Enable ES7210_INPUT_MIC3
I (1018) ES7210: Enable ES7210_INPUT_MIC4
I (1028) ES7210: The ES7210_CLOCK_OFF_REG01 value before stop is 0
I (1038) ES7210: Enable ES7210_INPUT_MIC1
I (1038) ES7210: Enable ES7210_INPUT_MIC2
[... 1580 lines cut ...]
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options

```

WeldonWangwang, 2022-01-14:
@PandaChen-pro 
你目前使用的这个 demo 是针对唤醒词的，只会运行唤醒词，如果需要使用 唤醒词+命令词，请使用 cn_speech_commands_recognition 这个 demo

PandaChen-pro, 2022-01-14:
好的，谢谢您，目前已经跑通三个官方Demo了

closed 2022-01-20 by feizi