espressif/esp-skainet #43: 关于文件缺失的问题

PandaChen-pro, 2022-02-02:
请问我在跑一个老例程时，出现了找不到MediaHal.h的问题，在github仓库内确实没有这个文件，请问是已经删除了吗。

```c
../main/main.c: In function 'wakenetTask':
../main/main.c:64:9: error: implicit declaration of function 'rb_read'; did you mean 'wl_read'? [-Werror=implicit-function-declaration]
         rb_read(agc_rb, (uint8_t *)buffer, audio_chunksize * sizeof(int16_t), portMAX_DELAY);
         ^~~~~~~
         wl_read
../main/main.c:72:17: error: implicit declaration of function 'rb_reset'; did you mean 'fd_set'? [-Werror=implicit-function-declaration]
                 rb_reset(rec_rb);
                 ^~~~~~~~
                 fd_set
../main/main.c: In function 'app_main':
../main/main.c:272:5: error: implicit declaration of function 'codec_init'; did you mean 'gpio_init'? [-Werror=implicit-function-declaration]
     codec_init();
     ^~~~~~~~~~
     gpio_init
../main/main.c:276:13: error: implicit declaration of function 'rb_init'; did you mean 'gpio_init'? [-Werror=implicit-function-declaration]
  dump1_rb = rb_init(BUFFER_PROCESS, 8 * 1024, 1, NULL);
             ^~~~~~~
             gpio_init
../main/main.c:276:21: error: 'BUFFER_PROCESS' undeclared (first use in this function)
  dump1_rb = rb_init(BUFFER_PROCESS, 8 * 1024, 1, NULL);
                     ^~~~~~~~~~~~~~
../main/main.c:276:21: note: each undeclared identifier is reported only once for each function it appears in
```

feizi, 2022-02-07:
是的,我们重构了硬件驱动模块

PandaChen-pro, 2022-02-07:
那请问还能不能找到之前的这个MediaHal.h文件

feizi, 2022-02-09:
之前的这个文件MediaHal.h被删掉了

closed 2022-02-14 by feizi