espressif/esp-skainet #44: ESP32-LyraT V4.3 No longer supported?

aswinda29, 2022-02-28:
I have been using ESP-Skainet with the lyraT board. In the new commits I see that the examples are no longer supported in this board. Is there a work around?

feizi, 2022-03-08:
I'm sorry that LyratV4.3 is no longer supported. If you want to use it, you can refer to the examples in ESP-ADF.
Generally, the performance of ESP32S3 on AI-related tasks is 5~10 times that of ESP32. Many new algorithms can only be run on the ESP32-S3 board.
we strongly recommend you move to the ESP32-S3 board.

Phil-f6czv, 2022-04-18:
Hello
I developped In July 2021 an application on LyraT 4.3 and Skainet with English commands. I did not see yesterday that the LyraT was no longer supported and I upgraded the IDF and the Skainet. Is it possible to get the July 2021Skainet version ?
Regards
Philippe

feizi, 2022-08-11:
We highly recommend you switch to esp32-s3 for English commands applications

closed 2022-08-23 by feizi