espressif/esp-skainet #45: esp32-s3-korvo-2加密的模组multinet有问题

LiTongXue98, 2022-03-05:
##### 环境

```
硬件：esp32-s3-korvo-2  
esp-skainet: dev2.0 Commit: 5aec971862b
esp-sr:   dev2.0  Commit: c873a359e7d
```

```
我现在的问题是在同样的工程代码（都是使用的cn_speech_commands_recognition的例程），在未加密的模组可以正常进行语音唤醒和识别，但是在另外一块已经加密的模组上，运行multinet->create(&MULTINET_COEFF, 5760);后就无法继续运行了，可以帮忙查看一下是什么问题导致的吗，万分感谢！下面是运行的一些log信息
```

#### 未加密的模组可以正常运行

```
I (25) boot: ESP-IDF v4.4-beta1-50-g5e13980e99-dirty 2nd stage bootloader
I (25) boot: compile time 10:03:07
I (25) boot: chip revision: 0
I (30) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (38) boot.esp32s3: SPI Flash Size : 8MB
I (43) boot: Enabling RNG early entropy source...
W (48) bootloader_random: RNG for ESP32-S3 not currently supported
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 factory          factory app      00 00 00010000 00200000
I (73) boot:  1 model            Unknown data     01 82 00210000 0050c000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=c3310h (799504) map
I (237) esp_image: segment 1: paddr=000d3338 vaddr=3fc9a2a0 size=028dch ( 10460) load
I (239) esp_image: segment 2: paddr=000d5c1c vaddr=40374000 size=0a3fch ( 41980) load
I (252) esp_image: segment 3: paddr=000e0020 vaddr=42000020 size=706ech (460524) map
I (335) esp_image: segment 4: paddr=00150714 vaddr=4037e3fc size=0be98h ( 48792) load
I (346) esp_image: segment 5: paddr=0015c5b4 vaddr=50000000 size=00010h (    16) load
I (354) boot: Loaded app from partition at offset 0x10000
I (355) boot: Disabling RNG early entropy source...
W (355) bootloader_random: RNG for ESP32-S3 not currently supported
Initializing SPIFFS
Partition size: total: 4857101, used: 3457023
model_name: xiaoaitongxue7 model_data: /srmodel/xiaoaitongxue7/wn7_data
MC Quantized wakeNet7: wakeNet7_v1_xiaoaitongxue_6_0.97_0.92, mode:2, p:3, (Feb  9 2022 16:22:38)
------------detect start------------
wakeword detected
-----------listening-----------
TOP 1, command_id: 0, phrase_id: 0, prob: 0.905920
TOP 2, command_id: 18, phrase_id: 18, prob: 0.375519
-----------listening-----------

-----------awaits to be waken up-----------
```

##### 加密模组运行到75行multinet->create(&MULTINET_COEFF, 5760);就挂掉了

```
I (29) boot: ESP-IDF v4.4-beta1-50-g5e13980e99-dirty 2nd stage bootloader
I (30) boot: compile time 10:03:07
I (30) boot: chip revision: 0
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (48) boot: Enabling RNG early entropy source...
W (53) bootloader_random: RNG for ESP32-S3 not currently supported
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 factory          factory app      00 00 00010000 00200000
I (78) boot:  1 model            Unknown data     01 82 00210000 0050c000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=c3310h (799504) map
I (257) esp_image: segment 1: paddr=000d3338 vaddr=3fc9a2a0 size=028dch ( 10460) load
I (260) esp_image: segment 2: paddr=000d5c1c vaddr=40374000 size=0a3fch ( 41980) load
I (273) esp_image: segment 3: paddr=000e0020 vaddr=42000020 size=7067ch (460412) map
I (365) esp_image: segment 4: paddr=001506a4 vaddr=4037e3fc size=0be98h ( 48792) load
I (377) esp_image: segment 5: paddr=0015c544 vaddr=50000000 size=00010h (    16) load
I (386) boot: Loaded app from partition at offset 0x10000
I (386) boot: Disabling RNG early entropy source...
W (386) bootloader_random: RNG for ESP32-S3 not currently supported
Initializing SPIFFS
Partition size: total: 4857101, used: 3457023
model_name: xiaoaitongxue7 model_data: /srmodel/xiaoaitongxue7/wn7_data
MC Quantized wakeNet7: wakeNet7_v1_xiaoaitongxue_6_0.97_0.92, mode:2, p:3, (Feb  9 2022 16:22:38)

assert failed: spinlock_release spinlock.h:140 (core_id == lock->owner)

Backtrace:0x40375cea:0x3fcac9700x40381cf5:0x3fcac990 0x40388105:0x3fcac9b0 0x40384ec7:0x3fcacad0 0x40382120:0x3fcacaf0 0x40382d1f:0x3fcacb10 0x4202a595:0x3fcacb50 0x4202a65a:0x3fcacb70 0x4037bf61:0x3fcacb90 0x4037ce31:0x3fcacbb0 0x4037ce3d:0x3fcacbd0 0x4037fcd9:0x3fcacbf0 0x4037caf9:0x3fcacc10 0x4203087d:0x3fcacc50 0x42025a95:0x3fcacc80 0x42029537:0x3fcacca0 0x4202716f:0x3fcaccd0 0x42028946:0x3fcacd10 0x42026084:0x3fcacd50 0x42025865:0x3fcacd80 0x420039b9:0x3fcacda0 0x420519c6:0x3fcacdc0 0x42051a69:0x3fcacdf0 0x420082aa:0x3fcace10 0x42008452:0x3fcace30 0x42016614:0x3fcace50 0x42016977:0x3fcacfc0 0x42006fdf:0x3fcacff0 0x40384cd6:0x3fcad020
0x40375cea: panic_abort at C:/esp-idf/components/esp_system/panic.c:402

0x40381cf5: esp_system_abort at C:/esp-idf/components/esp_system/esp_system.c:121

0x40388105: __assert_func at C:/esp-idf/components/newlib/assert.c:85

0x40384ec7: spinlock_release at C:/esp-idf/components/esp_hw_support/include/soc/spinlock.h:140
 (inlined by) vPortCPUReleaseMutex at C:/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:568
 (inlined by) vPortExitCritical at C:/esp-idf/components/freertos/port/xtensa/port.c:298

0x40382120: prvIsQueueEmpty at C:/esp-idf/components/freertos/queue.c:2474

0x40382d1f: xQueueSemaphoreTake at C:/esp-idf/components/freertos/queue.c:1667

0x4202a595: esp_ipc_call_and_wait at C:/esp-idf/components/esp_ipc/src/esp_ipc.c:150

0x4202a65a: esp_ipc_call at C:/esp-idf/components/esp_ipc/src/esp_ipc.c:162

0x4037bf61: spi_flash_disable_interrupts_caches_and_other_cpu at C:/esp-idf/components/spi_flash/cache_utils.c:163

0x4037ce31: cache_disable at C:/esp-idf/components/spi_flash/spi_flash_os_func_app.c:70

0x4037ce3d: spi1_start at C:/esp-idf/components/spi_flash/spi_flash_os_func_app.c:99

0x4037fcd9: spiflash_start_default at C:/esp-idf/components/spi_flash/esp_flash_api.c:128

0x4037caf9: esp_flash_read at C:/esp-idf/components/spi_flash/esp_flash_api.c:821

0x4203087d: esp_partition_read at C:/esp-idf/components/spi_flash/partition.c:415

0x42025a95: spiffs_api_read at C:/esp-idf/components/spiffs/spiffs_api.c:36

0x42029537: spiffs_phys_rd at C:/esp-idf/components/spiffs/spiffs/src/spiffs_cache.c:161

0x4202716f: spiffs_obj_lu_find_entry_visitor at C:/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:166

0x42028946: spiffs_object_find_object_index_header_by_name at C:/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:1694

0x42026084: SPIFFS_open at C:/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:230 (discriminator 2)

0x42025865: vfs_spiffs_open at C:/esp-idf/components/spiffs/esp_spiffs.c:479

0x420039b9: esp_vfs_open at C:/esp-idf/components/vfs/vfs.c:407 (discriminator 3)

0x420519c6: _fopen_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fopen.c:129

0x42051a69: fopen at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdio/fopen.c:168

0x420082aa: get_file_length at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc_1/build/../components/hufzip/hufzip_model.c:80

0x42008452: dl_matrixq_copy_to_psram_spiffs at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc_1/build/../components/hufzip/hufzip_model.c:232

0x42016614: model_create at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc_1/build/../components/multinet/multinet4_single_quantized_cn.c:795

0x42016977: model_init at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc_1/build/../components/multinet/multinet4_single_quantized_cn.c:1826

0x42006fdf: detect_Task at c:\esp-skainet\examples\en_speech_commands_recognition\build/../main/main.c:75

0x40384cd6: vPortTaskWrapper at C:/esp-idf/components/freertos/port/xtensa/port.c:130
```

LiTongXue98, 2022-03-07:
@feizi 请问一下，我发现的这个问题，你们测试的时候有发现吗，是我的工程配置有问题吗😥

feizi, 2022-03-08:
你好，有加密模组的详细型号吗？

LiTongXue98, 2022-03-08:
esp32-s3-wroom-1   D2N8R8(就是在官方淘宝店购买的esp32s3-korvo-2 v3.0)

LiTongXue98, 2022-03-08:
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01
秘钥设置的是这个

LiTongXue98, 2022-03-09:
@feizi 请问一下有什么新发现吗？运行NS/AEC/WAKENET都是OK的，multinet就失败了😂

LiTongXue98, 2022-03-15:
目前这个问题还困扰着我，我手上加密的esp32s3已经无法解密了，导致有一些程序测试在加密模组无法正常运行，在没有加密的模组就可以正常运行，我不知道是不是和我的menuconfig配置有关系

feizi, 2022-03-16:
你是怎样加密模组的，我们复现一下你的问题

LiTongXue98, 2022-03-18:
谷歌邮箱没我推送邮件，不好意思刚看到，我就是先烧写自定义的加密key,然后运行加密固件，后面烧写就直接把固件加密打包成加密固件进行烧写。<br/><br/>.\Flash_tool\python38\python.exe .\Flash_tool\esptool_py\esptool\espefuse.py --port COM1 burn_key BLOCK_KEY0 .\Flash_tool\esptool_py\esptool\espressif\efuse\my_key.bin XTS_AES_128_KEY --force-write-always<br/><br/>加密的key值<br/>00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01<html>
<head>
    <meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
</head>
<body>
<br/><br/><br/>        <a href="https://maas.mail.163.com/dashi-web-extend/html/proSignature.html?ftlId=1&name=litongxue98%40gmail.com&uid=litongxue98%40gmail.com&iconUrl=https%3A%2F%2Fmail-online.nosdn.127.net%2Fqiyelogo%2FdefaultAvatar.png&items=%5B%22%E9%82%AE%E7%AE%B1%EF%BC%9Alitongxue98%40gmail.com%22%5D" style="display:block;background:#fff; max-width: 400px; _width: 400px;padding:15px 0 10px 0;text-decoration: none; outline:none;-webkit-tap-highlight-color:transparent;-webkit-text-size-adjust:none !important;text-size-adjust:none !important;">
            <table cellpadding="0" style='width: 100%; max-width: 100%; table-layout: fixed; border-collapse: collapse;color: #9b9ea1;font-size: 14px;line-height:1.3;-webkit-text-size-adjust:none !important;text-size-adjust:none !important;'>
                <tbody style="font-family: 'PingFang SC', 'Hiragino Sans GB','WenQuanYi Micro Hei', 'Microsoft Yahei', '微软雅黑', verdana !important; word-wrap:break-word; word-break:break-all;-webkit-text-size-adjust:none !important;text-size-adjust:none !important;">
                    <tr>
                            <td width="38" style="padding:0; box-sizing: border-box; width: 38px;">
                                [image]
                            </td>
                            <td style='padding: 0 0 0 10px; color: #31353b;'>
                                <div style="font-size: 16px;font-weight:bold; width:100%; white-space: nowrap; overflow:hidden;text-overflow: ***@***.***</div>
                            </td>
                    </tr>
                        <tr width="100%" style="font-size: 14px !important; width: 100%;">
                            <td colspan='2' style="padding:10px 0 0 0; font-size:14px !important; width: 100%;">
                                    <div style="width: 100%;font-size: 14px ***@***.***</div>
                            </td>
                        </tr>
                </tbody>
            </table>
        </a><br/><br/><br/><div class="ntes-mailmaster-quote" style="padding-top: 1px; padding-bottom: 1px" >
    <div style=" margin-top: 2em; margin-bottom: 1em; font-size: 14px; line-height: 1.25; color: #78787a; " >---- 回复的原邮件 ----</div>
    <div style=" margin-bottom: 1em; font-size: 12px; line-height: 1.25; color: #232324; padding: 0.5em 0.25em; border-radius: 0.67em; background-color: #f0f0f0; " >
        <table width="100%" cellpadding="0" cellspacing="9" border="0">
            <tr>
                <td valign="top" style=" width: 4.25em; font-size: 12px; line-height: 1.25; color: #78787a; " >
                    发件人
                </td>
                <td valign="top" style=" font-size: 12px; line-height: 1.25; color: #232324; word-break: break-all; " >
                    <a class="mail-from" style="text-decoration:none;color:#0886e8;" ***@***.***">Sun ***@***.***&gt;</a>
                </td>
            </tr>
            <tr>
                <td valign="top" style=" width: 4.25em; font-size: 12px; line-height: 1.25; color: #78787a; " >
                    日期
                </td>
                <td class="mail-date" valign="top" style=" font-size: 12px; line-height: 1.25; color: #232324; word-break: break-all; " >
                    2022年03月16日 10:48
                </td>
            </tr>
            <tr style="">
                <td valign="top" style=" width: 4.25em; font-size: 12px; line-height: 1.25; color: #78787a; " >
                    收件人
                </td>
                <td valign="top" style=" font-size: 12px; line-height: 1.25; color: #232324; word-break: break-all; " >
                    <a class="mail-to" style="text-decoration:none;color:#0886e8;" ***@***.******@***.***&gt;</a>
                </td>
            </tr>
            <tr style="">
                <td valign="top" style=" width: 4.25em; font-size: 12px; line-height: 1.25; color: #78787a; " >
                    抄送至
                </td>
                <td valign="top" style=" font-size: 12px; line-height: 1.25; color: #232324; word-break: break-all; " >
                    <a class="mail-cc" style="text-decoration:none;color:#0886e8;" ***@***.******@***.***&gt;</a>、<a class="mail-cc" style="text-decoration:none;color:#0886e8;" ***@***.******@***.***&gt;</a>
                </td>
            </tr>
            <tr>
                <td valign="top" style=" width: 4.25em; font-size: 12px; line-height: 1.25; color: #78787a; " >
                    主题
                </td>
                <td class="mail-subject" valign="top" style=" font-size: 12px; line-height: 1.25; color: #232324; word-break: break-all; " >
                    Re: [espressif/esp-skainet] esp32-s3-korvo-2加密的模组multinet有问题 (AIS-854) (Issue #45)
                </td>
            </tr>
        </table>
    </div>
    <div><p></p>
<p dir="auto">你是怎样加密模组的，我们复现一下你的问题</p>

<p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">&mdash;<br />
<script type="application/ld+json">[
{
"target": "https://github.com/espressif/esp-skainet/issues/45#issuecomment-1068675859",
"url": "https://github.com/espressif/esp-skainet/issues/45#issuecomment-1068675859",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script></div>
</div>
</body>
</html>

LiTongXue98, 2022-03-30:
请问一下，有什么新发现吗，后面这个的esp-sr库会解决这个问题吗

feizi, 2022-04-08:
目前上海疫情，大家都居家隔离，暂时无法解决该问题。

LiTongXue98, 2022-04-09:
> 目前上海疫情，大家都居家隔离，暂时无法解决该问题。

好的，注意防护，有进展了也请您及时回复我

feizi, 2023-08-16:
如果有问题请重新提问

closed 2023-08-16 by feizi