espressif/esp-skainet #52: ADF和skainet的AEC效果不一致问题

LiTongXue98, 2022-05-05:
你好，我这边有测试在skainet上测试AEC的效果，发现并不是会消除的很彻底，我也做了一系列实验，利用相同的esp-sr的lib,利用同样的adf的NS单声道信号，发现ADF的参考信号采集的方式AEC效果会好一点，请问这两种方式对AEC的效果有很大影响吗，还是esp-sr需要严格的音频相位同步呢，请问有具体的要求吗

[https://github.com/espressif/esp-adf/issues/799](url)

feizi, 2022-05-07:
esp-sr里的AEC确实是需要参考信号的相位同步。
esp-skainet使用硬件电路回采的参考信号，在信号同步上做的更好，相应的性能也会更好，问题是需要额外的硬件电路来支持该功能  
你发现的ADF的性能更好一点，有可能是因为ADF上开启了NS，而esp-skainet上双麦是不使用NS的，你可以通过在esp-skainet的AEC输出加上NS,测试一下

LiTongXue98, 2022-05-07:
> 

我之前测试的方式就是在skainet上使用的NS的单麦处理，esp-sr也是同一git版本的lib

LiTongXue98, 2022-05-07:
我也有测试过保存adf上的音频数据，然后在skainet上feed到NS的算法，这样skainet AEC后的性能是和adf是一样的。

将skainet的数据保存，将数据通过algorithm stream 的接口传进去，adf上面的效果就和skainet上的一样差了，所以才提出了这个问题

feizi, 2022-05-17:
你用的是哪块开发板，我们测试一下．
现在上海都居家办公，复现这个问题会比较慢，非常抱歉．

LiTongXue98, 2022-05-17:
> 你用的是哪块板开发，我们测试一下。 现在上海都居家办公，复现这个问题会比较慢，非常抱歉。

esp32-s3-korvo-2 v3.0这块开发板，你可以看一下

closed 2022-07-23 by LiTongXue98