espressif/esp-skainet #54: 基于官方的2.0.3SDK,在Arduino上运行，能够唤醒，指令无法识别

tangjie133, 2022-06-10:
问题如题，基于官方的2.0.3SDK在Arduino上运行，唤醒词能够识别，但是指令词无法识别。
烧录如下：

> C:\Users\tangjie\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.3.0/esptool.exe --chip esp32s3 --port COM9 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 C:\Users\tangjie\AppData\Local\Temp\arduino_build_885412/wakeWordDetection.ino.bootloader.bin 0x8000 C:\Users\tangjie\AppData\Local\Temp\arduino_build_885412/wakeWordDetection.ino.partitions.bin 0x10000 C:\Users\tangjie\AppData\Local\Temp\arduino_build_885412/wakeWordDetection.ino.bin 0x281000 C:\Users\tangjie\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3/tools/sdk/esp32s3/bin/model.bin 
esptool.py v3.3
Serial port COM9
Connecting...
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 68:b6:b3:22:13:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x000b3fff...
Flash will be erased from 0x00281000 to 0x0078cfff...
Compressed 22208 bytes to 13805...
Writing at 0x00000000... (100 %)
Wrote 22208 bytes (13805 compressed) at 0x00000000 in 0.5 seconds (effective 391.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 121...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (121 compressed) at 0x00008000 in 0.0 seconds (effective 519.9 kbit/s)...
Hash of data verified.
Compressed 668992 bytes to 357539...
Writing at 0x00010000... (4 %)
Writing at 0x0001a52a... (9 %)
Writing at 0x00026c93... (13 %)
Writing at 0x00039cb1... (18 %)
Writing at 0x00040fbf... (22 %)
Writing at 0x00046ca1... (27 %)
Writing at 0x0004c49b... (31 %)
Writing at 0x00051469... (36 %)
Writing at 0x00056617... (40 %)
Writing at 0x0005b57b... (45 %)
Writing at 0x00060be1... (50 %)
Writing at 0x00066322... (54 %)
Writing at 0x0006bf47... (59 %)
Writing at 0x00073afa... (63 %)
Writing at 0x0007bab3... (68 %)
Writing at 0x0008304f... (72 %)
Writing at 0x00088715... (77 %)
Writing at 0x0008e8be... (81 %)
Writing at 0x00096967... (86 %)
Writing at 0x0009d95b... (90 %)
Writing at 0x000a6a68... (95 %)
Writing at 0x000ae736... (100 %)
Wrote 668992 bytes (357539 compressed) at 0x00010000 in 7.4 seconds (effective 720.6 kbit/s)...
Hash of data verified.
Compressed 5292032 bytes to 2127378...
Writing at 0x00281000... (0 %)
Writing at 0x00285ee9... (1 %)
Writing at 0x0028acc2... (2 %)
Writing at 0x0028fa56... (3 %)
Writing at 0x002948c0... (3 %)
Writing at 0x002996e4... (4 %)
Writing at 0x0029df59... (5 %)
Writing at 0x002a2802... (6 %)
Writing at 0x002a6f9e... (6 %)
Writing at 0x002ab833... (7 %)
Writing at 0x002b0133... (8 %)
Writing at 0x002b4a18... (9 %)
Writing at 0x002b93e4... (10 %)
Writing at 0x002bdce9... (10 %)
Writing at 0x002c26e8... (11 %)
[... 150 lines cut ...]
[0;32mI (139) esp_image: segment 2: paddr=00039844 vaddr=40374000 size=067d4h ( 26580) load[0m
[0;32mI (147) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=66a2ch (420396) map[0m

[16:43:00.246]收←◆[0;32mI (214) esp_image: segment 4: paddr=000a6a54 vaddr=4037a7d4 size=0cab0h ( 51888) load[0m
[0;32mI (225) esp_image: segment 5: paddr=000b350c vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (232) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (232) boot: Disabling RNG early entropy source...[0m
[   252][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Initializing SPIFFS

[16:43:00.633]收←◆Partition size: total: 4857101, used: 2629225

[16:43:00.679]收←◆model_name: xiaoaitongxue7q8 model_data: /srmodel/xiaoaitongxue7q8/wn7q8_data

[16:43:01.513]收←◆MC Quantized-8 wakeNet7: wakeNet7Q8_v1_xiaoaitongxue_5_0.97_0.90, mode:2, p:3, (Jan  5 2022 11:31:01)
Initial ONE-MIC auido front-end for speech recognition, mode:0, (Dec 10 2021 11:08:01)

[16:43:04.129]收←◆------------detect start------------

[16:44:06.602]收←◆wakeword detected
-----------LISTENING-----------

[16:44:17.908]收←◆
-----------awaits to be waken up-----------
请问这会是什么问题导致的呢？

feizi, 2022-06-15:
看你的log，Multinet(也就是命令词) 都没有初始化

feizi, 2022-06-15:
> 问题如题，基于官方的2.0.3SDK在Arduino上运行，唤醒词能够识别，但是指令词无法识别。 烧录如下：
>
>
> Leaving... Hard resetting via RTS pin... 运行
>
>
> [16:43:00.246]收←◆�[0;32mI (214) esp_image: segment 4: paddr=000a6a54 vaddr=4037a7d4 size=0cab0h [...]

> [16:43:01.513]收←◆MC Quantized-8 wakeNet7: wakeNet7Q8_v1_xiaoaitongxue_5_0.97_0.90, mode:2, p:3, (Jan 5 2022 11:31:01)
> Initial ONE-MIC auido front-end for speech recognition, mode:0, (Dec 10 2021 11:08:01)

feizi, 2022-06-15:
◆MC Quantized-8 wakeNet7: wakeNet7Q8_v1_xiaoaitongxue_5_0.97_0.90, mode:2, p:3, (Jan 5 2022 11:31:01) ，这个是唤醒词初始化话
Initial ONE-MIC auido front-end for speech recognition, mode:0, (Dec 10 2021 11:08:01)， 这个是音频前端初始化

closed 2022-06-29 by feizi

tangjie133, 2022-07-18:
将他配置为中文但是他去输出了英文得命令词

>
[image]