espressif/esp-skainet #55: 基于官方的SDK,在Arduino上运行，设置为中文但是却显示的英文命令，如下

tangjie133, 2022-07-19:
[image]

feizi, 2022-08-11:
请更新一下最新的master分支

feizi, 2022-08-23:
Please check the configuration you have set:
`
CONFIG_SR_WN_WN9_HILEXIN=y
CONFIG_USE_MULTINET=y
CONFIG_SR_MN_CHINESE=y
CONFIG_SR_MN_CN_MULTINET4_5_SINGLE_RECOGNITION=y
CONFIG_CN_SPEECH_COMMAND_ID0="da kai kong tiao"
CONFIG_CN_SPEECH_COMMAND_ID1="guan bi kong tiao"
CONFIG_CN_SPEECH_COMMAND_ID2="zeng da feng su"
CONFIG_CN_SPEECH_COMMAND_ID3="jian xiao feng su"
CONFIG_CN_SPEECH_COMMAND_ID4="sheng gao yi du"
CONFIG_CN_SPEECH_COMMAND_ID5="jiang di yi du"
CONFIG_CN_SPEECH_COMMAND_ID6="zhi re mo shi"
CONFIG_CN_SPEECH_COMMAND_ID7="zhi leng mo shi"
CONFIG_CN_SPEECH_COMMAND_ID8="jie neng mo shi"
CONFIG_CN_SPEECH_COMMAND_ID9="guan bi jie neng mo shi"
CONFIG_CN_SPEECH_COMMAND_ID10="chu shi mo shi"
CONFIG_CN_SPEECH_COMMAND_ID11="guan bi chu shi mo shi"
CONFIG_CN_SPEECH_COMMAND_ID12="shui mian mo shi"
CONFIG_CN_SPEECH_COMMAND_ID13="guan bi shui mian mo shi"
CONFIG_CN_SPEECH_COMMAND_ID14="ding shi yi xiao shi"
CONFIG_CN_SPEECH_COMMAND_ID15="ding shi liang xiao shi"
CONFIG_CN_SPEECH_COMMAND_ID16="zui da feng su,zui gao feng su"
`

closed 2023-08-16 by feizi