espressif/esp-skainet #56: esp32-korvo 1.1 运行cn_speech_commands_recognition例子错误

pighead1016, 2022-08-01:
使用的是esp32-korvo 1.1开发板。
除了这个例子还有什么其他的example能够启用mic 回声抑制的例子
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6612
load:0x40078000,len:14780
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (27) boot: compile time 15:09:43
I (27) boot: chip revision: 3
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 16MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 3, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=350ech (217324) map
I (188) esp_image: segment 1: paddr=00045114 vaddr=3ffb0000 size=025c0h (  9664) load
I (192) esp_image: segment 2: paddr=000476dc vaddr=40080000 size=0893ch ( 35132) load
I (209) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=5615ch (352604) map
I (337) esp_image: segment 4: paddr=000a6184 vaddr=4008893c size=07598h ( 30104) load
I (349) esp_image: segment 5: paddr=000ad724 vaddr=50000000 size=00010h (    16) load
I (357) boot: Loaded app from partition at offset 0x10000
I (357) boot: Disabling RNG early entropy source...
I (369) cpu_start: Pro cpu up.
I (370) cpu_start: Starting app cpu, entry point is 0x400814b8
0x400814b8: call_start_cpu1 at D:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/port/cpu_start.c:156
I (356) cpu_start: App cpu up.
I (384) cpu_start: Pro cpu start user code
I (384) cpu_start: cpu freq: 160000000
I (384) cpu_start: Application information:
I (388) cpu_start: Project name:     wake_word_detection
I (394) cpu_start: App version:      1
I (399) cpu_start: Compile time:     Aug  1 2022 15:08:58
I (405) cpu_start: ELF file SHA256:  71769bd431e63172...
I (411) cpu_start: ESP-IDF:          v4.4-dirty
I (416) heap_init: Initializing. RAM available for dynamic allocation:
I (423) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (429) heap_init: At 3FFB44D8 len 0002BB28 (174 KiB): DRAM
I (436) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (442) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (448) heap_init: At 4008FED4 len 0001012C (64 KiB): IRAM
I (456) spi_flash: detected chip: generic
I (459) spi_flash: flash io: dio
I (465) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (476) I2S: DMA Malloc info, datalen=blocksize=1280, dma_buf_count=6
I (486) I2S: I2S1, MCLK output by GPIO0
I (486) I2S: DMA Malloc info, datalen=blocksize=320, dma_buf_count=6
I (496) I2S: I2S0, MCLK output by GPIO0
I (496) board: Detected codec at 0x40. Name : ES7210
I (506) board: Detected codec at 0x18. Name : ES8311
I (516) ES7210: ES7210 in Slave mode
I (516) ES7210: Enable ES7210_INPUT_MIC1
I (526) ES7210: Enable ES7210_INPUT_MIC2
I (526) ES7210: Enable ES7210_INPUT_MIC3
I (526) ES7210: Enable ES7210_INPUT_MIC4
I (546) ES7210: The ES7210_CLOCK_OFF_REG01 value before stop is 0
I (546) ES7210: Enable ES7210_INPUT_MIC1
I (546) ES7210: Enable ES7210_INPUT_MIC2
I (546) ES7210: Enable ES7210_INPUT_MIC3
I (556) ES7210: Enable ES7210_INPUT_MIC4
I (566) DRV8311: ES8311 in Slave mode
I (576) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
Item psram alloc failed. Size: 2068 = 1024 x 2 + 16 + 4
Deprecated Quantized wakeNet5: wakeNet5_v1_hilexin_5_0.95_0.90, mode:0 (Dec 10 2021 20:15:15)
3
Item psram alloc failed. Size: 2068 = 2048 x 1 + 16 + 4
Initial ONE-MIC auido front-end for speech recognition, mode:1, (Dec 10 2021 20:29:24)
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d7d98  PS      : 0x00060c30  A0      : 0x800d6b4c  A1      : 0x3ffedcf0
0x400d7d98: afe_feed at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc/build/../components/esp_audio_front_end/esp_afe_sr_1mic.c:226 (discriminator 3)

A2      : 0x3ffbb2e4  A3      : 0x3ffeddb8  A4      : 0x00060023  A5      : 0x00000001
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x00000019  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00001000  A12     : 0x00000200  A13     : 0x3ffb79e0
A14     : 0x00000000  A15     : 0x00001000  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace:0x400d7d95:0x3ffedcf00x400d6b49:0x3ffedd10 0x4008ba49:0x3ffedd30
0x400d7d95: afe_feed at /home/wangwangwang/workspace/ai_vnc/728_support/esp_sr_lib_vnc/build/../components/esp_audio_front_end/esp_afe_sr_1mic.c:226 (discriminator 3)

0x400d6b49: feed_Task at d:\work\esp-skainet-master\examples\wake_word_detection\build/../main/main.c:40 (discriminator 1)

0x4008ba49: vPortTaskWrapper at D:/Espressif/frameworks/esp-idf-v4.4/components/freertos/port/xtensa/port.c:131

feizi, 2022-08-02:
看你的log，你使用的代码版本比较老，建议你使用最新的分支master 或者 dev/2.0

pighead1016, 2022-08-09:
最新分支里没有get_start的例子了啊

feizi, 2022-08-11:
你可以使用 en_speech_commands_recognition 或者 cn_speech_commands_recognition例子来代替之前get_started

closed 2023-08-16 by feizi