espressif/esp-skainet #58: Set i2s pins on custom board

ValeraDanger, 2022-08-14:
I am using TTGO T-camera [http://www.lilygo.cn/prod_view.aspx?TypeId=50030&Id=1273&FId=t3:50030:3](url) with I2S microphone. It's pinout:
- sck 26
- ws 32
- sdo 33

This module has ESP32 (not S3) chip, so i am using `sdkconfig.defaults.esp32`. I'm trying to change i2s pins. I've founded `esp-skainet\components\hardware_driver\boards\include\esp32_korvo_v1_1_board.h` file, where I changed `GPIO_I2S` block (not `GPIO_I2S0`, cause it seems to be second mic). This didn't work, so I disabled all useless blocks (change `FUNC_*func_name*_EN` parametrs to zero). Here you can see my actual full `esp32_korvo_v1_1_board.h` file:
```
/**
 * 
 * @copyright Copyright 2021 Espressif Systems (Shanghai) Co. Ltd.
 *
 *      Licensed under the Apache License, Version 2.0 (the "License");
 *      you may not use this file except in compliance with the License.
 *      You may obtain a copy of the License at
 *
 *               http://www.apache.org/licenses/LICENSE-2.0
 *
 *      Unless required by applicable law or agreed to in writing, software
 *      distributed under the License is distributed on an "AS IS" BASIS,
 *      WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 *      See the License for the specific language governing permissions and
 *      limitations under the License.
 */
#pragma once

#include "driver/gpio.h"

/**
 * @brief ESP32-KORVO-V1.1 I2C GPIO defineation
 * 
 */
#define FUNC_I2C_EN     (0)
#define GPIO_I2C_SCL    (GPIO_NUM_32)
#define GPIO_I2C_SDA    (GPIO_NUM_19)

/**
 * @brief ESP32-KORVO-V1.1 SDMMC GPIO defination
 * 
 * @note Only avaliable when PMOD connected
 */
#define FUNC_SDMMC_EN   (0)
#define SDMMC_BUS_WIDTH (1)
#define GPIO_SDMMC_CLK  (GPIO_NUM_5)
#define GPIO_SDMMC_CMD  (GPIO_NUM_18)
#define GPIO_SDMMC_D0   (GPIO_NUM_21)
#define GPIO_SDMMC_D1   (GPIO_NUM_NC)
#define GPIO_SDMMC_D2   (GPIO_NUM_NC)
#define GPIO_SDMMC_D3   (GPIO_NUM_23)
#define GPIO_SDMMC_DET  (GPIO_NUM_NC)

/**
 * @brief ESP32-KORVO-V1.1 SDSPI GPIO definationv
 * 
 */
#define FUNC_SDSPI_EN       (0)
#define SDSPI_HOST          (SPI2_HOST)
#define GPIO_SDSPI_CS       (GPIO_NUM_23)
#define GPIO_SDSPI_SCLK     (GPIO_NUM_5)
#define GPIO_SDSPI_MISO     (GPIO_NUM_21)
#define GPIO_SDSPI_MOSI     (GPIO_NUM_18)

/**
 * @brief ESP32-KORVO-V1.1 I2S GPIO defination
 * 
 */
#define FUNC_I2S_EN         (1)
#define GPIO_I2S_LRCK       (GPIO_NUM_32)
#define GPIO_I2S_MCLK       (GPIO_NUM_0)
#define GPIO_I2S_SCLK       (GPIO_NUM_26)
#define GPIO_I2S_SDIN       (GPIO_NUM_33)
#define GPIO_I2S_DOUT       (GPIO_NUM_NC)

/**
 * @brief ESP32-KORVO-V1.1 I2S GPIO defination
 * 
 */
#define FUNC_I2S0_EN         (0)
#define GPIO_I2S0_LRCK       (GPIO_NUM_22)
#define GPIO_I2S0_MCLK       (GPIO_NUM_0)
#define GPIO_I2S0_SCLK       (GPIO_NUM_25)
#define GPIO_I2S0_SDIN       (GPIO_NUM_NC)
#define GPIO_I2S0_DOUT       (GPIO_NUM_13)

/**
 * @brief ESP32-S3-HMI-DevKit power control IO
 * 
 * @note Some power control pins might not be listed yet
 * 
 */
#define FUNC_PWR_CTRL       (1)
#define GPIO_PWR_CTRL       (GPIO_NUM_12)
#define GPIO_PWR_ON_LEVEL   (1)

#define I2S0_CONFIG_DEFAULT() { \
    .mode                   = I2S_MODE_MASTER | I2S_MODE_TX, \
    .sample_rate            = sample_rate, \
    .bits_per_sample        = I2S_BITS_PER_SAMPLE_16BIT, \
    .channel_format         = channel_format, \
    .communication_format   = I2S_COMM_FORMAT_STAND_I2S, \
    .intr_alloc_flags       = ESP_INTR_FLAG_LEVEL1, \
    .dma_buf_count          = 6, \
    .dma_buf_len            = 160, \
    .use_apll               = false, \
    .tx_desc_auto_clear     = true, \
    .fixed_mclk             = 0, \
    .mclk_multiple          = I2S_MCLK_MULTIPLE_DEFAULT, \
    .bits_per_chan          = bits_per_chan, \
}

#define I2S_CONFIG_DEFAULT() { \
    .mode                   = I2S_MODE_MASTER | I2S_MODE_RX, \
    .sample_rate            = sample_rate, \
    .bits_per_sample        = I2S_BITS_PER_SAMPLE_32BIT, \
    .channel_format         = I2S_CHANNEL_FMT_RIGHT_LEFT, \
    .communication_format   = I2S_COMM_FORMAT_STAND_I2S, \
    .intr_alloc_flags       = ESP_INTR_FLAG_LEVEL1, \
    .dma_buf_count          = 6, \
    .dma_buf_len            = 160, \
    .use_apll               = false, \
    .tx_desc_auto_clear     = true, \
    .fixed_mclk             = 0, \
    .mclk_multiple          = I2S_MCLK_MULTIPLE_DEFAULT, \
    .bits_per_chan          = I2S_BITS_PER_CHAN_32BIT, \
}
```

This didn't help too. I appended `printf(i2s_buff);` in `\esp-skainet\examples\wake_word_detection\main\main.c` file in `feed_Task` function. to see, is there any data from my mic. . There is this func's code:
```
void feed_Task(void *arg)
{
    esp_afe_sr_data_t *afe_data = arg;
    int audio_chunksize = afe_handle->get_feed_chunksize(afe_data);
    int nch = afe_handle->get_channel_num(afe_data);
    int feed_channel = esp_get_feed_channel();
    int16_t *i2s_buff = malloc(audio_chunksize * sizeof(int16_t) * feed_channel);
    assert(i2s_buff);
    size_t bytes_read;

    while (1) {
        esp_get_feed_data(i2s_buff, audio_chunksize * sizeof(int16_t) * feed_channel);

        printf(i2s_buff);
        afe_handle->feed(afe_data, i2s_buff);    
    }
    afe_handle->destroy(afe_data);
    vTaskDelete(NULL);
}

``` 
But monitor is empty 
[image]
So, there is not any data from my microphone, 
Also, I've noticed, this warning 
[image]
I think, the problem is the absence of any codec on my board (before, I took data directly from mic by DMA (this sample worked for me [https://github.com/atomic14/esp32-i2s-mic-test](url)  ))

How can I tune this code to force it working on this board?

feizi, 2022-08-15:
I think you should refer to [esp32-s3-eye board setting](https://github.com/espressif/esp-skainet/blob/master/components/hardware_driver/boards/esp32s3-eye/bsp_board.c) to modify your board.

ValeraDanger, 2022-08-15:
I have renamed `sdkconfig.defaults.esp32s3eye` to `sdkconfig` and tryied to flash my board. While flashing, I tooked this error, cause my chip is not S3.
[image]

> I think you should refer to [esp32s3-eye board setting](https://github.com/espressif/esp-skainet/blob/master/components/hardware_driver/boards/esp32s3-eye/bsp_board.c) to modify your board.
> 
> [ESP32-S3-EYE_Getting_Started_Guide](https://github.com/espressif/esp-who/blob/master/docs/en/get-started/ESP32-S3-EYE_Getting_Started_Guide.md)

feizi, 2022-08-23:
I mean you can refer to https://github.com/espressif/esp-skainet/blob/master/components/hardware_driver/boards/esp32s3-eye/bsp_board.c to modify I2S setting.
The microphone of esp32_korvo_v1_1 is analog microphone.
The microphone of esp32s3-sys is digital microphone and I guess your mircophone is digital microphone too.

closed 2023-08-16 by feizi