espressif/esp-skainet #60: chinese_tts的some questions

casitelwithdean, 2022-08-28:
1.readme中提及“##### 2)使用该目录下的flash_voicedata.sh脚本，烧写voice data到指定分区:  ”使用的是“source flash_voicedata.sh ../../components/esp-sr/esp-tts/esp_tts_chinese/esp_tts_voice_data_xiaoxin.dat  /dev/ttyUSB0”这个命令，但是我使用的windows的idf-cmd，请问windows下应该使用什么命令烧录这个.dat文件呢？
2.es8311我是完全仿照着esp32-kovro的原理图自己画的一块小板子，然后用worver-e开发板测试，同样也是采用korvo的接口，当我只插上SCL和SDA的用来测试I2C接口的时候，es8311的初始化非常正常，当我再插上I2S的MCLK或者SCLK时，es8311的初始化就会报错，报错总是在bsp_borad.c  里的bsp_codec_dac_init（）函数。我追根溯源，发现在 es8311.c文件里   ret |= es8311_write_reg(ES8311_SDPIN_REG09, dac_iface); 我打印出ret，查询错误表对应的0x107 也就是operation time_out。我最初以为板子焊接的问题，于是我又打了两次板子，焊接了五个板子，都是同样的情况，困扰我半个月之久，还请帮忙解答，谢谢！

feizi, 2022-08-30:
@casitelwithdean ， 我不是很清楚 es8311 codec的问题，你可以在ESP-ADF中问该问题，他们更熟悉关于es8311的配置问题

casitelwithdean, 2022-08-30:
> @casitelwithdean ， 我不是很清楚 es8311 codec的问题，你可以在ESP-ADF中问该问题，他们更熟悉关于es8311的配置问题

好的谢谢

closed 2022-08-30 by casitelwithdean