espressif/esp-skainet #69: cn_speech_commands_recognition running with errors.

babadue, 2022-10-28:
The en_speech_commands_recognition worked fine.  It recognized "hi esp" and subsequent commands.

The cn_speech_commands_recognition compiled ok.  But when running, it will recognize the "hi lexin".  But then after that it just was not waiting for commands.  Here is log with errors from monitor:

------------detect start------------
WAKEWORD DETECTED
AFE_FETCH_CHANNEL_VERIFIED, channel index: 2

assert failed: dl_matrixq_slice dl_lib_matrixq.c:483 (in->flags & DL_MF_FOREIGNDATA)

Backtrace:0x40379e7a:0x3fceb2c00x40389991:0x3fceb2e0 0x40390001:0x3fceb300 0x4202053e:0x3fceb420 0x42022082:0x3fceb440 0x420210c2:0x3fceb4a0 0x4201462f:0x3fceb4d0 0x42014caf:0x3fceb500 0x42006e42:0x3fceb520 0x4038ca3d:0x3fceb550
0x40379e7a: panic_abort at C:/Users/uuumm/esp/esp-idf/components/esp_system/panic.c:402

0x40389991: esp_system_abort at C:/Users/uuumm/esp/esp-idf/components/esp_system/esp_system.c:128

0x40390001: __assert_func at C:/Users/uuumm/esp/esp-idf/components/newlib/assert.c:85

0x4202053e: dl_matrixq_slice at /home/sunxiangyu/aishare/master/esp_sr_lib/build/../components/dl_lib/dl_lib_matrixq.c:483

0x42022082: dl_convq16_lstm_layer_s3 at /home/sunxiangyu/aishare/master/esp_sr_lib/build/../components/dl_lib/dl_lib_convq_queue_s3.c:960 (discriminator 2)

0x420210c2: dl_convq16_lstm_layer at /home/sunxiangyu/aishare/master/esp_sr_lib/build/../components/dl_lib/dl_lib_convq_queue.c:620

0x4201462f: multinet4_model_run_step2 at /home/sunxiangyu/aishare/master/esp_sr_lib/build/../components/multinet/multinet4_single_quantized_cn.c:1141

0x42014caf: model_detect at /home/sunxiangyu/aishare/master/esp_sr_lib/build/../components/multinet/multinet4_single_quantized_cn.c:1607

0x42006e42: detect_Task at C:/Users/uuumm/cn_speech_commands_recognition/main/main.c:115

0x4038ca3d: vPortTaskWrapper at C:/Users/uuumm/esp/esp-idf/components/freertos/port/xtensa/port.c:131

feizi, 2022-10-31:
Hi @babadue,
The bug has been fixed. Please update esp-sr commponent.

babadue, 2022-11-01:
> Hi @babadue,
> The bug has been fixed. Please update esp-sr commponent.

谢了！

closed 2022-11-01 by babadue