espressif/esp-skainet #71: ESP32-Korvo V1.1 运行cn_speech_commands_recognition，watchdog复位问题

zxg623, 2022-11-14:
Executing action: monitor
Running idf_monitor in directory d:\stm32f030f4\esp\esp-idf\esp-skainet\esp-skainet\examples\cn_speech_commands_recognition
Executing "C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe D:\STM32F030F4\esp\esp-idf\esp-skainet\esp-skainet\esp-idf\tools/idf_monitor.py -p \\.\COM15 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 --print_filter  d:\stm32f030f4\esp\esp-idf\esp-skainet\esp-skainet\examples\cn_speech_commands_recognition\build\speech_commands_recognition.elf -m 'C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe' 'D:\STM32F030F4\esp\esp-idf\esp-skainet\esp-skainet
\esp-idf\tools\idf.py'"...
?[0;33m--- idf_monitor on \\.\COM15 115200 ---?[0m
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7048
load:0x40078000,len:15544
load:0x40080400,len:4492
0x40080400: _init at ??:?

entry 0x400806b4
I (27) boot: ESP-IDF v4.4-beta1-2-gdf0929009a 2nd stage bootloader
I (27) boot: compile time 08:38:06
I (27) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32: SPI Speed      : 80MHz
I (49) boot.esp32: SPI Mode       : QIO
I (53) boot.esp32: SPI Flash Size : 16MB
I (58) boot: Enabling RNG early entropy source...
I (63) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 factory          factory app      00 00 00010000 00500000
I (82) boot: End of partition table
I (86) boot_comm: chip revision: 3, min. application chip revision: 0
I (93) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1d5df4h (1924596) map
I (632) esp_image: segment 1: paddr=001e5e1c vaddr=3ffb0000 size=02d84h ( 11652) load
I (635) esp_image: segment 2: paddr=001e8ba8 vaddr=40080000 size=07470h ( 29808) load
I (647) esp_image: segment 3: paddr=001f0020 vaddr=400d0020 size=65e04h (417284) map
I (762) esp_image: segment 4: paddr=00255e2c vaddr=40087470 size=0ed64h ( 60772) load
I (782) esp_image: segment 5: paddr=00264b98 vaddr=50000000 size=00010h (    16) load
I (793) boot: Loaded app from partition at offset 0x10000
I (793) boot: Disabling RNG early entropy source...
D (805) efuse: In EFUSE_BLK0__DATA3_REG is used 3 bits starting with 9 bit
D (805) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 2 bit
I (807) psram: This chip is ESP32-D0WD
I (813) spiram: Found 64MBit SPI RAM device
I (817) spiram: SPI RAM mode: flash 80m sram 80m
I (822) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (829) cpu_start: Pro cpu up.
D (833) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 15 bit
D (840) efuse: In EFUSE_BLK0__DATA5_REG is used 1 bits starting with 20 bit
D (847) efuse: In EFUSE_BLK0__DATA3_REG is used 3 bits starting with 9 bit
D (854) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 2 bit
I (861) cpu_start: Starting app cpu, entry point is 0x40081830
0x40081830: call_start_cpu1 at D:/STM32F030F4/esp/esp-idf/esp-skainet/esp-skainet/esp-idf/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (1363) spiram: SPI SRAM memory test OK
D (1371) clk: RTC_SLOW_CLK calibration value: 3445542
I (1376) cpu_start: Pro cpu start user code
I (1376) cpu_start: cpu freq: 240000000
I (1376) cpu_start: Application information:
I (1381) cpu_start: Project name:     speech_commands_recognition
I (1388) cpu_start: App version:      v0.3.0-137-gaa36390-dirty
I (1395) cpu_start: Compile time:     Nov 14 2022 08:38:09
I (1401) cpu_start: ELF file SHA256:  b0e425b1cd0300b8...
I (1407) cpu_start: ESP-IDF:          v4.4-beta1-2-gdf0929009a
D (1413) memory_layout: Checking 8 reserved memory ranges:
D (1419) memory_layout: Reserved memory range 0x3f800000 - 0x3fc00000
D (1425) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (1432) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb51d0
D (1438) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
[... 810 lines cut ...]
W (24692) AFE_SR: ERROR! rb_out slow!!!

W (24702) AFE_SR: ERROR! rb_out slow!!!

W (24802) AFE_SR: ERROR! rb_out slow!!!

W (24842) AFE_SR: ERROR! rb_out slow!!!

W (24852) AFE_SR: ERROR! rb_out slow!!!

W (24862) AFE_SR: ERROR! rb_out slow!!!

W (24872) AFE_SR: ERROR! rb_out slow!!!

W (24972) AFE_SR: ERROR! rb_out slow!!!

W (25002) AFE_SR: ERROR! rb_out slow!!!

W (25022) AFE_SR: ERROR! rb_out slow!!!

W (25032) AFE_SR: ERROR! rb_out slow!!!

W (25042) AFE_SR: ERROR! rb_out slow!!!

-----------awaits to be waken up-----------

feizi, 2022-11-15:
ESP32-S3 was recommended to run speech commands recognition.

zxg623, 2022-11-15:
<html><body>
<h2 dir="auto">硬件准备</h2>
<p dir="auto">为了运行 ESP-Skainet，您需要一块集成了音频输入模块的开发板。
开发板支持：</p>

Example Name | Latest Models | Supported Board
-- | -- | --
cn_speech_commands_recognition | Multinet4_5 | ESP32-Korvo, ESP32-S3-Korvo-1, ESP-BOX, ESP-S3-Korvo-2, ESP32-S3-EYE
en_speech_commands_recognition | Multinet5 | ESP32-S3-Korvo-1, ESP-BOX, ESP-S3-Korvo-2, ESP32-S3-EYE
wake_word_detection | Wakenet8 | ESP32-Korvo, ESP32-S3-Korvo-1, ESP-BOX, ESP-S3-Korvo-2, ESP32-S3-EYE
chinese_tts | esp-tts-v1 | ESP32-Korvo, ESP32-S3-Korvo-1, ESP-BOX, ESP-S3-Korvo-2

closed 2023-08-16 by feizi