espressif/esp-skainet #72: English speech command not working on ESP32S3 eye.

nikhil-robinson, 2022-11-15:
Hello there. I was testing the English speech command recognition example on the esp32s3 eye module. It is not able to detect any wake words or the commands. Its stuck with the below output.
```

`I (106) cpu_start: Starting app cpu, entry point is 0x403ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x400570ec
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x470
load:0x403c9700,len:0xc04
load:0x403cc700,len:0x2bc8
entry 0x403c98e8
I (33) opi psram: vendor id : 0x0d (AP)
I (33) opi psram: dev id    : 0x02 (generation 3)
I (33) opi psram: density   : 0x03 (64 Mbit)
I (36) opi psram: good-die  : 0x01 (Pass)
I (41) opi psram: Latency   : 0x01 (Fixed)
I (46) opi psram: VCC       : 0x01 (3V)
I (50) opi psram: SRF       : 0x01 (Fast Refresh)
I (56) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (61) opi psram: BurstLen  : 0x01 (32 Byte)
I (66) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (72) opi psram: DriveStrength: 0x00 (1/1)
W (77) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (87) spiram: Found 64MBit SPI RAM device
I (91) spiram: SPI RAM mode: sram 80m
I (96) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (103) cpu_start: Pro cpu up.
I (106) cpu_start: Starting app cpu, entry point is 0x40379424
0x40379424: call_start_cpu1 at /home/bibox/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (402) spiram: SPI SRAM memory test OK
I (410) cpu_start: Pro cpu start user code
I (410) cpu_start: cpu freq: 240000000
I (411) cpu_start: Application information:
I (411) cpu_start: Project name:     speech_commands_recognition
I (412) cpu_start: App version:      403ae61-dirty
I (413) cpu_start: Compile time:     Nov 15 2022 13:23:19
I (414) cpu_start: ELF file SHA256:  66d165ff0240f4f2...
I (414) cpu_start: ESP-IDF:          v4.4.2-388-g755ce1077d
I (415) heap_init: Initializing. RAM available for dynamic allocation:
I (416) heap_init: At 3FCA81E0 len 00041530 (261 KiB): D/IRAM
I (416) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (417) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (418) spi_flash: detected chip: gd
I (418) spi_flash: flash io: qio
I (419) sleep: Configure to isolate all GPIO pins in sleep state
I (419) sleep: Enable automatic switching of GPIO sleep configuration
I (420) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (421) spiram: Reserving pool of 16K of internal memory for DMA/internal allocations
I (421) MODEL_LOADER: 
Initializing models from SPIFFS, partition label: model

I (741) MODEL_LOADER: Partition size: total: 4857101, used: 2490422

I (1011) I2S: DMA Malloc info, datalen=blocksize=640, dma_buf_count=6
I (1011) AFE_SR: afe interface for speech recognition

I (1011) AFE_SR: AFE version: SR_V220727

I (1011) AFE_SR: Initial auido front-end, total channel: 3, mic num: 2, ref num: 1

I (1011) AFE_SR: aec_init: 0, se_init: 1, vad_init: 1

I (1011) AFE_SR: wakenet_init: 1

model_name: wn9_hiesp model_data: /srmodel/wn9_hiesp/wn9_data
MC Quantized wakenet9: wakeNet9_v1h24_hiesp_3_0.63_0.635, tigger:v3, mode:2, p:0, (Oct  9 2022 17:21:08)
I (1751) AFE_SR: wake num: 3, mode: 0, (Oct 12 2022 19:13:45)

multinet:mn5q8_en
Quantized8 Multinet5: MN5Q8_v2_english_8_0.9_0.90, (Oct 12 2022 19:26:00)
esp_mn_commands_update_from_sdkconfig
I (5171) MN_COMMAND: ---------------------SPEECH COMMANDS---------------------
I (5171) MN_COMMAND: Command ID0, phrase ID0: hi PmSm
I (5171) MN_COMMAND: ---------------------------------------------------------

W (5171) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

------------detect start------------
```
`

Some thing i notice is.

why is it showing 2 mics when there is only one mic in esp32s3
`I (1011) AFE_SR: Initial auido front-end, total channel: 3, mic num: 2, ref num: 1`

And i am also getting this warning some times.

`W (5171) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!`

nikhil-robinson, 2022-11-15:
@feizi  any suggestions ?

lijunru-hub, 2022-11-21:
Your problem is that the configuration you set does not match your hardware.

You need to modify the configuration of afe_handle, total channel: 1, mic num: 1, ref num: 0.
Hardware schematic diagram:https://dl.espressif.com/dl/schematics/SCH_ESP32-S3-EYE-MB_20211201_V2.2.pdf
You can refer to esp-who:esp32-s3-eye example:Sr libraries will be older, and some api will be different.

nikhil-robinson, 2022-11-21:
@lijunru-hub i tried modifying the mic configuration but still getting this.

```
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x470
load:0x403c9700,len:0xc04
load:0x403cc700,len:0x2b40
entry 0x403c98e8
I (33) opi psram: vendor id : 0x0d (AP)
I (34) opi psram: dev id    : 0x02 (generation 3)
I (34) opi psram: density   : 0x03 (64 Mbit)
I (37) opi psram: good-die  : 0x01 (Pass)
I (42) opi psram: Latency   : 0x01 (Fixed)
I (47) opi psram: VCC       : 0x01 (3V)
I (51) opi psram: SRF       : 0x01 (Fast Refresh)
I (56) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (62) opi psram: BurstLen  : 0x01 (32 Byte)
I (67) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (73) opi psram: DriveStrength: 0x00 (1/1)
W (77) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (88) spiram: Found 64MBit SPI RAM device
I (92) spiram: SPI RAM mode: sram 80m
I (97) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (104) cpu_start: Pro cpu up.
I (107) cpu_start: Starting app cpu, entry point is 0x40379464
0x40379464: call_start_cpu1 at /home/nikhil/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (402) spiram: SPI SRAM memory test OK
I (411) cpu_start: Pro cpu start user code
I (411) cpu_start: cpu freq: 240000000
I (411) cpu_start: Application information:
I (411) cpu_start: Project name:     speech_commands_recognition
I (412) cpu_start: App version:      v0.3.0-137-gaa36390-dirty
I (412) cpu_start: Compile time:     Nov 21 2022 13:18:06
I (412) cpu_start: ELF file SHA256:  80ecfc929b673e6a...
I (412) cpu_start: ESP-IDF:          v4.4.3-145-g76caa64286
I (413) heap_init: Initializing. RAM available for dynamic allocation:
I (413) heap_init: At 3FCA8788 len 00040F88 (259 KiB): D/IRAM
I (413) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (413) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (414) spi_flash: detected chip: gd
I (414) spi_flash: flash io: qio
I (415) sleep: Configure to isolate all GPIO pins in sleep state
I (415) sleep: Enable automatic switching of GPIO sleep configuration
I (416) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (416) spiram: Reserving pool of 16K of internal memory for DMA/internal allocations
I (416) MODEL_LOADER: 
Initializing models from SPIFFS, partition label: model

I (736) MODEL_LOADER: Partition size: total: 4857101, used: 2490422

I (1006) I2S: DMA Malloc info, datalen=blocksize=640, dma_buf_count=6
I (1006) AFE_SR: afe interface for speech recognition

I (1006) AFE_SR: AFE version: SR_V220727

I (1006) AFE_SR: Initial auido front-end, total channel: 1, mic num: 1, ref num: 0

I (1006) AFE_SR: aec_init: 0, se_init: 1, vad_init: 1

I (1006) AFE_SR: wakenet_init: 1

model_name: wn9_hiesp model_data: /srmodel/wn9_hiesp/wn9_data
MC Quantized wakenet9: wakeNet9_v1h24_hiesp_3_0.63_0.635, tigger:v3, mode:2, p:0, (Oct  9 2022 17:21:08)
I (1676) AFE_SR: wake num: 2, mode: 0, (Oct 12 2022 19:13:45)

multinet:mn5q8_en
[... 401 lines cut ...]

W (4956) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

W (4966) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

W (4986) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

W (4996) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

W (5016) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

W (5026) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

W (5046) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

Quantized8 Multinet5: MN5Q8_v2_english_8_0.9_0.90, (Oct 12 2022 19:26:00)
esp_mn_commands_update_from_sdkconfig
I (5056) MN_COMMAND: ---------------------SPEECH COMMANDS---------------------
I (5056) MN_COMMAND: Command ID0, phrase ID0: hi PmSm
I (5056) MN_COMMAND: Command ID6, phrase ID1: LbcST VnLYoM
I (5056) MN_COMMAND: ---------------------------------------------------------

------------detect start------------
W (5066) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!
```

nikhil-robinson, 2022-11-21:
can any one point to an old commit of esp skainet which have the one mic support aec.

feizi, 2022-11-22:
The latest version also supports single mic AEC

nikhil-robinson, 2022-11-22:
Okay, great !

closed 2022-11-23 by feizi