espressif/esp-skainet #83: Help with ESP32 Korvo 1 v5.0

imdahsiaria90, 2022-12-30:
I newly bought the ESP32 Korvo-1 v5.0 development board. I have installed the latest release from [https://github.com/espressif/esp-skainet]

I flashed the example "en_speech_commands_recognition"

Observation/ Question:

1. There is hardly any documentation on the example. Where can I find this ?
2. How do I control the Leds ?
3. Audio does not seem to work. I have tried with both the headphones and speaker.
4. Is there any example on how to use the multinet commands in the code once they are detected ? 
5. Where can I find the source code of the default firmware ?

Help would highly be appreciated.

Sandra-lol, 2023-01-04:
Hi, for en speech command recog, ESP32-S3 boards are recommended:
[image]

imdahsiaria90, 2023-01-04:
Hello Sandra,

As mentioned in my description, I am using ESP32-S3-Korvo-1 v5.0.

feizi, 2023-01-04:
Hi imdahsiaria90,
en_speech_commands_recognition example do not play any audio.
If you want play audio, you can refer to cn_speech_commands_recognition or [esp-box/factory_demo](https://github.com/espressif/esp-box/tree/master/examples/factory_demo)

imdahsiaria90, 2023-01-04:
Hi Feizi.

OK thank you for the feedback. Atleast that solves for my question no 2.

Can you also provide any feedback on other questions ?

feizi, 2023-08-16:
Please create new one if you have

closed 2023-08-16 by feizi