espressif/esp-skainet #88: 唤醒未结束中途再次唤醒，timeout的时间以第一次为开始计算，而非再次唤醒后计算。

casitelwithdean, 2023-02-24:
Hi,我设置的listen时长为10s，当在7-8s左右时我再次唤醒时，马上就会timeout退出，我想实现的是，就算中途唤醒，是在中途唤醒的10s后再timeout退出。请问有什么方法可以实现吗？

feizi, 2023-02-27:
这应该是个bug,你用的是哪个模型?

casitelwithdean, 2023-02-27:
> 这应该是个bug,你用的是哪个模型?

小爱同学wn9  
Mn4_cn
目前是如果中途检测到了指令的话，那么会重新计时。如果中途检测到了唤醒词的话，并不会重新计时。

feizi, 2023-02-27:
你试试 mn5q8_cn

casitelwithdean, 2023-02-27:
> 你试试 mn5q8_cn

Hi,我试了mn5q8_cn，但是会导致重启。如下为日志。

feizi, 2023-02-28:
I will check it

feizi, 2023-02-28:
我没有遇到你这个问题，请更新一下最新的 [esp-sr](https://github.com/espressif/esp-sr)

feizi, 2023-02-28:
预计这周五，我们会添加Chinese Multinet6 会有更好的性能，到时你也可以试试。

casitelwithdean, 2023-02-28:
> 预计这周五，我们会添加Chinese Multinet6 会有更好的性能，到时你也可以试试。

好的，我会更新后再次尝试，非常感谢您耐心的解答，多谢！

casitelwithdean, 2023-03-01:
> 我没有遇到你这个问题，请更新一下最新的 [esp-sr](https://github.com/espressif/esp-sr)

Hi,我找到mn5q8_cn不断重启的问题了，原因是sdkconfig指令中 bo fang，zan ting lan ya的逗号误写成了中文的，因此导致报错重启（不过mn4倒是不会重启）

casitelwithdean, 2023-03-06:
> 预计这周五，我们会添加Chinese Multinet6 会有更好的性能，到时你也可以试试。

[hi,我下载了最新的esp-sr,但是当我选择mn6_cn之后，add_chinese_command的menuconfig不见了，mn5_cn和mn4_cn都会显示，请问该怎么处理？
[image]

feizi, 2023-03-07:
multinet6换了一种方法去定义命令词，可以看 [tool/README.md](https://github.com/espressif/esp-sr/tree/master/tool)

closed 2023-03-08 by casitelwithdean