espressif/esp-skainet #89: Build wake_word_detection example fails

waruqi, 2023-03-27:
I use esp-idf 5.1 to build it, but it fails.

```
idf.py set-target esp32s3
idf.py build
```

```
ruki-2:wake_word_detection ruki$ idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Executing action: set-target
Set Target to: esp32s3, new sdkconfig will be created.
Running cmake in directory /Users/ruki/projects/personal/esp-skainet/examples/wake_word_detection/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/ruki/.espressif/python_env/idf5.1_py3.10_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /Users/ruki/projects/personal/esp-skainet/examples/wake_word_detection"...
-- Existing sdkconfig '/Users/ruki/projects/personal/esp-skainet/examples/wake_word_detection/sdkconfig' renamed to '/Users/ruki/projects/personal/esp-skainet/examples/wake_word_detection/sdkconfig.old'.
-- Found Git: /usr/bin/git (found version "2.37.1 (Apple Git-137.1)") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/ruki/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/ruki/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/ruki/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Solving dependencies requirements
...Updating lock file at /Users/ruki/projects/personal/esp-skainet/examples/wake_word_detection/dependencies.lock
Processing 2 dependencies:
[1/2] espressif/esp-dsp (1.3.0)
[2/2] idf (5.1.0)
-- Project sdkconfig file /Users/ruki/projects/personal/esp-skainet/examples/wake_word_detection/sdkconfig
Loading defaults file /Users/ruki/projects/personal/esp-skainet/examples/wake_word_detection/sdkconfig.defaults...
Loading defaults file /Users/ruki/projects/personal/esp-skainet/examples/wake_word_detection/sdkconfig.defaults.esp32s3...
/Users/ruki/projects/personal/esp-skainet/examples/wake_word_detection/sdkconfig.defaults.esp32s3:9 CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240 was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240
/Users/ruki/projects/personal/esp-skainet/examples/wake_word_detection/sdkconfig.defaults.esp32s3:13 CONFIG_ESP32S3_SPIRAM_SUPPORT was replaced with CONFIG_SPIRAM
Compiler supported targets: xtensa-esp32s3-elf

-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /Users/ruki/.espressif/python_env/idf5.1_py3.10_env/bin/python (found version "3.10.8") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "wake_word_detection" version: v0.3.0-173-g5abc25b
-- Adding linker script /Users/ruki/projects/personal/esp-skainet/examples/wake_word_detection/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/ruki/projects/personal/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script /Users/ruki/projects/personal/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /Users/ruki/projects/personal/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /Users/ruki/projects/personal/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /Users/ruki/projects/personal/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /Users/ruki/projects/personal/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /Users/ruki/projects/personal/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-sr esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__esp-dsp esptool_py fatfs freertos hal hardware_driver heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon player protobuf-c protocomm pthread sdmmc soc spi_flash spiffs sr_ringbuf tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Users/ruki/projects/personal/esp-idf/components/app_trace /Users/ruki/projects/personal/esp-idf/components/app_update /Users/ruki/projects/personal/esp-idf/components/bootloader /Users/ruki/projects/personal/esp-idf/components/bootloader_support /Users/ruki/projects/personal/esp-idf/components/bt /Users/ruki/projects/personal/esp-idf/components/cmock /Users/ruki/projects/personal/esp-idf/components/console /Users/ruki/projects/personal/esp-idf/components/cxx /Users/ruki/projects/personal/esp-idf/components/driver /Users/ruki/projects/personal/esp-idf/components/efuse /Users/ruki/projects/personal/esp-skainet/components/esp-sr /Users/ruki/projects/personal/esp-idf/components/esp-tls /Users/ruki/projects/personal/esp-idf/components/esp_adc /Users/ruki/projects/personal/esp-idf/components/esp_app_format /Users/ruki/projects/personal/esp-idf/components/esp_coex /Users/ruki/projects/personal/esp-idf/components/esp_common /Users/ruki/projects/personal/esp-idf/components/esp_eth /Users/ruki/projects/personal/esp-idf/components/esp_event /Users/ruki/projects/personal/esp-idf/components/esp_gdbstub /Users/ruki/projects/personal/esp-idf/components/esp_hid /Users/ruki/projects/personal/esp-idf/components/esp_http_client /Users/ruki/projects/personal/esp-idf/components/esp_http_server /Users/ruki/projects/personal/esp-idf/components/esp_https_ota /Users/ruki/projects/personal/esp-idf/components/esp_https_server /Users/ruki/projects/personal/esp-idf/components/esp_hw_support /Users/ruki/projects/personal/esp-idf/components/esp_lcd /Users/ruki/projects/personal/esp-idf/components/esp_local_ctrl /Users/ruki/projects/personal/esp-idf/components/esp_mm /Users/ruki/projects/personal/esp-idf/components/esp_netif /Users/ruki/projects/personal/esp-idf/components/esp_netif_stack /Users/ruki/projects/personal/esp-idf/components/esp_partition /Users/ruki/projects/personal/esp-idf/components/esp_phy /Users/ruki/projects/personal/esp-idf/components/esp_pm /Users/ruki/projects/personal/esp-idf/components/esp_psram /Users/ruki/projects/personal/esp-idf/components/esp_ringbuf /Users/ruki/projects/personal/esp-idf/components/esp_rom /Users/ruki/projects/personal/esp-idf/components/esp_system /Users/ruki/projects/personal/esp-idf/components/esp_timer /Users/ruki/projects/personal/esp-idf/components/esp_wifi /Users/ruki/projects/personal/esp-idf/components/espcoredump /Users/ruki/projects/personal/esp-skainet/examples/wake_word_detection/managed_components/espressif__esp-dsp /Users/ruki/projects/personal/esp-idf/components/esptool_py /Users/ruki/projects/personal/esp-idf/components/fatfs /Users/ruki/projects/personal/esp-idf/components/freertos /Users/ruki/projects/personal/esp-idf/components/hal /Users/ruki/projects/personal/esp-skainet/components/hardware_driver /Users/ruki/projects/personal/esp-idf/components/heap /Users/ruki/projects/personal/esp-idf/components/http_parser /Users/ruki/projects/personal/esp-idf/components/idf_test /Users/ruki/projects/personal/esp-idf/components/ieee802154 /Users/ruki/projects/personal/esp-idf/components/json /Users/ruki/projects/personal/esp-idf/components/log /Users/ruki/projects/personal/esp-idf/components/lwip /Users/ruki/projects/personal/esp-skainet/examples/wake_word_detection/main /Users/ruki/projects/personal/esp-idf/components/mbedtls /Users/ruki/ [... cut ...]
HINT: You are maybe using pre FreeRTOS V8.0.0 APIs. The backward compatibility of such APIs is no longer enabled by default. Please turn on CONFIG_FREERTOS_ENABLE_BACKWARD_COMPATIBILITY explicitly to use such APIs.
HINT: The issue is better to resolve by replacing format specifiers to 'PRI'-family macros (include <inttypes.h> header file).
ninja failed with exit code 1, output of the command is in the /Users/ruki/projects/personal/esp-skainet/examples/wake_word_detection/build/log/idf_py_stderr_output_18021 and /Users/ruki/projects/personal/esp-skainet/examples/wake_word_detection/build/log/idf_py_stdout_output_18021
```

closed 2023-03-28 by waruqi

mph070770, 2023-05-12:
I have the same issue.  What's the fix?

feizi, 2023-05-15:
Currently, only esp-sr supports esp-idf v5.0 and later. The hardware driver of esp-skainet  only support esp-idf v4.4.

waruqi, 2023-05-15:
I use fixed commit 6cef5b859b2dd60c7197ecada8b0d271d22a1b1c for esp-sr. it supports idf 4.4 very well.

Henry586, 2023-08-01:
> Currently, only esp-sr supports esp-idf v5.0 and later. The hardware driver of esp-skainet only support esp-idf v4.4.

got it.

sun-xiangyu, 2023-08-02:
Yes, now esp-skainet has supported IDF5.0 .