espressif/esp-skainet #91: 当我运行cn_speech_commands_recognition例程的时候报错是为什么

ailunwowo, 2023-04-07:
我的板子是ESP-BOX-Lite 开发环境是ESP-IDF5.0，使用的master分支

-- Found Git: C:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1") 
CMake Error at F:/ESP5.0/esp-idf/tools/cmake/project.cmake:243 (message):
  Directory specified in EXTRA_COMPONENT_DIRS doesn't exist:      
  F:/ESP5.0/esp-idf/examples/common_components/led_strip
Call Stack (most recent call first):
  F:/ESP5.0/esp-idf/tools/cmake/project.cmake:327 (__project_init)
  CMakeLists.txt:9 (project)

-- Configuring incomplete, errors occurred!

feizi, 2023-04-07:
目前esp-skainet还不支持idf v5.0.我们正在支持中。
你可以使用 esp-box: https://github.com/espressif/esp-box/tree/master

ailunwowo, 2023-04-07:
我改用4.4.4编译没问题但是下载进去为什么没用呢？

PS D:\LiAiGuo\ESP32S3\Wake_space\esp-skainet-master\examples\en_speech_commands_recognition> set IDF_PATH=C:/Espressif/frameworks/esp-idf-v4.4.4/
PS D:\LiAiGuo\ESP32S3\Wake_space\esp-skainet-master\examples\en_speech_commands_recognition> C:/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe C:\Espressif\frameworks\esp-idf-v4.4.4\tools\idf_monitor.py -p COM6 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 d:\LiAiGuo\ESP32S3\Wake_space\esp-skainet-master\examples\en_speech_commands_recognition\build\speech_commands_recognition.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM6 instead...
--- idf_monitor on \\.\COM6 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x23 (DOWNLOAD(USB/UART0))
Saved PC:0x40041a76
waiting for download
ClearCommError failed (PermissionError(13, '???????????????????????????', None, 22))
Waiting for the device to reconnect.
I (234) esp_image: segment 1: paddr=000e0238 vaddr=3fc9a0I (736) spiram: SPI SRAM memory test OK
I (745) cpu_start: Pro cpu start user code
I (745) cpu_start: cpu freq: 240000000
I (745) cpu_start: Application information:
I (748) cpu_start: Project name:     speech_commands_recognition
I (755) cpu_start: App version:      1
I (759) cpu_start: Compile time:     Apr  7 2023 20:48:09
I (765) cpu_start: ELF file SHA256:  9addd05ad8a0c922...
I (771) cpu_start: ESP-IDF:          v4.4.4-dirty
I (777) heap_init: Initializing. RAM available for dynamic allocation:
I (784) heap_init: At 3FCA1908 len 00047E08 (287 KiB): D/IRAM
I (790) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (797) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (803) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (811) spi_flash: detected chip: gd
I (815) spi_flash: flash io: qio
I (820) sleep: Configure to isolate all GPIO pins in sleep state
I (826) sleep: Enable automatic switching of GPIO sleep configuration
I (833) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (853) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (853) MODEL_LOADER: partition model size: 5292032 by mmap
I (863) I2S: DMA Malloc info, datalen=blocksize=1280, dma_buf_count=6
ClearCommError failed (PermissionError(13, '???????????????????????????', None, 22))
Waiting for the device to reconnect.......

feizi, 2023-04-11:
目前esp-skainet支持的是esp-box,没有esp-box-lite的驱动，你可以自己修改一下驱动或是选择 esp-box: https://github.com/espressif/esp-box/tree/master

Liuyuehuixue, 2023-08-11:
> 目前esp-skainet支持的是esp-box,没有esp-box-lite的驱动，你可以自己修改一下驱动或是选择 esp-box: https://github.com/espressif/esp-box/tree/master

我遇到了一样的问题，想烧录中文语音识别的例子，请问esp-box-lite驱动具体如何修改