espressif/esp-skainet #96: esp-skainet\examples\chinese_tts总是自动重启

Henry586, 2023-05-05:
全新编译chinese_tts项目通过，但是总是自动重启，

仅仅使用idf.py menuconfig 设置 了一下 板子为Korvo-2，没有其他的改动：
(Top) → Audio Media HAL                                                                                                                                                                                                                                                                                                   Audio hardware board (ESP32-S3-Korvo-2)  --->

启动后报告错误:
`I (742) DRV8311: ES8311 in Slave mode
I (752) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
voice_data paration size:3983360
init voice set:template
ESP Chinese TTS v1.7 (Sep 22 2022 14:35:13, 1)
欢迎使用乐鑫语音合成
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420023b1
0x420023b1: panic_handler at D:/esp/Espressif/frameworks/esp-idf-v4.4.4/components/esp_system/port/panic_handler.c:148 (discriminator 3)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x17c4
load:0x403c9700,len:0xe78
load:0x403cc700,len:0x310c
entry 0x403c9980`

请问这个怎么调试，是不是配置有问题呢？
所有的日志在附件中：
[tts_boot_error.txt](https://github.com/espressif/esp-skainet/files/11407328/tts_boot_error.txt)

Henry586, 2023-05-05:
忘记写入语音库了，写入后就好了， 不过这个效果一言难尽啊

`d:\esp\esp-skainet\examples\chinese_tts>esptool.py --port com3 --baud 115200 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x410000 "esp_tts_voice_data_xiaoxin.dat"
esptool.py v3.3.2
Serial port com3
Connecting...
Failed to get PID of a device on com3, using standard reset sequence.

Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:42:f9:c4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 16MB
Flash will be erased from 0x00410000 to 0x0068bfff...
Compressed 2602608 bytes to 2447971...
Wrote 2602608 bytes (2447971 compressed) at 0x00410000 in 217.0 seconds (effective 96.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...`

feizi, 2023-05-06:
目前这个TTS是基于单词拼接的方法，词表大小限制了效果。  
如果是针对数字播报应用，应该还是可以的，我们针对这个应用加了更多的音频单元，听上去会更自然一点。

Henry586, 2023-05-06:
谢谢大佬，回复这么神速！

closed 2023-05-29 by feizi