espressif/esp-skainet #97: why set aec_init to false for esp32-box board?

IamFive, 2023-05-15:
code: https://github.com/espressif/esp-skainet/blob/master/examples/wake_word_detection/main/main.c#L96
why is esp32-s3-box special from other s3 boards?

IamFive, 2023-05-17:
[int feed_channel = esp_get_feed_channel();](https://github.com/espressif/esp-skainet/blob/5abc25bf0d61b089ebe414678a511d5d9c221de5/examples/wake_word_detection/main/main.c#L96)

`esp_get_feed_channel` 是什么作用，为什么esp32-s3-box 上返回的是4，硬件信息里显示只有2个i2s。

feizi, 2023-05-17:
> [ int nch = afe_handle->get_channel_num(afe_data); int feed_channel = esp_get_feed_channel();](https://github.com/espressif/esp-skainet/blob/5abc25bf0d61b089ebe414678a511d5d9c221de5/examples/wake_word_detection/main/main.c#L30C4-L31)
> 
> * `afe_handle->get_channel_num(afe_data)` 中返回的是 Mic 的数量（afe_config.pcm_config.mic_num）吗
     Yes

> * `esp_get_feed_channel` 是什么作用，为什么esp32-s3-box 上返回的是4，硬件信息里显示只有2个i2s。
     One i2s supports two microphone channles

closed 2023-05-17 by IamFive