espressif/esp-skainet #98: Esp-Box build error

suyulin, 2023-05-15:
```

github/esp-skainet/examples/chinese_tts/main/tts_urat.c:12:10: fatal error: esp_vfs.h: No such file or directory
   12 | #include "esp_vfs.h"
      |          ^~~~~~~~~~~
```

feizi, 2023-05-17:
esp-sr 核心的算法库已经支持 idf 5.0
hardware driver 目前在等esp_codec， 一个给所有基于音频的开发板公用的codec开源库和接口
如果你比较急且使用esp-box，可以使用[ esp-box ](https://github.com/espressif/esp-box), 这个仓库已经支持 idf 5.0

suyulin, 2023-05-17:
> 
我就是从 [esp-box ](https://github.com/espressif/esp-box)过来的，但是仓库没有中文的语音的example

IamFive, 2023-05-17:
> esp-sr 核心的算法库已经支持 idf 5.0 hardware driver 目前在等esp_codec， 一个给所有基于音频的开发板公用的codec开源库和接口 如果你比较急且使用esp-box，可以使用[ esp-box ](https://github.com/espressif/esp-box), 这个仓库已经支持 idf 5.0

esp-box 不是通用的吧？esp-skainet 才封装了通用的 driver。

feizi, 2023-05-17:
> > esp-sr 核心的算法库已经支持 idf 5.0 hardware driver 目前在等esp_codec， 一个给所有基于音频的开发板公用的codec开源库和接口 如果你比较急且使用esp-box，可以使用[ esp-box ](https://github.com/espressif/esp-box), 这个仓库已经支持 idf 5.0
> 
> esp-box 不是通用的吧？esp-skainet 才封装了通用的 driver。

是的esp-skainet支持更多的开发板，但只支持了音频相关的功能，你也可以看到例子都相对简单。
对于esp-box的开发，我推荐使用esp-box,支持了esp-box的显示和其他功能模块。
就我所知， https://github.com/espressif/esp-box/tree/master/examples/factory_demo 是支持中英文的，你可以通过`idf.py menuconfig` 选择相应的模型

closed 2023-05-17 by suyulin