espressif/esp-skainet #99: 与ADF一起编译不通过

dengbq1234, 2023-05-15:
IDF版本4.4，ADF已经安装配置好并且例子能够编译通过和运行，skainet的例子:cn_speech_commands_recognition 能够编译通过和正常运行，在cn_speech_commands_recognition例子的基础上，想增加使用ADF的一些组件，方法是在CMakeLists.txt增加一行:include($ENV{ADF_PATH}/CMakeLists.txt)

完整CMakeLists.txt如下:
cmake_minimum_required(VERSION 3.5)

set(EXTRA_COMPONENT_DIRS
    ../../components    
    )

include($ENV{ADF_PATH}/CMakeLists.txt)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)

project(speech_commands_recognition)

再idf.py build
输出：

idf.py build
Executing action: all (aliases: build)
Running ninja in directory d:\deng\work\skainet\git\esp-skainet\examples\cn_speech_commands_recognition_test02\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32s3
CMake Error at D:/Espressif/frameworks/esp-idf-v4.4/tools/cmake/build.cmake:200 (message):
  Failed to resolve component 'hardware_driver'.
Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v4.4/tools/cmake/build.cmake:226 (__build_resolve_and_add_req)
  D:/Espressif/frameworks/esp-idf-v4.4/tools/cmake/build.cmake:512 (__build_expand_requirements)
  D:/Espressif/frameworks/esp-idf-v4.4/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:11 (project)

-- Configuring incomplete, errors occurred!
See also "D:/deng/work/skainet/git/esp-skainet/examples/cn_speech_commands_recognition_test02/build/CMakeFiles/CMakeOutput.log".
[31mFAILED: [0mbuild.ninja
D:\Espressif\tools\cmake\3.20.3\bin\cmake.exe --regenerate-during-build -SD:\deng\work\skainet\git\esp-skainet\examples\cn_speech_commands_recognition_test02 -BD:\deng\work\skainet\git\esp-skainet\examples\cn_speech_commands_recognition_test02\build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1

请问怎样消除这个错误，能否给个例子，谢谢。

feizi, 2023-05-17:
ADF 中也包括 esp-sr（主要提供 wakenet & multinet）, 可能会和esp-skainet 中的esp-sr 冲突。
相较于ADF， esp-skainet更容易被集成，所以我建议你吧 esp-skainet中的例子集成到ADF.
可以参考： https://github.com/espressif/esp-adf/tree/master/examples/speech_recognition/wwe

IamFive, 2023-05-17:
> ADF 中也包括 esp-sr（主要提供 wakenet & multinet）, 可能会和esp-skainet 中的esp-sr 冲突。 相较于ADF， esp-skainet更容易被集成，所以我建议你把 esp-skainet中的例子集成到ADF. 可以参考： https://github.com/espressif/esp-adf/tree/master/examples/speech_recognition/wwe

ADF 和 esp-skainet 的主要区别是什么呢，该使用哪个进行音频开发呢。

feizi, 2023-05-17:
esp-skainet 是一个简单的仓库，主要提供如何使用AI算法,音频算法，主要用于学习。
ADF 提供更多的模块，包括音乐播放，音频格式编解码，Wi-Fi，蓝牙等，主要用于产品开发

dengbq1234, 2023-05-18:
adf中的wwe不能实现无唤醒词的情况进行语音识别，所以只能使用skainet来达成目标。但我的工程中还有其它功能模块要使用ADF，因此同时使用skainet和ADF成为当前唯一的选择。
ADF中esp_sr目标是空的，我看了下，好像是skainet和adf的音频驱动存在冲突，找不到解决的办法。

IamFive, 2023-05-18:
> 像是skainet和adf的音频驱动存在冲突，找不到解决的办法。

- skainet 的代码非常简单，只是做了一些初始化的操作。除去hardware下的代码，都是esp-sr的，在adf里都是可以直接用的。
- adf里，wwe同一目录下，不是有vad。

dengbq1234, 2023-05-18:
我把skainet componets下面的内容拷贝到adf的componets下，正常编译adf的示例工程没有问题，如果要在示例工程中使用skainet的组件，编译会报错。

dengbq1234, 2023-05-18:
报错内容:Running ninja in directory d:\deng\work\test\adf_examples\recorder\pipeline_raw_http_s3_test_skainet\build
Executing "ninja all"...
[0/1] Re-running CMake...
fatal: not a git repository (or any of the parent directories): .git
-- ccache will be used for faster recompilation
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
CMake Error at D:/Espressif/frameworks/esp-idf-v4.4/tools/cmake/build.cmake:200 (message):
  Failed to resolve component 'hardware_driver'.
Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v4.4/tools/cmake/build.cmake:226 (__build_resolve_and_add_req)
  D:/Espressif/frameworks/esp-idf-v4.4/tools/cmake/build.cmake:512 (__build_expand_requirements)
  D:/Espressif/frameworks/esp-idf-v4.4/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:8 (project)

dengbq1234, 2023-05-18:
ok

closed 2023-05-18 by dengbq1234