espressif/esp-skainet #101: esp-skainet master分支的wake-word-detect例程问题

ZorkyZorky, 2023-06-25:
依赖：esp-sr-master、esp-box
问题1：直接编译了wake-word-detect例程，

espressif2022, 2023-06-25:
https://github.com/espressif/esp-box/tree/master/examples/factory_demo

你这边试下 factory_demo 的 demo，这边有个 SR 的应用可以参考下先。SR 这边的库更新了，之前和他们沟通应用这边还没同步，我明天也试下他们的 demo。

feizi, 2023-06-26:
> 问题1：
我刚用esp-box测试了一下，并没有复现你的问题，请把完整的log贴一下

> 问题2：尝试了使能aec，合并feed和detect两个任务到同一任务下，在feed()完afe_data后再fetch()，可以消除警告，但是在50000个时间戳左右esp-box会由fetch导致重启。

feed 和 detect 两个任务不可以合并，feed和detect内部通过一个ring buffer沟通数据，如果放在一个任务内，就会有时序问题。

ZorkyZorky, 2023-06-29:
问题1确实是我有改动导致的，抱歉！但是两次唤醒后大约等1~2分钟会自动重启。log如下
Load: wn9_xiaoaitongxue
I (701) AFE_SR: afe interface for speech recognition

I (711) AFE_SR: AFE version: SR_V220727

I (711) AFE_SR: Initial auido front-end, total channel: 3, mic num: 2, ref num: 1

I (721) AFE_SR: aec_init: 1, se_init: 1, vad_init: 1

I (731) AFE_SR: wakenet_init: 1

MC Quantized wakenet9: WakeNet9_v1h24_xiaoaitongxue_3_0.620_0.627, tigger:v3, mode:2, p:0, (Jun 14 2023 11:15:21)
I (971) AFE_SR: wake num: 3, mode: 0, (Jun 14 2023 11:15:21)

------------detect start------------
wakeword detected
-----------LISTENING-----------
wakeword detected
-----------LISTENING-----------
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x42046bc5
0x42046bc5: add_rv16_rv16_rv16_728 at /home/sunxiangyu/workspace/esp_sr_lib/build/../components/esp_audio_processor/asm_functions/add_rv16_rv16_rv16_728.S:29

espressif2022, 2023-07-03:
> 

我看 shainet （master） `wake_word_detection`，默认选不到 `WakeNet9_` 的唤醒词，也选不了 `esp-box` 的硬件，所以你工程这还是自己修改了的 ？

ZorkyZorky, 2023-07-03:
没有。图一截的wake detect的仓库，里面就有配置文件啊；另外wake detect代码也有写如果配置了box，……

[image]

#if defined CONFIG_ESP32_S3_BOX_BOARD || defined CONFIG_ESP32_S3_EYE_BOARD
    afe_config.aec_init = false;
    #if defined CONFIG_ESP32_S3_EYE_BOARD
        afe_config.pcm_config.total_ch_num = 2;
        afe_config.pcm_config.mic_num = 1;
        afe_config.pcm_config.ref_num = 1;
    #endif
#endif

espressif2022, 2023-07-03:
> 没有。图一截的wake detect的仓库，里面就有配置文件啊；另外wake detect代码也有写如果配置了box，……
>
> [image]
>
> #if defined CONFIG_ESP32_S3_BOX_BOARD || defined CONFIG_ESP32_S3_EYE_BOARD afe_config.aec_init = false; #if defined [...]

ok, 我一会测试一下

espressif2022, 2023-07-03:
> 没有。图一截的wake detect的仓库，里面就有配置文件啊；另外wake detect代码也有写如果配置了box，……
>
> [image]
>
> #if defined CONFIG_ESP32_S3_BOX_BOARD || defined CONFIG_ESP32_S3_EYE_BOARD afe_config.aec_init = false; #if defined [...]

[image]
我这测试结果，好像也是正常的，没有重启或者异常现象。原始工程文件，未作任何改动。

espressif2022, 2023-07-07:
> 可以把测试的例程打包发给我试试吗？我的邮箱[1053714527@qq.com](mailto:1053714527@qq.com)。我排查下是不是硬件导致的@espressif2022

我把 bin 发你下

ZorkyZorky, 2023-07-11:
试了新买的esp32s3-korvo-2，没有这个问题了，怀疑是这个box的问题

espressif2022, 2023-07-11:
> 试了新买的esp32s3-korvo-2，没有这个问题了，怀疑是这个box的问题

我发过去的 bin 也是不行？硬件有问题的话，你这联系下售后看看吧。

closed 2023-07-19 by feizi