espressif/esp-skainet #102: esp-skainet master分支的wake_word_detect例程VAD问题

ZorkyZorky, 2023-07-11:
开发板：esp32-s3-korvo-2
情况：在detect任务里fetch后打印VAD的状态，第一次被唤醒后vad状态会一直是1（没有说话），直到第二次被唤醒。

I (3230) SR: vad_state: 0
I (3280) SR: vad_state: 0
I (3290) SR: vad_state: 0
I (3360) SR: vad_state: 0
I (3360) SR: vad_state: 0
I (3420) SR: vad_state: 0
I (3420) SR: vad_state: 0
I (3480) SR: vad_state: 0
I (3490) SR: vad_state: 0
I (3550) SR: vad_state: 0
I (3550) SR: vad_state: 1
wakeword detected
-----------LISTENING-----------
I (3610) SR: vad_state: 1
I (3610) SR: vad_state: 1
I (3670) SR: vad_state: 1
I (3680) SR: vad_state: 1
I (3730) SR: vad_state: 1
I (3740) SR: vad_state: 1
I (3810) SR: vad_state: 1
I (3810) SR: vad_state: 1
I (3860) SR: vad_state: 1
I (3870) SR: vad_state: 1
I (3920) SR: vad_state: 1
I (3930) SR: vad_state: 1
I (4000) SR: vad_state: 1
I (4000) SR: vad_state: 1
I (4050) SR: vad_state: 1
I (4060) SR: vad_state: 1
I (4120) SR: vad_state: 1
I (4130) SR: vad_state: 1
I (4190) SR: vad_state: 1
I (4190) SR: vad_state: 1
I (4240) SR: vad_state: 1
I (4250) SR: vad_state: 1
I (4310) SR: vad_state: 1
I (4320) SR: vad_state: 1
I (4380) SR: vad_state: 1
I (4380) SR: vad_state: 1
I (4440) SR: vad_state: 1
I (4450) SR: vad_state: 1
I (4500) SR: vad_state: 1
I (4510) SR: vad_state: 1
I (4570) SR: vad_state: 1
I (4570) SR: vad_state: 1
I (4630) SR: vad_state: 1
I (4640) SR: vad_state: 1
I (4690) SR: vad_state: 1
I (4700) SR: vad_state: 1
I (4770) SR: vad_state: 1
I (4770) SR: vad_state: 1
I (4820) SR: vad_state: 1
I (4830) SR: vad_state: 1
I (4880) SR: vad_state: 1
I (4890) SR: vad_state: 1
I (4960) SR: vad_state: 1
I (4960) SR: vad_state: 1
I (5010) SR: vad_state: 1
I (5020) SR: vad_state: 1
I (5080) SR: vad_state: 1
I (5090) SR: vad_state: 1
I (5150) SR: vad_state: 1
I (5150) SR: vad_state: 1
I (5200) SR: vad_state: 1
I (5210) SR: vad_state: 1
I (5270) SR: vad_state: 1
I (5280) SR: vad_state: 1
I (5340) SR: vad_state: 1
I (5340) SR: vad_state: 1
I (5400) SR: vad_state: 1
I (5410) SR: vad_state: 1
I (5460) SR: vad_state: 1
I (5470) SR: vad_state: 1
I (5530) SR: vad_state: 1
I (5530) SR: vad_state: 1
I (5590) SR: vad_state: 1
I (5600) SR: vad_state: 1
I (5650) SR: vad_state: 1
I (5660) SR: vad_state: 1
I (5730) SR: vad_state: 1
I (5730) SR: vad_state: 1
I (5780) SR: vad_state: 1
I (5790) SR: vad_state: 1
I (5840) SR: vad_state: 1
I (5850) SR: vad_state: 1
I (5920) SR: vad_state: 1
I (5920) SR: vad_state: 1
I (5970) SR: vad_state: 1
I (5980) SR: vad_state: 1
I (6040) SR: vad_state: 1
I (6050) SR: vad_state: 1
I (6110) SR: vad_state: 1
I (6110) SR: vad_state: 1
I (6160) SR: vad_state: 1
I (6170) SR: vad_state: 1
I (6230) SR: vad_state: 1
I (6240) SR: vad_state: 1
I (6300) SR: vad_state: 1
I (6300) SR: vad_state: 1
I (6360) SR: vad_state: 1
I (6370) SR: vad_state: 1
I (6420) SR: vad_state: 1
I (6430) SR: vad_state: 1
I (6490) SR: vad_state: 1
I (6490) SR: vad_state: 1
I (6550) SR: vad_state: 1
I (6560) SR: vad_state: 1
I (6610) SR: vad_state: 1
I (6620) SR: vad_state: 1
I (6690) SR: vad_state: 1
I (6690) SR: vad_state: 1
I (6740) SR: vad_state: 1
I (6750) SR: vad_state: 1
I (6800) SR: vad_state: 1
I (6810) SR: vad_state: 1
I (6880) SR: vad_state: 1
I (6880) SR: vad_state: 1
I (6930) SR: vad_state: 1
I (6940) SR: vad_state: 1
I (7000) SR: vad_state: 1
I (7010) SR: vad_state: 1
I (7070) SR: vad_state: 1
I (7070) SR: vad_state: 1
I (7120) SR: vad_state: 1
wakeword detected
-----------LISTENING-----------
I (7130) SR: vad_state: 1
I (7190) SR: vad_state: 1
I (7200) SR: vad_state: 1
I (7260) SR: vad_state: 0
I (7260) SR: vad_state: 0
I (7320) SR: vad_state: 0
I (7330) SR: vad_state: 0
I (7380) SR: vad_state: 0
I (7390) SR: vad_state: 0
I (7450) SR: vad_state: 0
I (7450) SR: vad_state: 0
I (7510) SR: vad_state: 0
I (7520) SR: vad_state: 0
I (7570) SR: vad_state: 0
I (7580) SR: vad_state: 0
I (7650) SR: vad_state: 0
I (7650) SR: vad_state: 0
I (7700) SR: vad_state: 0
I (7710) SR: vad_state: 0
I (7760) SR: vad_state: 0
I (7770) SR: vad_state: 0
I (7840) SR: vad_state: 0
I (7840) SR: vad_state: 0
I (7890) SR: vad_state: 0

BlueSkyB, 2023-07-12:
这个问题后续我们会修复

feizi, 2023-07-28:
修复已经提交

closed 2023-07-28 by feizi

ZorkyZorky, 2023-07-28:
是在commits里吗？上周的？没有看到