espressif/esp-skainet #104: Using Multinet without timeout

PrathamG, 2023-07-19:
Hi! I was wondering if it is possible to let the multinet run indefinitely after the wakeword is detected. 

From #81 I found out that the second parameter in 
`multinet->create(mn_name, time_ms)`
is the timeout parameter, which determines how long after the last keyword should the multinet be stopped.

Is there an option to allow the multinet to keep running forever? 
Also, are there any disadvantages to letting the multinet run without timeout?

feizi, 2023-07-27:
> Is there an option to allow the multinet to keep running forever? Also, are there any disadvantages to letting the multinet run without timeout?

Yes, the second parameter is timeout duration. You can ignore the timeout state when you want to run multinet always.

PrathamG, 2023-08-02:
Okay got it, thanks!

closed 2023-08-02 by PrathamG