espressif/esp-skainet #105: BSS (Blind Source Separation)

tareqAmen, 2023-07-27:
how can i use the BSS (Blind Source Separation) and NS (Noise Suppression in your code, I didn't find anything related to that in the examples.
Please help to use these algorithms.

feizi, 2023-07-27:
Hi @tareqAmen , BSS and NS are in AFE module. Please refer to [AFE docs](https://docs.espressif.com/projects/esp-sr/en/latest/esp32s3/audio_front_end/README.html) about how to use them.

tareqAmen, 2023-07-27:
> Hi @tareqAmen , BSS and NS are in AFE module. Please refer to [AFE docs](https://docs.espressif.com/projects/esp-sr/en/latest/esp32s3/audio_front_end/README.html) about how to use them.

Thank you for reply, But there are no sources code for BSS or NS

feizi, 2023-07-27:
> > Hi @tareqAmen , BSS and NS are in AFE module. Please refer to [AFE docs](https://docs.espressif.com/projects/esp-sr/en/latest/esp32s3/audio_front_end/README.html) about how to use them.
> 
> Thank you for reply, But there are no sources code for BSS or NS

We don't have open source code for BSS and NS yet

tareqAmen, 2023-07-29:
> We don't have open source code for BSS and NS yet

can you help me to build esp32 code make human voice activity detection using esp-skainet framework?
In my project, I want to turn on my device when there is only a human voice, so that it is not affected by any other sound, such as the sound of the wind or any other sounds in the environment.

I'm using esp32s3 with inmp441 i2s mic.