espressif/esp-skainet #108: get core panic with esp_mn_iface_t interface

adonishong, 2023-08-04:
Hi, we are trying transplant offline voice recognition to lilygo twatch2020-v3, which is based on esp32. By instruction in test_board.c and my_codec.c and etc, we modify bsp_board.c related stuff, we have made wakenet works well. 

Version info is here, esp-skainet is latest version from GitHub, and esp-idf is 4.4.

However with multinet, we could not make it work, with official cn_speech_commands_recognition, we found that, when we call esp_mn_commands_update_from_sdkconfig, program will meet core panic error, and we trace into this function, here is the path esp_mn_commands_update_from_sdkconfig  call esp_mn_commands_add call esp_mn_model_handle->check_speech_command make program panic.
We tried comment out this call, and then, esp_mn_commands_update_from_sdkconfig call esp_mn_commands_update call esp_mn_model_handle->set_speech_commands, this also will get us core panic ...

What shall we do now? From an example wwe from esp-adf, we noticed that those guy claims multinet does not support esp32 now. If this is the situation, is there any elder version could make esp32 works with multinet to do offline voice recognition?

feizi, 2023-08-10:
Yes, the latest multinet does not support ESP32. Due to performance limitations, esp32 can only run models less than 1M at most, and cannot get a good offline command word experience, so our latest multinet no longer supports esp32.

closed 2023-08-11 by adonishong

adonishong, 2023-08-11:
> Yes, the latest multinet does not support ESP32. Due to performance limitations, esp32 can only run models less than 1M at most, and cannot get a good offline command word experience, so our latest multinet no longer supports esp32.
> @adonishong ， If you want to use the offline speech command, I recommend that you use the ESP32-S3 chip or module.

At least claim this clear in all versions of documents please. Otherwise, this will waste bunch of application developer's money and time. And eventually, this will push developers away ...

feizi, 2023-08-11:
Sorry for the confusion, we will mark it more clearly in the documentation later