espressif/esp-skainet #109: test  example  cn_speech_commands_recognition error

Liuyuehuixue, 2023-08-10:
请问使用ESP32-S3-BOX-Lite 测试cn_speech_commands_recognition例子，flash monitor 后出现如下信息：
I (322) opi psram: BurstLen  : 0x01 (32 Byte)
I (327) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (333) opi psram: DriveStrength: 0x00 (1/1)
W (338) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (349) spiram: Found 64MBit SPI RAM device
I (353) spiram: SPI RAM mode: sram 80m
I (358) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (365) cpu_start: Pro cpu up.
I (368) cpu_start: Starting app cpu, entry point is 0x40379488
0x40379488: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v4.4.4/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (660) spiram: SPI SRAM memory test OK
I (669) cpu_start: Pro cpu start user code
I (669) cpu_start: cpu freq: 240000000
I (669) cpu_start: Application information:
I (672) cpu_start: Project name:     speech_commands_recognition
I (678) cpu_start: App version:      v0.3.0-258-ga9429c3-dirty
I (685) cpu_start: Compile time:     Aug 10 2023 17:26:01
I (691) cpu_start: ELF file SHA256:  fd258b55f4ebd1dc...
I (697) cpu_start: ESP-IDF:          v4.4.4-dirty
I (702) heap_init: Initializing. RAM available for dynamic allocation:
I (710) heap_init: At 3FC9FC28 len 00049AE8 (294 KiB): D/IRAM
I (716) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (723) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (729) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (737) spi_flash: detected chip: gd
I (741) spi_flash: flash io: qio
I (745) sleep: Configure to isolate all GPIO pins in sleep state
I (752) sleep: Enable automatic switching of GPIO sleep configuration
I (759) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (779) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (779) MODEL_LOADER: The storage free size is 15360 KB
I (789) MODEL_LOADER: The partition size is 5168 KB
I (799) MODEL_LOADER: Successfully map model partition
I (799) I2S: DMA Malloc info, datalen=blocksize=320, dma_buf_count=6
I (809) I2S: I2S0, MCLK output by GPIO42
I (809) I2S: DMA Malloc info, datalen=blocksize=1280, dma_buf_count=6

无error

之后串口工具，对应COM调试不通。
sdkconfig  esp32s3,esp32s3eye, esp32s3box都尝试过，都是一样的问题，请问是什么原因呢？

BlueSkyB, 2023-08-11:
Hi Liuyuehuixue, 
    The skainet don't support ESP32-S3-BOX-Lite. You can view the supported boards from README.md.

Liuyuehuixue, 2023-08-11:
thank you for your reply. Hmm, I've seen the documentation. I tried to use the configuration file of ESP32S3-Box, and the compilation was successful. The terminal can print the identification command and start detection information, but the following error message appears:
E (820) I2C_If: Fail to write to dev 80

Is it feasible to change the relevant configuration of the file configuration esp32_s3_box_board.h? For example, try to change the following configuration:
/**
  * @brief ESP32-S3-BOX I2C GPIO definition
  *
  */
#define FUNC_I2C_EN (1)
#define I2C_NUM (0)
#define I2C_CLK (600000)
#define GPIO_I2C_SCL (GPIO_NUM_18)
#define GPIO_I2C_SDA (GPIO_NUM_8)

/**
  * @brief ESP32-S3-BOX I2S GPIO definition
  *
  */
#define FUNC_I2S_EN (1)
#define GPIO_I2S_LRCK (GPIO_NUM_47)
#define GPIO_I2S_MCLK (GPIO_NUM_2)
#define GPIO_I2S_SCLK (GPIO_NUM_17)
#define GPIO_I2S_SDIN (GPIO_NUM_16)
#define GPIO_I2S_DOUT (GPIO_NUM_15)

I would be very grateful if you could give me some advice.

feizi, 2023-08-16:
Hi @Liuyuehuixue , I do not know about this issues. May you can ask for help under [esp-bsp](https://github.com/espressif/esp-bsp) project. They know more about hardware drivers.