espressif/esp-skainet #110: 引入某些库文件如nvs_flash.h  会报错

casitelwithdean, 2023-08-11:
我下载了最新版本的skainet，然后选用esp-box编译了工程，一切正常，烧录到板子上功能也正常。当我在main.c里添加了#include "nvs_flash.h"之后，就会报如下错：其余的什么都没有改动
[image]
又多测试了几个 #include "esp_http_client.h"   #include "esp_wpa2.h"都会报同样的错误
但是在idf的示例中，即使添加了#include "nvs_flash.h"也不会报错的，只是在skainet里面的示例会报错。请问时有什么设置需要做吗？
谢谢

feizi, 2023-08-11:
hi @casitelwithdean , esp-skainet 例子只添加了部分所需的esp-idf components，如果你想用到其他component需要在CMakeList.txt中指定，比如你想添加“nvs_flash.h”, 对应于esp-idf的 [components/nvs_flash](https://github.com/espressif/esp-idf/tree/master/components/nvs_flash)， 可以在[cn_speech_commands_recognition/main/CMakeList.txt](https://github.com/espressif/esp-skainet/blob/master/examples/cn_speech_commands_recognition/main/CMakeLists.txt)中添加一行

```
set(requires
    esp-sr
    led_strip
    hardware_driver
    nvs_flash
    )
```

casitelwithdean, 2023-08-11:
> p

好的，明白了，十分感谢！

casitelwithdean, 2023-08-11:
> hi @casitelwithdean , esp-skainet 例子只添加了部分所需的esp-idf components，如果你想用到其他component需要在CMakeList.txt中指定，比如你想添加“nvs_flash.h”, 对应于esp-idf的 [...]

好的，明白了，十分感谢！

closed 2023-08-11 by casitelwithdean