espressif/esp-skainet #112: 关于将esp-box的aec算法启用

ZorkyZorky, 2023-08-18:
esp-box没有引入aec算法，参照esp-skainet的写法将esp-box的唤醒程序做了如下修改，但是没办法唤醒（alexa），有什么地方疏忽的没有修改吗？

1. ret |= es7210_write_reg(ES7210_SDP_INTERFACE2_REG12, 0x02); //在es7210.c init()中
2.afe_config.aec_init = true;
3.i2s_read(I2S_NUM_0, feed_buffer, chunksize * 4 * sizeof(int16_t),&bytes_read, portMAX_DELAY);//读取4个通道
4.for (int i = 0; i < chunksize; i++) {
int16_t ref = feed_buffer[4 * i + 0];
feed_buffer[3 * i + 0] = feed_buffer[4 * i + 1];
feed_buffer[3 * i + 1] = feed_buffer[4 * i + 3];
feed_buffer[3 * i + 2] = ref;
}//adjust channel

feizi, 2023-08-18:
AFE config 中的microphone通道修改了吗
```
        afe_config.pcm_config.total_ch_num = 3;
        afe_config.pcm_config.mic_num = 2;
        afe_config.pcm_config.ref_num = 1;
```

BlueSkyB, 2023-08-21:
esp-box那边的程序可能和目前esp-skainet中的代码存在差异，esp-box的代码情况不是很了解，只根据目前esp-skainet中的代码，回答你提到的几个问题：

1. 目前skainet中的音频驱动，是基于esp-adf中的esp_codec_dev组件调用进行的，并没有直接调用es7210的驱动接口，驱动请不要随意修改，目前集成的esp_codec_dev版本是满足esp-skainet中的功能需求的。如果确实有需求要修改音频驱动，可能需要移步到esp-adf那边提issue，看如何修改移植新的驱动。我们这边无法提供更多有效的帮助。

2. 目前skainet中，并没有使用TDM模式，可以理解为还是I2S，只是底层的IDF做了适配。当然，这些动作都是基于esp_codec_dev组件接口进行的。

3. 目前的skainet示例代码，是支持在esp-box上运行的，不需要进行代码修改。在示例代码目录下，只需要将sdkconfig.defaults.esp32s3box文件，拷贝替换之前的sdkconfig文件，即可编译运行。

4. 如果想体验AEC，打开afe_config.aec_init = true配置即可。

以上都是基于skainet项目情况进行回答的，如果是想基于esp-box体验更多功能，碰到音频问题，可能需要在esp-box那边提issue，请他们看看项目的驱动问题。

ZorkyZorky, 2023-09-11:
不太理解，如果不启用TDM模式，是怎么读到回声通道的？标准的I2S只有双通道呀

feizi, 2023-09-13:
因为一个I2S通道32bit,可以采集两路16bit麦克风的信号，所以两个I2S通道可以采集4路麦克风信号

ZorkyZorky, 2023-09-13:
你们有采用两个I2S通道吗，在ESP-box里吗？请问方便分享一下相关代码吗？1053714526@qq.com谢谢

ZorkyZorky, 2023-09-13:
你们有采用两个I2S通道吗，在ESP-box里吗？请问方便分享一下相关代码吗？1053714527@qq.com谢谢

espressif2022, 2023-09-14:
> 你们有采用两个I2S通道吗，在ESP-box里吗？请问方便分享一下相关代码吗？[1053714527@qq.com](mailto:1053714527@qq.com)谢谢

[image]
这边是一路 I2S， 然后 esp-box 仓库也是依赖的 `esp_codec_dev` 组件。
硬件上 ADC 3,4 也是接到了同一个 I2S，如上 ES7210 的寄存器 `SDOUT_MODE` 应该是要配置的，现在默认是只接受 ADC12

closed 2024-06-18 by feizi