espressif/esp-skainet #115: 看门狗复位以及AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!的问题

Liuyuehuixue, 2023-09-27:
开发板：esp32s3eye
在examples\cn_speech_commands_recognition中detect_Task添加一些代码后，出现以下错误:

W (5929) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

W (5939) AFE_SR: ERROR! afe_feed_aec_init_false, rb_out slow!!!

E (5949) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (5949) task_wdt:  - IDLE (CPU 0)
E (5949) task_wdt: Tasks currently running:
E (5949) task_wdt: CPU 0: detect
E (5949) task_wdt: CPU 1: IDLE
E (5949) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x4205A0A6:0x3FC99CA0 0x4037AEF1:0x3FC99CC0 0x4200C0A3:0x3FCEE090 0x42014F7A:0x3FCEE390 0x420086B3:0x3FCEE3C0 0x42006725:0x3FCEE3E0 0x420064A0:0x3FCEE400 0x40382375:0x3FCEE450
0x4205a0a6: task_wdt_isr at D:/Espressif/frameworks/esp-idf-v4.4.4/components/esp_system/task_wdt.c:183 (discriminator 3)
请问如何解决？

feizi, 2023-09-27:
esp-sr的任务负载比较重，添加多个task的时候，要注意两点：
1. CPU 负载比较重的任务，尽可能分开执行，比如启用multinet识别离线语音命令的时候，不要同时传输数据
2. 必须同时执行的task， 分配到不同的cpu core

Liuyuehuixue, 2023-09-27:
好的，谢谢回复，添加新功能可以新建一个xTaskCreatePinnedToCore是吧

closed 2023-12-14 by feizi