espressif/esp-skainet #120: Examples don't work with ESP32-S3 Box 3?

mapigit, 2024-01-01:
Hi there,

I've tried for several days now to get the Wake-word example working on my new ESP32-S3 Box 3. Different IDF versions gave the same reult: The example compiles without problems (yes, I've set the target to ESP32S3, and in menuconfig the Esp32-S3 Box), but there is no reaction to any wakeword. Same for the en speech recognition. However, the factory demo works fine: I've built and flashed the demo v1.2.4 with IDF 5.1.2; wakeword and english commands are recognized.

Is it possible that the Skainet example don't work with the Box 3 Hardware version yet? BTW, the LOG of the wakeword example shows MIC1-MIC4, which I find strange, whereas the factory demo correctly print only MIC1, MIC2.

Can you please help?
Thanks, cheers,
Martin

feizi, 2024-01-03:
@mapigit , Yes, currently, Skainet examples don't work with the Box 3 Hardware version yet. We will support it this week.

feizi, 2024-01-03:
Thank you for the question. Now Skainet has supported Box-3. Please try again.
The commit: https://github.com/espressif/esp-skainet/commit/8824d838a72b4a16d1cf059fcf677d2c169ffd46

mapigit, 2024-01-03:
@feizi, could you please also add "|| CONFIG_ESP32_S3_BOX_3_BOARD" in main.c? E.g. in the english speech recognition example in lines 170 and 154. Thanks!

mapigit, 2024-01-03:
Hi there,

if I select "multiple wakewords" in the  english speech recognition example  and then "Alexa" as well as "Hi, ESP!", only "Alexa" will be recognized as a wakeword by the Box3. If I keep "multiple wakewords", deselect "Alexa" and select only "Hi, ESP!", then "Hi, ESP!" works. Is this a bug? I'd like to be able to use bothe wakewords alternatively in the same program.

Thanks, cheers,
Martin

mapigit, 2024-01-22:
Hi @feizi 
that's exactly what I did. After selecting "multiple wakewords"  as well as both wakewords, still only "Alexa" works. 
Thanks, cheers,
Martin

sun-xiangyu, 2024-01-23:
You also need to bind the model name of the WakeNet in the AFE config.
Please refer to [this example](https://github.com/espressif/esp-skainet/blob/master/examples/wake_word_detection/main/main.c#L106-L107)

```
    afe_config.wakenet_model_name = wn_name;
    afe_config.wakenet_model_name_2 = wn_name_2;
```