espressif/esp-skainet #121: support for xiao esp32s3 sense

gparvind, 2024-01-12:
I tried configuring wake word detection example for xiao esp32s3 sense but it does't detect any spoken wake words.
I have set the config for pdm mono mode single mic config. Tried with sample rate 441000 and 16000 both not working. When I print feed data, I can it change when spoken something in mic. With same mic i2s configuration, esp-idf i2s_recorder exanple is working.
Anything specifig I am missing?

gparvind, 2024-01-15:
vad_state shows speech activity so microphone is getting detected but not the wake words.

gparvind, 2024-01-19:
@feizi any help with this?

sun-xiangyu, 2024-01-22:
Can you save audio data of your board by WiFi or SD card?

wzdl, 2024-03-28:
Hi, I'm also trying to run wake word detection on the Xiao ESP32-S3 Sense board. I've encountered similar issues, but I managed to get it working by modifying the `pcm_config` and enabling `nsent`.
```c
afe_config.pcm_config.total_ch_num = 1;
afe_config.pcm_config.mic_num = 1;
afe_config.pcm_config.ref_num = 0;
```
﻿
However, even though I've loaded multiple models, only the `wakenet9l_tts1h8_你好小鑫` model seems to work, and the recognition accuracy is not satisfactory.
﻿
Are there any methods to improve recognition accuracy?

lixy123, 2024-06-01:
https://blog.csdn.net/weixin_44135213/article/details/137217731 
There are some clues. Actual measurement cannot be used.  it still cannot be awakened. Further efforts are needed.

paul356, 2025-03-14:
With little tweaks I can run wake_word_detection example on XIAO ESP32S3 SENSE. If you are interested, required changes are here. https://github.com/paul356/esp-skainet/commit/4ea3ffd1447aaa4bd1c76e034f2500d9a2d3bd57