espressif/esp-skainet #124: en_speech_commands_recognition crashes after wake word detected.

femahmudzadeh, 2024-01-18:
I tried two of the examples `en_speech_commands_recognition` and `wake_word_detection` on **ESP32-S3-Korvo-2 V3.0**. Both are built and flashed successfully. When I say the wake word, it detects it but crashes immediately after with the following error:

> ------------detect start------------
> res:1, prob:0.660444, prob_sum:3.022056 
> res:1, prob:0.530206, prob_sum:3.022056 
> res:1, prob:0.525404, prob_sum:2.882793 
> res:1, prob:0.502554, prob_sum:3.022056 
> res:1, prob:0.502381, prob_sum:2.882793 
> res:1, prob:0.628529, prob_sum:3.032536 
> res:1, prob:0.644184, prob_sum:3.118070 
> WAKEWORD DETECTED
> res:1, prob:0.611815, prob_sum:3.000803 
> res:1, prob:0.621404, prob_sum:3.049605 
> res:1, prob:0.611808, prob_sum:3.086167 
> res:1, prob:0.632071, prob_sum:3.163918 
> res:1, prob:0.608281, prob_sum:3.163918 
> res:1, prob:0.601389, prob_sum:3.163918 
> Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
> 
> Core  1 register dump:
> PC      : 0x00000000  PS      : 0x00060330  A0      : 0x820091d8  A1      : 0x3fcbc8f0  
> A2      : 0x3fcad140  A3      : 0x3fcadcc4  A4      : 0x00000000  A5      : 0xffffffff  
> A6      : 0x00000000  A7      : 0x3c39673e  A8      : 0x8200b23c  A9      : 0x3fcbc8d0  
> A10     : 0x3fcad968  A11     : 0x41200000  A12     : 0x00000066  A13     : 0x3fcb4734  
> A14     : 0x3fcb4400  A15     : 0x00000002  SAR     : 0x00000014  EXCCAUSE: 0x00000014  
> EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  
> 0x400556d5: strlen in ROM
> 
> 0x400556e5: strlen in ROM
> 
> 
> 
> Backtrace: 0xfffffffd:0x3fcbc8f0 0x420091d5:0x3fcbc910 0x40383805:0x3fcbc950
> 0x420091d5: detect_Task at /Users/fereshtehmahmudzadeh/Documents/esp-skainet-master/examples/en_speech_commands_recognition_copy/main/main.c:88
> 
> 0x40383805: vPortTaskWrapper at /Users/fereshtehmahmudzadeh/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
> 
> 
> 
> 
> 
> ELF file SHA256: 88e5fda1eced1711
> 
> Rebooting...

feizi, 2024-01-18:
Hi @femahmudzadeh , 
I can not reproduce your log. What's your version of esp-sr. I recommend you update to esp-sr v1.6.0.

femahmudzadeh, 2024-01-18:
> Hi @femahmudzadeh , I can not reproduce your log. What's your version of esp-sr. I recommend you update to esp-sr v1.6.0.

I updated esp-sr as you recommended and worked. Thank you so much for your help.