espressif/esp-skainet #129: ESP32-Korvo运行wake_word_detection例程出现IO口映射错误导致不停自动reboot怎么解决？

HaoWenT, 2024-03-05:
load:0x40080404,len:4312
entry 0x4008065c
I (29) boot: ESP-IDF v5.2 2nd stage bootloader
I (29) boot: compile time Mar  5 2024 20:39:27
I (29) boot: Multicore bootloader
I (33) boot: chip revision: v3.0
I (37) qio_mode: Enabling default flash chip QIO
I (42) boot.esp32: SPI Speed      : 80MHz
I (47) boot.esp32: SPI Mode       : QIO
I (52) boot.esp32: SPI Flash Size : 16MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 factory          factory app      00 00 00010000 00500000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=29c0ch (171020) map
I (137) esp_image: segment 1: paddr=00039c34 vaddr=3ffb0000 size=026f0h (  9968) load
I (140) esp_image: segment 2: paddr=0003c32c vaddr=40080000 size=03cech ( 15596) load
I (147) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=35328h (217896) map
I (207) esp_image: segment 4: paddr=00075350 vaddr=40083cec size=101dch ( 66012) load
I (237) boot: Loaded app from partition at offset 0x10000
I (237) boot: Disabling RNG early entropy source...
I (248) cpu_start: Multicore app
I (248) quad_psram: This chip is ESP32-D0WD
I (249) esp_psram: Found 8MB PSRAM device
I (251) esp_psram: Speed: 80MHz
I (254) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (262) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (785) esp_psram: SPI SRAM memory test OK
I (793) cpu_start: Pro cpu start user code
I (793) cpu_start: cpu freq: 240000000 Hz
I (793) cpu_start: Application information:
I (796) cpu_start: Project name:     wake_word_detection
I (802) cpu_start: App version:      v0.3.0-308-ge5c8310
I (808) cpu_start: Compile time:     Mar  5 2024 21:20:49
I (814) cpu_start: ELF file SHA256:  71a9e7571...
I (819) cpu_start: ESP-IDF:          v5.2
I (824) cpu_start: Min chip rev:     v0.0
I (829) cpu_start: Max chip rev:     v3.99 
I (834) cpu_start: Chip rev:         v3.0
I (838) heap_init: Initializing. RAM available for dynamic allocation:
I (846) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (852) heap_init: At 3FFB34F8 len 0002CB08 (178 KiB): DRAM
I (858) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (864) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (871) heap_init: At 40093EC8 len 0000C138 (48 KiB): IRAM
I (877) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
I (885) spi_flash: detected chip: generic
I (889) spi_flash: flash io: qio
W (893) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (904) main_task: Started on CPU0
I (914) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (914) main_task: Calling app_main()
E (914) esp_clock_output: esp_clock_output_start(188): Selected io is already mapped by another signal
E (924) i2s_common: i2s_check_set_mclk(755): mclk configure failed
E (934) i2s_std: i2s_std_set_gpio(162): mclk config failed
E (934) i2s_std: i2s_channel_init_std_mode(225): initialize channel failed while setting gpio pins
E (944) i2s_common: i2s_channel_enable(995): the channel has already enabled or not initialized
I (964) ES7210: Work in Slave mode
I (964) ES7210: Enable ES7210_INPUT_MIC1
I (964) ES7210: Enable ES7210_INPUT_MIC2
I (974) ES7210: Enable ES7210_INPUT_MIC3
I (974) ES7210: Enable ES7210_INPUT_MIC4
I (984) ES7210: Enable TDM mode
E (984) i2s_common: i2s_channel_disable(1021): the channel has not been enabled yet
I (994) I2S_IF: channel mode 0 bits:32/32 channel:2 mask:3
E (994) i2s_std: i2s_channel_reconfig_std_slot(317): invalid state, I2S should be disabled before reconfiguring the slot
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400db5a4  PS      : 0x00060330  A0      : 0x800db702  A1      : 0x3ffb5540
0x400db5a4: set_drv_fs at F:/Github/esp-skainet/examples/wake_word_detection/managed_components/espressif__esp_codec_dev/platform/audio_codec_data_i2s.c:137

A2      : 0x3f800cfc  A3      : 0x00000000  A4      : 0x00000020  A5      : 0x3ffbe6cc
A6      : 0x00000003  A7      : 0x00000020  A8      : 0x00000000  A9      : 0x3ffb5510
A10     : 0x00000103  A11     : 0x3ffb5574  A12     : 0x3f4078e0  A13     : 0x000003e2
A14     : 0x3f4077bc  A15     : 0x3f800cfc  SAR     : 0x00000004  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x40089445  LEND    : 0x40089455  LCOUNT  : 0xfffffff8
0x40089445: strlen at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:84
0x40089455: strlen at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:96

Backtrace: 0x400db5a1:0x3ffb5540 0x400db6ff:0x3ffb55d0 0x400db76d:0x3ffb5610 0x400db935:0x3ffb5640 0x400dad4e:0x3ffb5670 0x400da74a:0x3ffb5690 0x400da831:0x3ffb56f0 0x400da8c1:0x3ffb5710 0x400da4c1:0x3ffb5730 0x400da3df:0x3ffb5750 0x4010499f:0x3ffb57f0 0x4008c87d:0x3ffb5820
0x400db5a1: set_drv_fs at F:/Github/esp-skainet/examples/wake_word_detection/managed_components/espressif__esp_codec_dev/platform/audio_codec_data_i2s.c:137
0x400db6ff: set_fs at F:/Github/esp-skainet/examples/wake_word_detection/managed_components/espressif__esp_codec_dev/platform/audio_codec_data_i2s.c:2420x400db76d: check_fs_compatible at F:/Github/esp-skainet/examples/wake_word_detection/managed_components/espressif__esp_codec_dev/platform/audio_codec_data_i2s.c:265
0x400db935: _i2s_data_set_fmt at F:/Github/esp-skainet/examples/wake_word_detection/managed_components/espressif__esp_codec_dev/platform/audio_codec_data_i2s.c:440
0x400dad4e: esp_codec_dev_open at F:/Github/esp-skainet/examples/wake_word_detection/managed_components/espressif__esp_codec_dev/esp_codec_dev.c:179    
0x400da74a: bsp_codec_adc_init at F:/Github/esp-skainet/components/hardware_driver/boards/esp32-korvo/bsp_board.c:114
0x400da831: bsp_codec_init at F:/Github/esp-skainet/components/hardware_driver/boards/esp32-korvo/bsp_board.c:368
0x400da8c1: bsp_board_init at F:/Github/esp-skainet/components/hardware_driver/boards/esp32-korvo/bsp_board.c:424
0x400da4c1: esp_board_init at F:/Github/esp-skainet/components/hardware_driver/esp_board_init.c:30
0x400da3df: app_main at F:/Github/esp-skainet/examples/wake_word_detection/main/main.c:78
0x4010499f: main_task at C:/Espressif/frameworks/esp-idf-v5.2/components/freertos/app_startup.c:208
0x4008c87d: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

HaoWenT, 2024-03-05:
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7524
load:0x40078000,len:16520
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:4312
entry 0x4008065c
I (29) boot: ESP-IDF v5.2 2nd stage bootloader
I (29) boot: compile time Mar  5 2024 20:39:27
I (29) boot: Multicore bootloader
I (33) boot: chip revision: v3.0
I (37) qio_mode: Enabling default flash chip QIO
I (42) boot.esp32: SPI Speed      : 80MHz
I (47) boot.esp32: SPI Mode       : QIO
I (52) boot.esp32: SPI Flash Size : 16MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 factory          factory app      00 00 00010000 00500000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=29c0ch (171020) map
I (137) esp_image: segment 1: paddr=00039c34 vaddr=3ffb0000 size=026f0h (  9968) load
I (140) esp_image: segment 2: paddr=0003c32c vaddr=40080000 size=03cech ( 15596) load
I (147) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=35328h (217896) map
I (207) esp_image: segment 4: paddr=00075350 vaddr=40083cec size=101dch ( 66012) load
I (237) boot: Loaded app from partition at offset 0x10000
I (237) boot: Disabling RNG early entropy source...
I (248) cpu_start: Multicore app
I (248) quad_psram: This chip is ESP32-D0WD
I (249) esp_psram: Found 8MB PSRAM device
I (251) esp_psram: Speed: 80MHz
I (254) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (262) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (785) esp_psram: SPI SRAM memory test OK
I (793) cpu_start: Pro cpu start user code
I (793) cpu_start: cpu freq: 240000000 Hz
I (793) cpu_start: Application information:
I (796) cpu_start: Project name:     wake_word_detection
I (802) cpu_start: App version:      v0.3.0-308-ge5c8310
I (808) cpu_start: Compile time:     Mar  5 2024 21:20:49
I (814) cpu_start: ELF file SHA256:  71a9e7571...
I (819) cpu_start: ESP-IDF:          v5.2
I (824) cpu_start: Min chip rev:     v0.0
I (829) cpu_start: Max chip rev:     v3.99 
I (834) cpu_start: Chip rev:         v3.0
I (838) heap_init: Initializing. RAM available for dynamic allocation:
I (846) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (852) heap_init: At 3FFB34F8 len 0002CB08 (178 KiB): DRAM
I (858) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (864) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (871) heap_init: At 40093EC8 len 0000C138 (48 KiB): IRAM
I (877) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
I (885) spi_flash: detected chip: generic
I (889) spi_flash: flash io: qio
W (893) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (904) main_task: Started on CPU0
I (914) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (914) main_task: Calling app_main()
E (914) esp_clock_output: esp_clock_output_start(188): Selected io is already mapped by another signal
E (924) i2s_common: i2s_check_set_mclk(755): mclk configure failed
E (934) i2s_std: i2s_std_set_gpio(162): mclk config failed
E (934) i2s_std: i2s_channel_init_std_mode(225): initialize channel failed while setting gpio pins
E (944) i2s_common: i2s_channel_enable(995): the channel has already enabled or not initialized
[... 226 lines cut ...]

Core  0 register dump:
PC      : 0x400db5a4  PS      : 0x00060330  A0      : 0x800db702  A1      : 0x3ffb5540
0x400db5a4: set_drv_fs at F:/Github/esp-skainet/examples/wake_word_detection/managed_components/espressif__esp_codec_dev/platform/audio_codec_data_i2s.c:137

A2      : 0x3f800cfc  A3      : 0x00000000  A4      : 0x00000020  A5      : 0x3ffbe6cc
A6      : 0x00000003  A7      : 0x00000020  A8      : 0x00000000  A9      : 0x3ffb5510
A10     : 0x00000103  A11     : 0x3ffb5574  A12     : 0x3f4078e0  A13     : 0x000003e2
A14     : 0x3f4077bc  A15     : 0x3f800cfc  SAR     : 0x00000004  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x40089445  LEND    : 0x40089455  LCOUNT  : 0xfffffff8
0x40089445: strlen at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:84
0x40089455: strlen at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:96

Backtrace: 0x400db5a1:0x3ffb5540 0x400db6ff:0x3ffb55d0 0x400db76d:0x3ffb5610 0x400db935:0x3ffb5640 0x400dad4e:0x3ffb5670 0x400da74a:0x3ffb5690 0x400da831:0x3ffb56f0 0x400da8c1:0x3ffb5710 0x400da4c1:0x3ffb5730 0x400da3df:0x3ffb5750 0x4010499f:0x3ffb57f0 0x4008c87d:0x3ffb5820
0x400db5a1: set_drv_fs at F:/Github/esp-skainet/examples/wake_word_detection/managed_components/espressif__esp_codec_dev/platform/audio_codec_data_i2s.c:137
0x400db6ff: set_fs at F:/Github/esp-skainet/examples/wake_word_detection/managed_components/espressif__esp_codec_dev/platform/audio_codec_data_i2s.c:2420x400db76d: check_fs_compatible at F:/Github/esp-skainet/examples/wake_word_detection/managed_components/espressif__esp_codec_dev/platform/audio_codec_data_i2s.c:265
[... cut ...]
0x400da8c1: bsp_board_init at F:/Github/esp-skainet/components/hardware_driver/boards/esp32-korvo/bsp_board.c:424
0x400da4c1: esp_board_init at F:/Github/esp-skainet/components/hardware_driver/esp_board_init.c:30
0x400da3df: app_main at F:/Github/esp-skainet/examples/wake_word_detection/main/main.c:78
0x4010499f: main_task at C:/Espressif/frameworks/esp-idf-v5.2/components/freertos/app_startup.c:208
0x4008c87d: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

sun-xiangyu, 2024-03-06:
你使用的是哪款开发板？

HaoWenT, 2024-03-06:
ESP32-Korvo

viiv, 2024-06-18:
最后这个问题解决了吗? 我也遇到同样的问题

feizi, 2024-06-18:
esp-idf v5.1 运行是正常的，如果你比较急，建议切换到这个版本
esp-idf v5.2 中i2s的配置被修改了，目前还没有支持