espressif/esp-skainet #132: 关于遇到Deprecated Quantized wakenet5: wakeNet5_v1_hilexin_5_0.95_0.90, mode:0 (Nov 21 2023 19:15:51)

Euphie1, 2024-03-26:
在esp32-korvo-v1.1
上遇到警告
[0m
Deprecated Quantized wakenet5: wakeNet5_v1_hilexin_5_0.95_0.90, mode:0 (Nov 21 2023 19:15:51)
Please switch to ESP32-S3 and wakenet9 !
[0;32mI (1698) AFE_SR: wake num: 1, mode: 1, (Nov 21 2023 19:15:51)
[0m
multinet:mn2_cn
SINGLE_RECOGNITION: v2; (Nov 21 2023 19:40:21)
[0;33mW (1698) MN: MultiNet may cause WatchDog on ESP32 sometimes![0m
那我是无法再使用唤醒词了吗？

使用唤醒词确实也没有反应，但默认固件可以正常运行使用唤醒词

sun-xiangyu, 2024-03-27:
@Euphie1 
wakenet5不在维护和更新了， 应该还是可以使用的。和默认固件不一样，目前的版本阈值定的比较高（为了防止误触发），你多试几次，或是在代码里重新修改一下阈值。

Euphie1, 2024-03-28:
好的，感谢

hyuan-kamuda, 2024-05-07:
> 好的，感谢

请问您后来如何解决的呢？我也遇到同样的问题以及后续死机，用的是ESP32-LyraTD-MSC开发板跑官方ADF sample (examples/speech_recognition/wwe/)：

I (10539) AFE_SR: wakenet_init: 1

Deprecated Quantized wakenet5: wakeNet5_v1_nihaoxiaozhi_6_0.983_0.95, mode:0 (Nov 21 2023 19:15:51)
Please switch to ESP32-S3 and wakenet9 !
I (10659) AFE_SR: wake num: 1, mode: 1, (Nov 21 2023 19:15:51)

W (10659) RECORDER_SR: Multinet is not enabled in SDKCONFIG
I (10669) main_task: Returned from app_main()
Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.

官方文档说的是这个wwe