espressif/esp-skainet #133: 关于multinet返回结果ESP_MN_STATE的问题。

welkinchan, 2024-04-07:
你好，我近期在尝试开发离线语音+在线语音同时实现的方案。
我原本的思路是，默认一直使用离线语音，当用户的语音命令在命令词列表之外时，再调用在线语音的功能。

但是实际测试的时候，发现mn_state只有3种返回结果，detecting,detected和timeout。
当用户的语音命令在命令词列表之外时，mn_state的状态也一直时detecting直到timeout触发。
此时，mn_result->command_id[i]列表也是空的。

所以有几个问题想请教一下：
1. 我的上述的结论是正确的吗？还是我的结论是错误的？
2. 如果我想拿到用户在命令词列表外的语音输入的判断结果的话，是否有办法？
3. 如果我拿不到命令词列表外的语音输入的判断结果的话，那我上述中的离线语音+在线语音同时实现的思路是不是就不可行了？

非常希望能抽空答疑解惑，多谢！

if (detect_flag == 1) {
            esp_mn_state_t mn_state = multinet->detect(model_data, res->data);

            if (mn_state == ESP_MN_STATE_DETECTING) {
                continue;
            }

            if (mn_state == ESP_MN_STATE_DETECTED) {
                esp_mn_results_t *mn_result = multinet->get_results(model_data);
                for (int i = 0; i < mn_result->num; i++) {
                    printf("TOP %d, command_id: %d, phrase_id: %d, string:%s prob: %f\n", 
                    i+1, mn_result->command_id[i], mn_result->phrase_id[i], mn_result->string, mn_result->prob[i]);
                }
                printf("\n-----------listening-----------\n");
            }

            if (mn_state == ESP_MN_STATE_TIMEOUT) {
                esp_mn_results_t *mn_result = multinet->get_results(model_data);
                printf("timeout, string:%s\n", mn_result->string);
                afe_handle->enable_wakenet(afe_data);
                detect_flag = 0;
                printf("\n-----------awaits to be waken up-----------\n");
                continue;
            }
        }

sun-xiangyu, 2024-04-07:
> * 我的上述的结论是正确的吗？还是我的结论是错误的？
> * 如果我想拿到用户在命令词列表外的语音输入的判断结果的话，是否有办法？
> * 如果我拿不到命令词列表外的语音输入的判断结果的话，那我上述中的离线语音+在线语音同时实现的思路是不是就不可行了？

1. 你的理解是对的
2. 如果要拿到词表之外的结果，可以通过 
```
esp_mn_results_t *mn_result = multinet->get_results(model_data);
printf("timeout, string:%s\n", mn_result->string);
```
3. 使用该方法实现离线+在线语音识别，主要的问题是延时，你可以修改multinet timeout时间让他提前返回, 但在没有触发的情况下，multinet拿到结果至少有 500~800 ms的延时，这时你再送到云端加上网络的延时，可能会有 1.5s 左右。

welkinchan, 2024-04-07:
> 1. 你的理解是对的
> 2. 如果要拿到词表之外的结果，可以通过
> 
> ```
> esp_mn_results_t *mn_result = multinet->get_results(model_data);
> printf("timeout, string:%s\n", mn_result->string);
> ```
> 
> 3. 使用该方法实现离线+在线语音识别，主要的问题是延时，你可以修改multinet timeout时间让他提前返回, 但在没有触发的情况下，multinet拿到结果至少有 500~800 ms的延时，这时你再送到云端加上网络的延时，可能会有 1.5s 左右。

非常感谢回复，但您的建议有一个问题我不是很理解。

如果想拿到命令词以外的结果的话，如果要使用如下命令：
 esp_mn_results_t *mn_result = multinet->get_results(model_data);
 printf("timeout, string:%s\n", mn_result->string);
 
 这个命令应该放在什么时候运行呢？
 由于命令词之外的结果并不会让mn_state =ESP_MN_STATE_DETECTED，
 此时mn_state =ESP_MN_STATE_DETECTING，
 这个时候，multinet->get_results(model_data)会有结果返回吗？还是他是空的？
 
 我很想知道multinet->get_results(model_data)在不同的mn_state情况下的值分别是什么样的。

sun-xiangyu, 2024-04-07:
只有mn_state 在 detected 和 timeout 的时候可以拿到

welkinchan, 2024-04-10:
> 只有mn_state 在 detected 和 timeout 的时候可以拿到

请问afe_fetch_result_t *res = afe_handle->fetch(afe_data);
从哪里可以看到res的返回内容的资料吗？

feizi, 2024-04-10:
你可以在esp-sr 中查看相关接口和文档：
https://github.com/espressif/esp-sr/blob/ed2311821412f52457fd75a9eee5c99da8a587dd/include/esp32s3/esp_afe_sr_iface.h#L28

welkinchan, 2024-04-12:
> 只有mn_state 在 detected 和 timeout 的时候可以拿到

> 你可以在esp-sr 中查看相关接口和文档： https://github.com/espressif/esp-sr/blob/ed2311821412f52457fd75a9eee5c99da8a587dd/include/esp32s3/esp_afe_sr_iface.h#L28

好的，非常感谢。

closed 2024-04-12 by welkinchan