espressif/esp-skainet #135: Wake Word Detection failed

beer-to-code, 2024-04-23:
i am using esp32s3-devKit-c1 board  for experimant this project , i tried to flash wake word detetion example but got error , as mentioned below 
also i used the provided skdconfig, as mentioned in the example's readme , cp sdkconfig.default.esp32s3 sdkconfig

`ELF file SHA256: 8e9852ecfd7e6ea5

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379e2c
0x40379e2c: esp_restart_noos at C:/Users/deepe/esp/v5.0/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:158 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1860
load:0x403c9700,len:0xd90
load:0x403cc700,len:0x2f68
entry 0x403c9930
I (29) boot: ESP-IDF v5.0.6-318-g78af5840a1 2nd stage bootloader
I (29) boot: compile time 09:22:34
I (29) boot: Multicore bootloader
I (32) boot: chip revision: v0.2
I (36) qio_mode: Enabling default flash chip QIO
I (42) boot.esp32s3: Boot SPI Speed : 80MHz
I (46) boot.esp32s3: SPI Mode       : QIO
I (51) boot.esp32s3: SPI Flash Size : 16MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 factory          factory app      00 00 00010000 00271000
I (80) boot:  1 model            Unknown data     01 82 00281000 0050c000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=199a4h (104868) map
I (116) esp_image: segment 1: paddr=000299cc vaddr=3fc99b00 size=04df8h ( 19960) load
I (120) esp_image: segment 2: paddr=0002e7cc vaddr=40378000 size=0184ch (  6220) load
I (123) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=5016ch (328044) map
I (180) esp_image: segment 4: paddr=00080194 vaddr=4037984c size=10298h ( 66200) load
I (200) boot: Loaded app from partition at offset 0x10000
I (200) boot: Disabling RNG early entropy source...
I (212) cpu_start: Multicore app
E (212) octal_psram: PSRAM ID read error: 0x00000000, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (216) cpu_start: Failed to init external RAM!

abort() was called at PC 0x40379744 on core 0
0x40379744: call_start_cpu0 at C:/Users/deepe/esp/v5.0/esp-idf/components/esp_system/port/cpu_start.c:456 (discriminator 3)

Backtrace: 0x4037a1be:0x3fceb220 0x403810c1:0x3fceb240 0x403874f6:0x3fceb260 0x40379744:0x3fceb2d0 0x403cdb96:0x3fceb330 0x403cdfa1:0x3fceb380 0x403c9999:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
0x4037a1be: panic_abort at C:/Users/deepe/esp/v5.0/esp-idf/components/esp_system/panic.c:425

0x403810c1: esp_system_abort at C:/Users/deepe/esp/v5.0/esp-idf/components/esp_system/esp_system.c:153

0x403874f6: abort at C:/Users/deepe/esp/v5.0/esp-idf/components/newlib/abort.c:38

0x40379744: call_start_cpu0 at C:/Users/deepe/esp/v5.0/esp-idf/components/esp_system/port/cpu_start.c:456 (discriminator 3)

ELF file SHA256: 8e9852ecfd7e6ea5
[image]
`

feizi, 2024-04-23:
> E (212) octal_psram: PSRAM ID read error: 0x00000000, PSRAM chip not found or not supported, or wrong PSRAM line mode
> E (216) cpu_start: Failed to init external RAM!

1. Please check whether your ESP32-S3 module supports PSRAM(>= 8 MB)
2. If you do not use the boards[ in the list](https://github.com/espressif/esp-skainet?tab=readme-ov-file#hardware-preparation), you need to add driver for your board. Please refer to [here](https://github.com/espressif/esp-skainet/blob/master/components/hardware_driver/boards/esp32s3-korvo-2/bsp_board.c)

SleepInfinity, 2024-10-31:
Have you tried to switch the PSRAM mode from DIO to QUAD or vise versa ?

russelh15, 2025-06-26:
If the board does not support PSRAM, can it still be used if we add it to the list?